DSheirer/sdrtrunk #291: MPT1327 Channel Map - Unknown Frequency

Sparkz71, 2017-11-24:
I'm trying to follow MPT1327 traffic channels. I have entered a channel map with 2 ranges (1-251 & 301-364).  Listening to the control ch' - under 'events' - a channel number is shown for traffic (eg 323), however the 'details' state "unknown frequency".  P25 works. 
Thanks

Sparkz71, 2017-12-17:
I think I found a bug.  I turned on Channel MPT1327 - Decoder - AFC. Now it works, with AFC  turned on or off. The channel maps are now working & following traffic freqs' & details state "GTC".
 
I compared the playlist & the following was added after I selected AFC:
"      <channelMapName>New Channel Map</channelMapName>
</decode_configuration> "

[playlist_v2 - Ch Map Not Working.xml.txt](https://github.com/DSheirer/sdrtrunk/files/1565717/playlist_v2.-.Ch.Map.Not.Working.xml.txt)

[playlist_v2 - Ch Map Works.xml.txt](https://github.com/DSheirer/sdrtrunk/files/1565714/playlist_v2.-.Ch.Map.Works.xml.txt)

[playlist_v2 - Ch Map Works - AFC OFF.xml.txt](https://github.com/DSheirer/sdrtrunk/files/1565715/playlist_v2.-.Ch.Map.Works.-.AFC.OFF.xml.txt)

I hope this helps. I don't think AFC was checked as default when I selected MPT1327. But checking it added the magic line to the playlist & solved the problem! 

Regards,
Ian

DSheirer, 2017-12-19:
Hi Ian!  Do you remember if you hit 'save' after selecting the Channel Map for the MPT1327 configuration (prior to changing AFC)?  It seems that enabling/disabling AFC may have been coincidental, but hitting save after you changed AFC caused the channel map setting to take effect.

When you hit save, it stores all of the settings from the editor, whether you changed any of the values or not.

Sparkz71, 2017-12-26:
Hi Denny. I can't recall, so I just repeated the initial setup with the same results.

I entered the frequency, selected MPT1327 & created a channel map. After saving the channel map I closed the channel map manager window & returned to the decoder settings.

I attempted to select the channel map name from the Channel Map pull-down but it did not appear. I pressed save & now the channel map name is displayed in the selection box.  I pressed save again & restarted the program. The unknown frequency problem exists with the channel map name correctly selected. (*Note: Each time I create a new channel map, it does not appear in the selection box until I press the save button, beside enable ). 

I check all the settings again, press save & repeat. The problem remains the same - unknown frequency. 

Next, I select AFC & press save. Now the correct traffic frequencies are displayed & followed.
 
I can now create a new control channel, repeating the same procedure as before,, select the channel map (new or old) & without selecting AFC - the channel map works correctly.
However, in a previous setup, I had 2 control channels with the unknown frequency problem & I had to select AFC for each channel before the channel map worked. But the channel map would now work with  any new channel created, without selecting AFC.  

I also tried the following once I had the channel map working with only one channel in the channel list (I also restarted sdrtrunk between changes with the same result):
1* Delete channel map & recreate - GTC works.
2* Delete control channel & recreate - GTC works.
3* Delete channel map & control channel & recreate - UNKNOWN FREQUENCY GTC problem.
4* Select AFC - everything works.
5* Repeat - nothing works until I select AFC 

Please tell me if I've made a mistake.