DSheirer/sdrtrunk #293: Alias with 2 control channels

Sparkz71, 2017-12-01:
Minor problem.
I created an xyz alias with a talkgroup id: 7726.
I am monitoring 2 x P25 control channels from different sites. Same network.
When I receive both control/traffic channels, the alias is only converted to xyz on one channel & remains 7726 on the other.

DSheirer, 2017-12-01:
Are both of the channel configurations specifying the same alias list name.  For the channel where the 7726 talkgroup does not get assigned the alias, do you see any of the other aliases working correctly?

Sparkz71, 2017-12-03:
I have specified both control channels with the same alias list - P25.  It’s seems the alias list is ‘stuck’ to one control channel. I defined the talk group alias as 7726 = (eg) fire group. If I listen to 2 x P25 control ch, one detects talkgroup 7726 , as 7726 – ignoring the alias. The other channel converts 7726 to the ‘fire group’ alias. I am listening with ‘stereo’.  I don’t understand Java – just old school BASIC – I wish I could help further.

Regards.... Ian 

From: Denny Sheirer
Sent: Friday, 1 December 2017 8:16 PM
To: DSheirer/sdrtrunk
Cc: Sparkz71; Author
Subject: Re: [DSheirer/sdrtrunk] Alias with 2 control channels (#293)

Are both of the channel configurations specifying the same alias list name. For the channel where the 7726 talkgroup does not get assigned the alias, do you see any of the other aliases working correctly?

DSheirer, 2017-12-03:
Could you please post your playlist.xml file here?

Note: if you have setup audio streaming, your streaming password will be listed in the file and would be available for anyone to see.  You can delete your streaming configuration(s) before posting it.

Sparkz71, 2017-12-04:
False error, my mistake - Sorry! I apologise. Please close this issue. I had 2 similar alias list names & set the wrong list on 1 channel.

closed 2017-12-04 by Sparkz71