DSheirer/sdrtrunk #296: What are the chances of Tone Coded Squelch and P25 Phase II

photounit, 2017-12-03:
Denny, 
Thanks for all your efforts in writing SDR Trunk. I was wondering what the odds would be in adding TCS (or at least CSQ) and P25 Phase II  to SDRTrunk ....   I would be willing to make a contribution to your PayPal account to assist in your effort :-) .

DSheirer, 2017-12-04:
Issue #80 is the requirement for the Tone coded squelch.

I'll update the title on this one to P25 Phase 2.

I have planned to do Phase 2, but I'm finishing the Polyphase Channelizer and a DMR decoder is the next big thing in the hopper.  There's a Phase 2 system coming online near where I work, so I'll be able to work on this after I finish the first two issues.

omgitsgela, 2018-03-20:
We are eagerly awaiting P25 Phase 2 here in our locale. Unitrunker is awful software for our needs, and yours seems to be written much better. We'd love to get our hands on a Phase 2 working build!

One of the things Unitrunker failed us with is trunking specific talkgroups to output audio channels. Our system is being used as a multi audio broadcast server, and we need to pull specific talkgroups out of the system and put them to separate audio channels. Unitrunker was configured with an archaic scan ranking system, which meant that it was impossible to make direct assignments, the scans would overlap no matter what we would do. A good implementation of trunked radio scanning would allow you to set a talkgroup to output to a specific audio card, and create multiple hooks like this.

CGUTH, 2018-10-10:
Looking forward to Phase 2 support, Maricopa County (P25) (AZ) is in the testing phase now and they are using P25 Phase 2 for their test radios.

jkcameron, 2019-02-01:
Hey guys, I was just checking on the status of the p25 p2 update?  I was trying to decide if I should purchase a scanner or wait for this update.  Could I trouble you for a quick update on your progress.  Thanks so much... Appreciate you.

jkcameron, 2019-02-01:
OK well thank you for your time... I'm know I speak for everyone.  When you are finished I expect to see SDRTrunk's user-base increase dramatically.  Kick ass.

LordDonkey, 2019-06-20:
Any ETA on when Phase 2 will be ready?

DSheirer, 2019-06-22:
I finished cleaning up the IMBE (Phase 1) support in the JMBE library ... fixed a couple issues.  

Right now I'm finishing up the JMBE creator component that allows users to build the JMBE library without having to install Java or Gradle.  This seems to be a major hurdle for sdrtrunk users, so I'm trying to make it easier.

I still have the Phase 2 decoder state section to code, but the rest of the phase 2 decoder is finished.

I'm at least 3-4 weekends away from the first usable test build.

Dizzy-Pi, 2019-06-22:
Great news Denny. No rush, no one is doing what you are. Loved this from
day one!

jkcameron, 2019-06-24:
You are the man.

hunterbeach, 2019-11-17:
Any updates on Phase 2?

memtech3, 2019-11-18:
does this mean its ready?

memtech3, 2019-11-18:
Just built with gradle and found that when the jmbe library is selected sdrtrunk refuses to start receiving (clicking the start button under the channels tab doesn't do anything)

DSheirer, 2019-11-18:
@memtech3 Did you see any errors in the log when this happened?

memtech3, 2019-11-18:
I can privately send my log file

memtech3, 2019-11-18:
Here is my log file a brief glance shows no errors except for when I loaded the wrong jar and unplugged my sdr
[20191117_sdrtrunk_app.log](https://github.com/DSheirer/sdrtrunk/files/3861101/20191117_sdrtrunk_app.log)

DSheirer, 2019-11-19:
@memtech3 If you're running sdrtrunk out of master, it requires JMBE library version 1.0.0.  In the log file, you're using the older version 0.3.3 which doesn't have support for Phase II audio codec (AMBE).

memtech3, 2019-11-19:
I see...

memtech3, 2019-11-19:
can you add the current version to the releases tab on that repo?

hans109h, 2019-11-19:
Some of the code used is copyrighted so a binary cannot be provided for download.  You need to build the library yourself by following the instructions in the readme https://github.com/DSheirer/jmbe/blob/master/README.md

If I remember correctly Denny has updated the gradle wrapper to work with Java 11, but I think the readme stll refers to 8+, but any version should work.  

After a successful build the location of the jar file will be echoed and you can insert that in the sdrTrunk preferences.

closed 2019-11-26 by DSheirer