DSheirer/sdrtrunk #297: Adjust Bitrate For Cutom Icecast Stream

nomorecats, 2017-12-06:
Hi Denny,

If possible, could users be able to define their on bitrate for Icecast2.4. I am currently hosting my own Icecast server and would like to decrease latency.

DSheirer, 2017-12-06:
Changes to bitrate will primarily affect the audio quality (ie higher bitrates = better audio quality), but won't reduce latency.  

Are you trying to reduce the latency between transmission time and streaming time?  I think right now a call has to close out before it's queued for streaming which introduces a latency equal to the call duration, in addition to the streaming audio queue size.

nomorecats, 2017-12-07:
Better audio quality would be nice too, I don't have Intellij to I would recompile it myself. Is there any way to output audio by means of the speaker talkgroup selectively? In past versions I've had to "block" all other talkgroups.

DSheirer, 2017-12-07:
Audio (speaker) output is still controlled the same way using aliases by setting audio priority and/or setting a 'Do Not Monitor'

DSheirer, 2023-11-15:
Additional bit rates added for streaming.

closed 2023-11-15 by DSheirer