DSheirer/sdrtrunk #298: Create alias from Now Playing window

DSheirer, 2017-12-06:
Add capability to create an alias from any of the primary or secondary identifier fields in each row of the now playing table for any channel or traffic channel that is displayed.

This should launch you over to the Aliases window with a new alias created and the appropriate alias identifier type auto-created from the values.  Also, the alias list should be auto-filled from the alias list specified for the channel.

Sparkz71, 2017-12-09:
Great idea. If I identified a unit id / talkgroup whilst listening etc - I'd have to write it down, then add the alias.

Sparkz71, 2017-12-09:
Regarding ID's, can you add an option to convert HEX to decimal?

DSheirer, 2018-02-02:
Add ability to assign the id to an existing alias or simply create a new alias for the alias identifier.