DSheirer/sdrtrunk #307: No audio after gradle compile on Windows

endorphinrush, 2017-12-31:
Not a bug just a suggestion.

If following "[Build the project](https://github.com/DSheirer/sdrtrunk#instructions)" instructions, sdrtrunk produces no audio. This is not resolved until adding jmbe back to <../build/libs>. 

<This would add gradle, download all project dependencies and build JAR>

Would this be considered a dependency that should be installed during build? If not another line might be useful to indicate the additional dependency needs.

DSheirer, 2018-01-01:
The JMBE audio library is excluded from the normal build (and release) process due to potential copyright restrictions.  We leave it to the end-user to compile JMBE and co-locate the library in the execution folder to enable P25 audio decoding.

closed 2018-01-01 by DSheirer

endorphinrush, 2018-01-01:
No problem, I get that. It was just a suggestion for the readme file. Thanks for the hard work!

chudgoo, 2018-01-01:
Would there be anything wrong with including a routine that checks for JMBE
and fetches/installs if needed?
You're not including the code itself in your software, but could provide an
automated way for it to be compiled. Maybe a prompt like "are you sure you
agree to the terms of the third party EULA?   Seems fairly safe from a
legal standpoint.... I think...  ???