DSheirer/sdrtrunk #331: Broadcastify connecting, but not streaming audio

KD0FTJ, 2018-02-10:
I suspect that I am missing some simple thing, but I cannot seem to get audio to stream to my Broadcastify feed: "Queued", "Streamed", and "Aged Off" are all zeros.  The speaker audio changes to active calls, so the hardware setup is working and SDRtrunk is distinguishing intelligible signals.  I do not see anything obvious in the log that looks like it is attempting to stream to Broadcastify and failing.

Please offer any suggestions for what may be missing from my configuration below:

1) I setup the stream and SDRtrunk connects to the feed and Broadcastify confirms that it is connected, so apparently the server settings are correct.

2) I setup aliases and specified the "Broadcast Channel".
[image]
[image]

3) I entered and enabled frequencies of interest, specifying an "Alias List" that had the "Broadcast Channel" specified.
[image]

Thank you for sharing any suggestions.

DSheirer, 2018-02-10:
You're really close.  The missing piece is the talkgroup identifiers.  Each alias can have one or more Talkgroups and/or Radio Identifiers.  They have to be in hexadecimal format with 4-characters for Talkgroups and 6-characters for radio identifiers.  Use leading zeros to pad out the values to either 4 or 6 characters.

You can also use the asterisk (*) as a wildcard for each and every digit in a talkgroup or radio identifier.  

So, to illustrate, pick an alias and add a talkgroup identifier with a value of **** 4 asterisks.  This will cause that alias to match every call and make the audio from that call stream to broadcastify.  You can later change that talkgroup to be specific for the alias, and then add talkgroup identifiers to each of the other aliases.

DSheirer, 2018-02-11:
Please let me know if/when the issue is resolved.  You can either close this ticket or add a comment and I'll close it.

wolfmd, 2020-06-03:
Thank you so much for this answer and this software! I'm a noob to radio, but you've made it easy. Much love

DSheirer, 2020-06-03:
@wolfmd you're welcome.

closed 2023-11-15 by DSheirer