DSheirer/sdrtrunk #332: Alias priority should prevent 'Channel Start Rejected' errors

davidmckenzie, 2018-02-11:
Hey Denny,

Apologies if this is fixed in a newer version, I still haven't had a chance to update from 0.3.0.

One thing I'm seeing during high traffic days is that the 'channel start rejected' errors will appear when the only other channels being monitored are those with a 'do not monitor' priority.

To solve this I'm thinking one of three things: 
* The priority should determine whether channel start is rejected or not (taking the channel from another lower priority alias if necessary), or
* If an alias is "do not monitor" and there are no streams or recording methods attached, then don't start the channel, or
* We could have a flag for "_really_ do not monitor" aliases, that prevents channel start from ever occurring on that alias.

This would allow us to blacklist certain aliases from taking up resources when they're not used for audio or streaming.

Hoping that makes sense :)

Thanks,
Dave

waterbwuk, 2020-05-21:
I see this is still open. Was anything ever implemented that makes this work? I'm trying to monitor a HUGE system with tons of traffic, but only care about 3 or 4 talkgroups out of the hundreds on there and the SDR's are getting tied up on channels that I don't care about. @DSheirer

DSheirer, 2020-05-22:
I haven't made any progress on this.

Bisquit4407, 2021-11-12:
This would be a great help. I'm not monitoring a huge system but I have limited resources. I'm missing audio I need because my tuner is locked onto a channel that is encrypted and is marked for no audio (muted).

closed 2023-11-15 by DSheirer

DiivM5, 2025-03-06:
I know it's been awhile, but has there been any update on this? For me, it's rejecting channels that I actually want to listen to in favour of channels that I have listen turned off.