DSheirer/sdrtrunk #335: Application startup error when no output sound mixer available

DSheirer, 2018-02-22:
When the host computer does not have any speakers plugged in, the system may not provide an output mixer and it produces this error, preventing the application from starting:

Exception in thread "main" java.lang.IllegalArgumentException: Mixer not supported: null
        at java.desktop/javax.sound.sampled.AudioSystem.getMixer(Unknown Source)
        at io.github.dsheirer.audio.AudioManager.loadSettings(AudioManager.java:97)
        at io.github.dsheirer.audio.AudioManager.<init>(AudioManager.java:85)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:143)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:200)

ImagoTrigger, 2018-03-11:
see #213

DSheirer, 2023-11-09:
Closing.  Issue likely resolved under #660

closed 2023-11-09 by DSheirer