DSheirer/sdrtrunk #343: SDRtrunk hardware support for LimeSDR?

Hans14, 2018-03-26:
Hello, is it possible to add LimeSDR hardware support for SDRtrunk or somekinda plugin that enables SDRtrunk to work with LimeSDR?

DSheirer, 2018-03-26:
Yes.  I'll be adding LimeSDR support.

FFY00, 2019-06-01:
@DSheirer any ETA on this? It shouldn't be that hard since it can work as an UHD device using SoapyUHD. It would also make sense for you to support plain SoapySDR as that would bring a wide range of supported hardware.

DSheirer, 2019-06-02:
No updates yet.  It's still planned, but I'm working on issue #296 at the moment.

vulet, 2019-12-29:
I see Phase II has hit beta, awesome work @DSheirer!

bossco79, 2020-06-08:
i'm waiting for you)))

tibbon, 2020-06-11:
So excited for this!

ivkraft, 2022-09-29:
There is no updates?

jpooley-hearst, 2024-08-05:
Bumping this request

DSheirer, 2026-11-15:
Imlemented.

closed 2026-11-15 by DSheirer