DSheirer/sdrtrunk #346: Ignore Sync bit parameters on mpt1327

fe8769a, 2018-04-18:
European mpt1327 mainly french mpt1327 are using ignore sync bit parameter . It was added after my request to trunkview . Is it possible to have it or any explaination for the idle status displayed on STATE

DSheirer, 2018-04-20:
The 'IDLE' state means that it is NOT decoding correctly.

Can you please make an audio recording of the MPT1327 control channel and send it to me?  I'll resolve the issue with the decoding.

You can either post a download link here or I can send you my email address.

Denny

fe8769a, 2018-05-08:
MPT1327 signals are very strong . For sure , I asked to trunkview
developper and trunksniffer (many , many time ago) to modify software and
include "ignore sync bit" parameter
You can check in trunkview settings
As always It is a french particularity in MPT1327
Pascal

DSheirer, 2018-07-01:
Support for the French variant of MPT1327 was added previously.  In the channel configuration, use the dropdown list to select the 'French' variant.  Please reopen this issue if this does not resolve your problem.

closed 2018-07-01 by DSheirer

fe8769a, 2018-07-01:
Is the Version 0.3.4 Beta 2
<https://github.com/DSheirer/sdrtrunk/releases/tag/v0.3.4-beta.2>
version the right one for this "french" modification ?

thanks for answer

fe8769a, 2018-07-01:
Compilation throught JDK with error

fe8769a, 2018-07-06:
Hello

I did compilation of 0.40 alpha . I suppose that i have to fill the
"channel map editor" but i'm not sure . In fact the ignore Sync bit is an
option of the way mpt1327 signal is decoded . you can take exemple with
trunview (there is a special option "use sync bit" .
please give me the menu I have to go to check this

Thanks