DSheirer/sdrtrunk #348: Alias Actions are not fully working in P25 Decoder

DSheirer, 2018-04-22:
Not all P25 messages expose their aliasable identifiers via the getAliases() method.  This causes the Alias Actions to not be executed.

DCHuber, 2018-08-13:
Unsure if this is a related issue or new issue, but posting here since it appears to be the same as above.   

V0.3.2 on Windows, P25 Phase 1, Normal (C4FM)

Identified talkgroups are not always recognized, causing the audio to play for aliases set to "Do Not Monitor".    

In the now playing view, the "To" field will show the talkgroup identifier, but the "Alias" field will be blank even though the talkgroup is added in the Audio/Identifier list.    Editing/saving the specific Talkgroup in the alias resolves the issue.

closed 2018-12-24 by DSheirer

ChrisABQ, 2021-04-29:
Version 0.5.0 Alpha 6 - Alias Actions "Play Clip" does not function over speakers or stream.  Is this still a bug?  P25 LSM system monitored by Windows 10