DSheirer/sdrtrunk #352: Freezing up every couple days

nr0q, 2018-05-04:
Running sdtrunk on the latest version of CentOS with a minimal desktop, I have to restart the computer every couple of days. I'm only monitoring one P25 system and the control channel will switch to "Idle" and won't decode until after a reboot.

DSheirer, 2018-07-01:
This issue should be resolved with the recent commit for feature branch #39 and will be available in upcoming release 0.4.x

closed 2018-07-01 by DSheirer

kd4ab, 2018-08-29:
Running sdr-trunk-all-0.4.0-alpha.2.jar on Fedora 28 (Red Hat) LXDE desktop on an Intel celeron 2957U in an ASUS mini with 4GB RAM.
Using a single RTL-SDR Nooelec NESDR Mini 2+.  Sample rate shows 2.4M complex samples per second.
Decoding a one control channel, 3 voice channel APCO P25 system to decode a single talk group.  That single talk group is streamed to an audio server.
Works fine for some portion of the day.  Then fails with the GUI active, but the Control channel shows "Idle".
If I am running properly, the CPU utilization is about 66 to 68 percent.  When in "Idle" fail mode, it drops to 56-57 percent.  If I am running properly and just stop the channel, the percent utilization stays in the mid 60 percent range.
Stopping and restarting the java main program restores the system until the next random channel dropping to "Idle" on the Control channel.

This appears similar to nr0q's observation from back in May.

If there are nightly builds with potential corrections for this "Idle" problem, I will be happy to download and try a new jar.

As an aside, this is excellent software and amazingly capable to do just what I need using limited hardware.
Thanks for all your hard work and continuing efforts.

kd4ab, 2018-08-29:
In feeble attempt to see if feature enhancement #39 to 0.4.0 is causing "Idle" problem:
Falling back to 0.3.4-beta.2
Pausing the waterfall.
CPU utilization 85 percent.  Running OK.
Will see if this stays online.

kd4ab, 2018-08-30:
Locked up Control channel showing "Idle" appeared after 18 hours using 0.3.4-beta.2.  Changing to new RTL-SDR NESDR Nano2+ and Intel i3 CPU with 16GB.  Upgrading to 0.4.0-alpha.2 tried earlier.