DSheirer/sdrtrunk #355: Need help with alias's not working

ImDroided, 2018-06-09:
So I have a issue I can't solve. I used the script linked elsewhere here to take my talkgroup csv from RadioReference.com and convert it ti the xml format. I inserted the output into playlist_v2.xml and the alias's show up under the alias tab. The issue is it will not display those on the now playing screen. Calls just show up as a talkgroup id. I am stumped on why this is happening. I noticed I have 3 and 4 letter talkgroups and the 3 letter ones show up with a 0 in front so A31 is displaying as 0A31 in the primary to field. But it isnt even displaying the 4 letter ones either. I know this is something stupid that I am missing here. 

Here are some screenshots and a small sample of my playlist_v2.xml file I cut it down it has a ton of talkgroups 

[image]
[image]
[image]
[image]

`<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<playlist>
<alias color="-16777216" iconName="No Icon" list="StarComm DuPage" name="DC Police 1E: Police 1 East Lombard Villa Park - Law Dispatch - DuPage County  DUCOMM">
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="talkgroupID" talkgroup="A2F"/>
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="priority" priority="50"/>
    </alias>
    <alias color="-16777216" iconName="No Icon" list="StarComm DuPage" name="DC Police 3E: Police 3 East Elmhurst Oakbrook Terrace - Law Dispatch - DuPage County  DUCOMM">
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="talkgroupID" talkgroup="A30"/>
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="priority" priority="50"/>
    </alias>
    <alias color="-16777216" iconName="Police" list="StarComm DuPage" name="DC Police 1W: Police 1 West Bartlett Carol Stream - Law Dispatch - DuPage County  DUCOMM">
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="priority" priority="1"/>
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="talkgroupID" talkgroup="A31"/>
    </alias>
    <alias color="-16777216" iconName="No Icon" list="StarComm DuPage" name="DC Police 3W: Police 3 West Glen Ellyn Wheaton - Law Dispatch - DuPage County  DUCOMM">
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="talkgroupID" talkgroup="A32"/>
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="priority" priority="50"/>
    </alias>
    <alias color="-16777216" iconName="No Icon" list="StarComm DuPage" name="DC Police 7W: Police 7 West Warrenville West Chicago Winfield - Law Dispatch - DuPage County  DUCOMM">
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="talkgroupID" talkgroup="A33"/>
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="priority" priority="50"/>
    </alias>
    <alias color="-16777216" iconName="No Icon" list="StarComm DuPage" name="DC Police 1N: Police 1 North Hanover Park Roselle - Law Dispatch - DuPage County  DUCOMM">
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="talkgroupID" talkgroup="A34"/>
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="priority" priority="50"/>
    </alias>
<alias color="-16777216" iconName="No Icon" list="StarComm DuPage" name="Grundy 36030: Grundy County Reserved - Multi-Tac - Grundy County">
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="talkgroupID" talkgroup="8CBE"/>
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="priority" priority="50"/>
    </alias>
    <alias color="-16777216" iconName="No Icon" list="StarComm DuPage" name="Grundy 36031: Grundy County Reserved - Multi-Tac - Grundy County">
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="talkgroupID" talkgroup="8CBF"/>
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="priority" priority="50"/>
    </alias>
    <alias color="-16777216" iconName="No Icon" list="StarComm DuPage" name="NW Rescue Helo: Northwest Rescue Helicopter Based at the Whiteside County AirportRock Falls - EMS Dispatch - AeroMedical  Statewide">
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="talkgroupID" talkgroup="8E45"/>
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="priority" priority="50"/>
    </alias>
    <alias color="-16777216" iconName="No Icon" list="StarComm DuPage" name="Superior Air Med: Superior Air Med Elmhurst  Kankakee  West Chicago - EMS Dispatch - AeroMedical  Statewide">
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="talkgroupID" talkgroup="8EB3"/>
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="priority" priority="50"/>
    </alias>
    <alias color="-16777216" iconName="No Icon" list="StarComm DuPage" name="IDOC K28 TrnAcad: Training Academy K28 Springfield - Corrections - Illinois Department of Corrections IDOC  Statewide">
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="talkgroupID" talkgroup="9097"/>
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="priority" priority="50"/>
    </alias>
    <alias color="-16777216" iconName="No Icon" list="StarComm DuPage" name="testy">
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="priority" priority="43"/>
        <id xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="talkgroupID" talkgroup="0A32"/>
    </alias>
    <channel enabled="true" name="StarComm Dupage" site="Site" system="System">
        <aux_decode_configuration/>
        <decode_configuration xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="decodeConfigP25Phase1" ignore_data_calls="false" modulation="CQPSK" traffic_channel_pool_size="30" afc="false" AFCMaximumCorrection="1000"/>
        <event_log_configuration/>
        <record_configuration/>
        <source_configuration xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:type="sourceConfigTuner" frequency="773843750" source_type="TUNER"/>
    </channel>
</playlist>`

DSheirer, 2018-06-09:
It looks like your channel configuration may be missing the aliaslist name.  That's the final piece that tells the application to use all of your new aliases with calls produced by that channel.

ImDroided, 2018-06-09:
Any links as to how to set that up? I have created a playback xml file and
stuck the output in it. The playback then showed up in the channel tabs
alias drop-down but nothing worked. Maybe I formatted it wrong.

I even added under the alias tab all the talk groups to a list in that
first column and typed that list name in the alias part of the channel tab

Thanks for the reply.

ImDroided, 2018-06-09:
I just redid the setup and it works now. It must not have liked the space in the list name. renamed it to just dupage and it works now. thanks.

closed 2018-06-09 by ImDroided