DSheirer/sdrtrunk #358: Feature Request:  EDACS tracking

DaveNF2G, 2018-07-01:
There is currently only one Windows program that claims to be able to track EDACS trunking.  Said program is difficult and unreliable to use, and support appears to be waning.  SDRTrunk seems to be evolving into the new Windows trunktracking/decoding multi-use program.  There are still many EDACS systems around and this would be a useful addition.

ScanCaster, 2018-07-10:
I triple Double secret request this! I could throw away 3 radios on my streams for a program that could do EDACS, and use the JMBE to do ProVoice!. DOUBLE TRIPLE PLEASE!

See the following which is what started ETrunker which then others took and turned into a closed source non linux program with a some what similar name.

This is the source that was published on UseNet when ETrunker first came about.

[EDACS.C.](http://fccdata.com/EDACS.c.txt)

brookesrook, 2019-06-20:
Also wishing for EDACS and ProVoice support :)

MikonosII, 2020-04-20:
I also support this!!!

milo-trujillo, 2020-06-08:
This would be fantastic :+1:

dmosleyx, 2020-08-07:
Please, please, pleeeeease do this. I would be forever grateful to finally then have an option for a stand alone RPi mobile unit and I could finally incorporate EDACS into my OpenAuto setup.

davidmckenzie, 2022-11-22:
Bumping this as I've found a couple of linux apps supporting EDACS now, so there may be more source to borrow from or integrate in. Sadly my java skills are non existent so I won't be much help except to point fingers here:

https://github.com/lwvmobile/edacs-fm
https://github.com/lwvmobile/dsd-fme/tree/dev#very-experimental-edacsp25nxdn-simplesingle-vfo-trunking

lokyi792, 2023-01-19:
Bump. Only reason I'm still using UniTrunker.

tailnode-rjlprojects, 2024-09-05:
Bump as well, will to throw some beer money too if wanted. I am a sysadmin and pretty good at making things work but Java is out of my skillset, I'd be willing to help anyway I can. To start I took the C code from DSD-FME and did my best to convert it to Java and attached both the original C files used and the Java files that fingers crossed will work. I did take a look in Visial Studio Code and I must say FANTASIC job on the layout and organization of SDRTrunk
[EDACSConv.zip](https://github.com/user-attachments/files/16883058/EDACSConv.zip)