DSheirer/sdrtrunk #362: IceCast2 V2.3.x Streaming - no connection

ScanCaster, 2018-07-12:
I have several IceCast2 2.3.x servers I use for back ups and internally and I can not connect to them, either external or internal ones. 

User Name/Source and Pass are correct.

Attempts to connect, but tail'ng the access and error logs on these servers, noting, but where I access the status page to see if the mount comes up, and they are blank. 

sdtrunk latest release, just is "Connecting...." for the mount I am testing with. Alias/TG's are defined to send to that stream. Tried the intermal (local LAN) server and external backups VPS in DC's.

Using my regular feed software Darkice or EZStream on various Pi's and desktops (*buntu's 12.04-18.04, ALL ESR's) test mounts all come up fine. Servers all *buntu. I don't do windows. 

I'd like to run some tests internally, backup server on another potential issue I ran into, but want to see what happens on this first.

ScanCaster, 2018-07-15:
If needed I can provide access to the external 2.3.x servers for testing.

DSheirer, 2018-08-05:
@ScanCaster I'm working on the sdrtrunk streaming this morning.  I've modified the Icecast 2.3 code to be more robust in connecting/reconnecting and I found an issue in the metadata updater that may have been causing the issue you were seeing with the inability to update metadata from another app while sdrtrunk is streaming.

Can you send me the connection details for your servers so that I can troubleshoot the connection issues you were seeing?

ScanCaster, 2018-08-05:
I will gladly send info on the servers, but I need a way to send it privately.