DSheirer/sdrtrunk #375: 0.4.0 Alpha 1 does not decode Control Channel as well as 0.3.4

infecticide, 2018-07-26:
Hi,

Love the application so far!

I'm trying to get the app running on an RPI 3B+, which I've managed to do (mostly) successfully.   

The 0.4.0_alpha_1 build promises to run better on low power devices.   When I start 0.4.0, the channel auto starts and I can see "CONTROL" for about 2 - 3 seconds with messages being decoded.   Then it goes to IDLE.    The spectrum shows a strong signal and that it is tuned correctly however there are no messages being decoded most of the time. 

I am running a NooElec NESDR Smart (R820T) on both my RPI and my laptop, both systems show the same issue.

I have tried changing the P25-1 Decoder to use C4FM modulation instead of LSM but it seems to make no difference.  Traffic pool is set to 3.

waltdjr, 2018-07-27:
Experiencing the same issue.  3.4 beta 2 works almost flawlessly with decoding the P25-1.  However, new .4.0 alpha 1 decodes very little.  Voice is very broken when it does stops on a channel.

One difference I noticed between the two versions is on the tuners tab, under channels column, in version 3.4, the number bounces around.  In version 4.0, it shows 1 (Locked).

Using a RTL-SDR v3.

KR0SIV, 2018-07-30:
I'm also unable to decode the control channel at all with an rtl-sdr v3 on 4 while on 3.4 beta 2 it worked quite well.

masterplayer30, 2018-08-08:
0.4.0 Alpha 2 still won't decode the P25 control. Still on 0.3.4 as a result.

waltdjr, 2018-08-08:
Interesting, alpha 2 did fix my p25 issues.

infecticide, 2018-08-08:
I still have an issue decoding the control channel, however the issue now is that it works flawlessly for about 20 - 30 seconds and then it goes to IDLE.   If I close and re-open the app, when I start the channel, it sits at IDLE and the waterfall and spectrum don't move for about 15 seconds and then race to catch up.    It struggles after that, again once 20 - 30 seconds comes around it goes to IDLE and stays there.

DSheirer, 2018-08-10:
@infecticide what cpu/memory/os are you using?  This kinda sounds like a processing/performance issue.  Are you getting any 'Overflow' indicators?

DSheirer, 2018-08-10:
@masterplayer30 What type of P25 system, C4FM or LSM?  Do you know if it's a Motorola system, Harris, or other?

infecticide, 2018-08-10:
@DSheirer My laptop has a Quad Core i5 @ 2.67GHz, 8 GB RAM.

I use the following java command line to start the app:

java -XX:+UseG1GC -Xmx1024M -jar sdrtrunk_040_a2.jar

No Overflow indicators in the app or the command line console.

DSheirer, 2018-08-15:
@infecticide you might experiment with increasing the max memory you're giving to the jvm via your -Xmx flag.  By forcing the JVM to use 1 GB max, the JVM has to keep the garbage collection as a top priority.  Which tuner(s)/sample rate are you running with this configuration?

infecticide, 2018-08-16:
I have removed the memory limit.

The issue remains.

Tuner: R820T
ID: 00000001
Sample Rate: 2.400MHz
Freq: 139.12750MHz (Which is odd because the channel is configured for 139.245MHz)
Channels: 1 (Locked)
Current PPM: 0.0
Measured Error: Blank
Gain Master: 337

Let me know if you need anything else.

infecticide, 2018-08-16:
I've disabled PPM Autocorrect and its been chugged away happily for 10 mins before it went to IDLE and stayed there.

If I stop and start the channel, it starts decoding again briefly (2 mins) before going back to IDLE.

Measured Error shows -0.5PPM - 1.0PPM

If I re-enable that PPM autocorrect the issue occurs immediately (less than 10 seconds).

infecticide, 2018-08-16:
This issue seems to be related to it chasing calls and not tuning back to the control channel.

If I get a burst of calls (Traffic Pool is set to 3 / C4FM) 2 or more at once, it usually dies.

When no calls are coming in, it seems to decode the messages without issues.

DSheirer, 2018-08-17:
@infecticide Can you run the software until it gets into this state (control channel goes to idle) and then take two screenshots for me:

1. Spectral view with all of the currently enabled channels (control + traffic) in view, zooming in as much as possible while still keeping all of the channels in view.
2. Spectral view zoomed in on the control channel specifically

infecticide, 2018-08-18:
Channel zoom in:  https://i.imgur.com/lE0eqzU.png
Spectral View of Control Channel:  https://i.imgur.com/EGNewIx.png

DSheirer, 2018-08-19:
@infecticide Everything looks good with what you're doing.  I suspect the issue is in the code that I'm using to recombine channels from the polyphase channelizer.  The channelizer breaks up the entire spectrum into 12.5 kHz channels.  The system that you're monitoring has somewhat random channel spacing and that's causing sdrtrunk to select a tuner center frequency that places the control channel frequency between two 12.5 kHz channels.  The new channelizer will recombine those two channels and recenter to the channel frequency.  But, I think this may be causing the issue with the loss of sync on the decode.  I'll have to do some more testing to see if that's indeed the issue and see if I can correct it.

infecticide, 2018-08-25:
Would any additional logging on this side assist you in that?

DSheirer, 2018-08-30:
@infecticide I found the problem and have coded a fix for it.  I want to test it for a bit and then I'll merge the code into master and push out a new release shortly.

commit 96a56cab (2018-09-16): Resolves #375.

Reverts PLL gain values to version 0.3.4beta2 values to restore P25
decoder performance.

This change in conjuction with recent channel sample rate change should
restore P25 channel decoding to comparable levels from 0.3.4b2.