DSheirer/sdrtrunk #387: P25 Audio Recordings Limited to 10 seconds

DSheirer, 2018-08-08:
A user reports that recordings of P25 audio calls that exceed 10 seconds are being truncated at 10 seconds.

ImagoTrigger, 2018-10-07:
I can confirm and reproduce every time an automated county fire dispatch is called because it has a slow synthesized speech cadence with tones, grid box, description, etc. the call length is always >10 seconds and >158KB in .wav data.

160980 bytes seems to be a hard limit in v4.  v3 does not exhibit this behavior.

ImagoTrigger, 2018-10-08:
I've spent a good part of the day messing with this.  I can now get recordings up to 480948 bytes (again, give or take a few bytes I assume for INFO/RIFF stuff).  Attached are my changes, I don't know which one of the 3 made a difference, yet.  I'm hoping someone can figure it out besides me ;-)

[10sec_1.diff.txt](https://github.com/DSheirer/sdrtrunk/files/2454510/10sec_1.diff.txt)
[10sec_2.diff.txt](https://github.com/DSheirer/sdrtrunk/files/2454511/10sec_2.diff.txt)
[10sec_3.diff.txt](https://github.com/DSheirer/sdrtrunk/files/2454512/10sec_3.diff.txt)

DSheirer, 2018-10-08:
Thanks for finding this!  It's the resizing of the audio queue that did it.  Actually, I think the issue is that the reusable audio packet queue is not decrementing the total buffer count correctly when it takes buffers out of the queue and that's causing the overflow mechanism to kick in.

commit cb12d314 (2018-10-08): Merge pull request #436 from DSheirer/387-audio-10-minute-time-limit

Resolves #387 (P25) 10 second/10 minute audio recording time limit.
commit eb79d9e0 (2018-10-08): Resolves #387 10 second/10 minute audio recording time limit.

Updated audio packet wave recorder to correctly dequeue audio packets
to ensure that the overflowable buffer queue accounting process was
working correctly.

Updated overflowable buffer queue's poll() method to correctly decrement
the queue size counter.

Resolved issue where the mutable metadata was being prematurely reset
and causing the end audio packet to be mis-identified.  This resulted
in the recording manager not stopping the recording on command with the
timeout mechanism kicking in after a brief delay.
closed 2018-10-08 by DSheirer