DSheirer/sdrtrunk #403: Unable to see tuner RTL2832U

rayketcham, 2018-09-05:
E:\SDR>java -jar sdr-trunk-all-0.4.0-alpha.3.jar
17:12:05.851 INFO  i.g.dsheirer.gui.SDRTrunk - *******************************************************************  [5MB/15MB 34%]
17:12:05.855 INFO  i.g.dsheirer.gui.SDRTrunk - **** sdrtrunk: a trunked radio and digital decoding application ***  [5MB/15MB 34%]
17:12:05.855 INFO  i.g.dsheirer.gui.SDRTrunk - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [5MB/15MB 34%]
17:12:05.855 INFO  i.g.dsheirer.gui.SDRTrunk - *******************************************************************  [5MB/15MB 34%]
17:12:05.856 INFO  i.g.dsheirer.gui.SDRTrunk - Memory Logging Format: [Used/Allocated PercentUsed%]  [5MB/15MB 34%]
17:12:05.856 INFO  i.g.dsheirer.gui.SDRTrunk - Host CPU Cores:        8  [5MB/15MB 34%]
17:12:05.857 INFO  i.g.dsheirer.gui.SDRTrunk - Host OS Name:          Windows 10  [5MB/15MB 34%]
17:12:05.857 INFO  i.g.dsheirer.gui.SDRTrunk - Host OS Arch:          x86  [5MB/15MB 34%]
17:12:05.857 INFO  i.g.dsheirer.gui.SDRTrunk - Host OS Version:       10.0  [5MB/15MB 34%]
17:12:05.858 INFO  i.g.dsheirer.gui.SDRTrunk - Host Max Java Memory:  247 MB  [5MB/15MB 34%]
17:12:05.865 INFO  i.g.d.util.ThreadPool - Application thread pool created with [8] threads  [5MB/15MB 35%]
17:12:05.865 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: C:\Users\rayke\SDRTrunk  [5MB/15MB 35%]
17:12:05.868 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [C:\Users\rayke\SDRTrunk\SDRTrunk.properties]  [5MB/15MB 36%]
17:12:05.869 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [C:\Users\rayke\SDRTrunk\SDRTrunk.properties]  [5MB/15MB 36%]
17:12:05.881 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [C:\Users\rayke\SDRTrunk\settings\settings.xml]  [3MB/15MB 20%]
17:12:06.164 INFO  i.g.d.s.m.MixerManager - loading system mixer devices
        [LOADED]     Output: Primary Sound Driver CHANNELS: [MONO, STEREO]
        [LOADED]     Output: Headphones (2- High Definition Audio Device) CHANNELS: [MONO, STEREO]
        [LOADED]     Output: Speakers (Yeti Stereo Microphone) CHANNELS: [MONO, STEREO]
        [LOADED]     Output: Digital Audio (S/PDIF) (2- High Definition Audio Device) CHANNELS: [MONO, STEREO]
        [LOADED]     Output: Digital Audio (S/PDIF) (2- High Definition Audio Device) CHANNELS: [MONO, STEREO]
        [LOADED]     Input:  Primary Sound Capture Driver CHANNELS: [LEFT, RIGHT, MONO]
        [LOADED]     Input:  Microphone (Yeti Stereo Microph CHANNELS: [LEFT, RIGHT, MONO]
  [7MB/15MB 49%]
17:12:06.165 INFO  i.g.d.s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations  [7MB/15MB 49%]
17:12:06.257 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 16777474  [8MB/15MB 55%]
17:12:06.257 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.18.10866  [8MB/15MB 55%]
17:12:06.263 INFO  i.g.d.s.t.TunerManager - discovered [16] attached USB devices  [8MB/15MB 55%]
17:12:06.267 INFO  i.g.d.s.t.TunerManager - usb device [1532:0029] NOT LOADED: Unknown Device  [8MB/15MB 56%]
17:12:06.268 INFO  i.g.d.s.t.TunerManager - usb device [1B1C:0C04] NOT LOADED: Unknown Device  [8MB/15MB 56%]
17:12:06.272 ERROR i.g.d.s.t.TunerManager - couldn't determine RTL2832 tuner type  [9MB/15MB 58%]
io.github.dsheirer.source.SourceException: couldn't open device - check permissions Linux: udev.rule? , or Windows: reinstall Zadig? [LIBUSB_ERROR_NOT_SUPPORTED]
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.identifyTunerType(RTL2832TunerController.java:337)
        at io.github.dsheirer.source.tuner.TunerManager.initRTL2832Tuner(TunerManager.java:373)
        at io.github.dsheirer.source.tuner.TunerManager.initTuner(TunerManager.java:215)
        at io.github.dsheirer.source.tuner.TunerManager.initTuners(TunerManager.java:130)
        at io.github.dsheirer.source.tuner.TunerManager.<init>(TunerManager.java:66)
        at io.github.dsheirer.source.SourceManager.<init>(SourceManager.java:41)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:150)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:585)
17:12:06.272 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:2838] NOT LOADED: SDRTRunk doesn't currently support RTL2832 Dongle with [UNKNOWN] tuner for tuner class[USB Tuner:RTL2832_VARIOUS Vendor:RTL2832 Device:SDR Address:0BDA:2838]  [9MB/15MB 58%]
17:12:06.273 ERROR i.g.d.s.t.TunerManager - couldn't determine RTL2832 tuner type  [9MB/15MB 58%]
io.github.dsheirer.source.SourceException: couldn't open device - check permissions Linux: udev.rule? , or Windows: reinstall Zadig? [LIBUSB_ERROR_NOT_SUPPORTED]
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.identifyTunerType(RTL2832TunerController.java:337)
        at io.github.dsheirer.source.tuner.TunerManager.initRTL2832Tuner(TunerManager.java:373)
        at io.github.dsheirer.source.tuner.TunerManager.initTuner(TunerManager.java:215)
        at io.github.dsheirer.source.tuner.TunerManager.initTuners(TunerManager.java:130)
        at io.github.dsheirer.source.tuner.TunerManager.<init>(TunerManager.java:66)
        at io.github.dsheirer.source.SourceManager.<init>(SourceManager.java:41)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:150)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:585)
17:12:06.273 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:2838] NOT LOADED: SDRTRunk doesn't currently support RTL2832 Dongle with [UNKNOWN] tuner for tuner class[USB Tuner:RTL2832_VARIOUS Vendor:RTL2832 Device:SDR Address:0BDA:2838]  [9MB/15MB 58%]
17:12:06.274 INFO  i.g.d.s.t.TunerManager - usb device [1B1C:1B3D] NOT LOADED: Unknown Device  [9MB/15MB 58%]
17:12:06.274 INFO  i.g.d.s.t.TunerManager - usb device [B58E:9E84] NOT LOADED: Unknown Device  [9MB/15MB 58%]
17:12:06.275 ERROR i.g.d.s.t.r.RTL2832TunerController - error performing dummy write - attempting device reset  [9MB/15MB 59%]
org.usb4java.LibUsbException: USB error 2: error writing byte buffer: Invalid parameter
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.write(RTL2832TunerController.java:816)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.write(RTL2832TunerController.java:796)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.writeRegister(RTL2832TunerController.java:765)
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.identifyTunerType(RTL2832TunerController.java:351)
        at io.github.dsheirer.source.tuner.TunerManager.initRTL2832Tuner(TunerManager.java:373)
        at io.github.dsheirer.source.tuner.TunerManager.initTuner(TunerManager.java:215)
        at io.github.dsheirer.source.tuner.TunerManager.initTuners(TunerManager.java:130)
        at io.github.dsheirer.source.tuner.TunerManager.<init>(TunerManager.java:66)
        at io.github.dsheirer.source.SourceManager.<init>(SourceManager.java:41)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:150)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:585)
17:12:06.276 ERROR i.g.d.s.t.r.RTL2832TunerController - device reset attempted, but lost device handle.  Try restarting the application to use this device  [9MB/15MB 59%]
17:12:06.277 ERROR i.g.d.s.t.r.RTL2832TunerController - error while determining tuner type  [9MB/15MB 59%]
io.github.dsheirer.source.SourceException: couldn't reset device
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.identifyTunerType(RTL2832TunerController.java:392)
        at io.github.dsheirer.source.tuner.TunerManager.initRTL2832Tuner(TunerManager.java:373)
        at io.github.dsheirer.source.tuner.TunerManager.initTuner(TunerManager.java:215)
        at io.github.dsheirer.source.tuner.TunerManager.initTuners(TunerManager.java:130)
        at io.github.dsheirer.source.tuner.TunerManager.<init>(TunerManager.java:66)
        at io.github.dsheirer.source.SourceManager.<init>(SourceManager.java:41)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:150)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:585)
17:12:06.277 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:2838] NOT LOADED: SDRTRunk doesn't currently support RTL2832 Dongle with [UNKNOWN] tuner for tuner class[USB Tuner:RTL2832_VARIOUS Vendor:RTL2832 Device:SDR Address:0BDA:2838]  [9MB/15MB 59%]
17:12:06.278 INFO  i.g.d.s.t.TunerManager - usb device [8086:8C31] NOT LOADED: Unknown Device  [9MB/15MB 59%]
[... cut ...]
17:12:06.693 INFO  i.g.d.icon.IconManager - loading icons file [C:\Users\rayke\SDRTrunk\settings\icons.xml]  [10MB/15MB 65%]
17:12:06.840 INFO  i.g.d.p.PlaylistManager - Attempting to load version 2 playlist file [C:\Users\rayke\SDRTrunk\playlist\playlist_v2.xml]  [9MB/22MB 42%]
17:12:06.841 INFO  i.g.d.p.PlaylistManager - PlaylistManager - playlist not found at [C:\Users\rayke\SDRTrunk\playlist\playlist_v2.xml]  [9MB/22MB 42%]
17:12:06.842 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [9MB/22MB 42%]

rayketcham, 2018-09-05:
I installed a libusb driver and still getting a couple of errors in the startup.  I can now see the tuner inside the app.

E:\SDR>java -jar sdr-trunk-all-0.4.0-alpha.3.jar
18:27:19.512 INFO  i.g.dsheirer.gui.SDRTrunk - *******************************************************************  [15MB/491MB 3%]
18:27:19.516 INFO  i.g.dsheirer.gui.SDRTrunk - **** sdrtrunk: a trunked radio and digital decoding application ***  [15MB/491MB 3%]
18:27:19.517 INFO  i.g.dsheirer.gui.SDRTrunk - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [15MB/491MB 3%]
18:27:19.518 INFO  i.g.dsheirer.gui.SDRTrunk - *******************************************************************  [15MB/491MB 3%]
18:27:19.518 INFO  i.g.dsheirer.gui.SDRTrunk - Memory Logging Format: [Used/Allocated PercentUsed%]  [15MB/491MB 3%]
18:27:19.518 INFO  i.g.dsheirer.gui.SDRTrunk - Host CPU Cores:        8  [15MB/491MB 3%]
18:27:19.518 INFO  i.g.dsheirer.gui.SDRTrunk - Host OS Name:          Windows 10  [15MB/491MB 3%]
18:27:19.519 INFO  i.g.dsheirer.gui.SDRTrunk - Host OS Arch:          amd64  [15MB/491MB 3%]
18:27:19.519 INFO  i.g.dsheirer.gui.SDRTrunk - Host OS Version:       10.0  [15MB/491MB 3%]
18:27:19.519 INFO  i.g.dsheirer.gui.SDRTrunk - Host Max Java Memory:  7 GB  [15MB/491MB 3%]
18:27:19.526 INFO  i.g.d.util.ThreadPool - Application thread pool created with [8] threads  [15MB/491MB 3%]
18:27:19.527 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: C:\Users\rayke\SDRTrunk  [15MB/491MB 3%]
18:27:19.531 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [C:\Users\rayke\SDRTrunk\SDRTrunk.properties]  [15MB/491MB 3%]
18:27:19.531 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [C:\Users\rayke\SDRTrunk\SDRTrunk.properties]  [15MB/491MB 3%]
18:27:19.539 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [C:\Users\rayke\SDRTrunk\settings\settings.xml]  [15MB/491MB 3%]
18:27:19.820 INFO  i.g.d.s.m.MixerManager - loading system mixer devices
        [LOADED]     Output: Primary Sound Driver CHANNELS: [MONO, STEREO]
        [LOADED]     Output: Headphones (2- High Definition Audio Device) CHANNELS: [MONO, STEREO]
        [LOADED]     Output: Speakers (Yeti Stereo Microphone) CHANNELS: [MONO, STEREO]
        [LOADED]     Output: Digital Audio (S/PDIF) (2- High Definition Audio Device) CHANNELS: [MONO, STEREO]
        [LOADED]     Output: Digital Audio (S/PDIF) (2- High Definition Audio Device) CHANNELS: [MONO, STEREO]
        [LOADED]     Input:  Primary Sound Capture Driver CHANNELS: [LEFT, RIGHT, MONO]
        [LOADED]     Input:  Microphone (Yeti Stereo Microph CHANNELS: [LEFT, RIGHT, MONO]
  [38MB/491MB 7%]
18:27:19.822 INFO  i.g.d.s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations  [38MB/491MB 7%]
18:27:19.895 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 16777474  [41MB/491MB 8%]
18:27:19.896 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.18.10866  [41MB/491MB 8%]
18:27:19.901 INFO  i.g.d.s.t.TunerManager - discovered [16] attached USB devices  [41MB/491MB 8%]
18:27:19.905 INFO  i.g.d.s.t.TunerManager - usb device [1532:0029] NOT LOADED: Unknown Device  [41MB/491MB 8%]
18:27:19.905 INFO  i.g.d.s.t.TunerManager - usb device [1B1C:0C04] NOT LOADED: Unknown Device  [41MB/491MB 8%]
18:27:19.909 ERROR i.g.d.s.t.TunerManager - couldn't determine RTL2832 tuner type  [41MB/491MB 8%]
io.github.dsheirer.source.SourceException: couldn't open device - check permissions Linux: udev.rule? , or Windows: reinstall Zadig? [LIBUSB_ERROR_NOT_SUPPORTED]
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.identifyTunerType(RTL2832TunerController.java:337)
        at io.github.dsheirer.source.tuner.TunerManager.initRTL2832Tuner(TunerManager.java:373)
        at io.github.dsheirer.source.tuner.TunerManager.initTuner(TunerManager.java:215)
        at io.github.dsheirer.source.tuner.TunerManager.initTuners(TunerManager.java:130)
        at io.github.dsheirer.source.tuner.TunerManager.<init>(TunerManager.java:66)
        at io.github.dsheirer.source.SourceManager.<init>(SourceManager.java:41)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:150)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:585)
18:27:19.910 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:2838] NOT LOADED: SDRTRunk doesn't currently support RTL2832 Dongle with [UNKNOWN] tuner for tuner class[USB Tuner:RTL2832_VARIOUS Vendor:RTL2832 Device:SDR Address:0BDA:2838]  [41MB/491MB 8%]
18:27:19.910 ERROR i.g.d.s.t.TunerManager - couldn't determine RTL2832 tuner type  [41MB/491MB 8%]
io.github.dsheirer.source.SourceException: couldn't open device - check permissions Linux: udev.rule? , or Windows: reinstall Zadig? [LIBUSB_ERROR_NOT_SUPPORTED]
        at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.identifyTunerType(RTL2832TunerController.java:337)
        at io.github.dsheirer.source.tuner.TunerManager.initRTL2832Tuner(TunerManager.java:373)
        at io.github.dsheirer.source.tuner.TunerManager.initTuner(TunerManager.java:215)
        at io.github.dsheirer.source.tuner.TunerManager.initTuners(TunerManager.java:130)
        at io.github.dsheirer.source.tuner.TunerManager.<init>(TunerManager.java:66)
        at io.github.dsheirer.source.SourceManager.<init>(SourceManager.java:41)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:150)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:585)
[... cut ...]
18:27:21.953 INFO  i.g.d.icon.IconManager - loading icons file [C:\Users\rayke\SDRTrunk\settings\icons.xml]  [49MB/288MB 17%]
18:27:22.077 INFO  i.g.d.p.PlaylistManager - Attempting to load version 2 playlist file [C:\Users\rayke\SDRTrunk\playlist\playlist_v2.xml]  [59MB/288MB 20%]
18:27:22.077 INFO  i.g.d.p.PlaylistManager - PlaylistManager - playlist not found at [C:\Users\rayke\SDRTrunk\playlist\playlist_v2.xml]  [59MB/288MB 20%]
18:27:22.078 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [59MB/288MB 20%]

DSheirer, 2018-09-07:
@rayketcham it looks like one of your RTL2832 dongles is using an unsupported tuner type.  The one with the R820T tuner appears to have loaded correctly, but the other one did not.

pgram1, 2020-07-07:
Similar situation here on Arch Linux on latest AUR build

```
02:59:25.088 INFO  i.g.d.log.ApplicationLog - Application Log File: /root/SDRTrunk/logs/sdrtrunk_app.log  [9MB/60MB 16%]
02:59:25.132 INFO  i.g.d.log.ApplicationLog -   [10MB/60MB 18%]
02:59:25.132 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [10MB/60MB 18%]
02:59:25.133 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [10MB/60MB 18%]
02:59:25.133 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [10MB/60MB 18%]
02:59:25.133 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [10MB/60MB 18%]
02:59:25.134 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [10MB/60MB 18%]
02:59:25.134 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Linux  [10MB/60MB 18%]
02:59:25.134 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [10MB/60MB 18%]
02:59:25.135 INFO  i.g.d.log.ApplicationLog - Host OS Version:       5.7.0-3-MANJARO  [10MB/60MB 18%]
02:59:25.135 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        4  [10MB/60MB 18%]
02:59:25.136 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  934 MB  [10MB/60MB 18%]
02:59:25.136 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [10MB/60MB 18%]
02:59:25.137 INFO  i.g.d.log.ApplicationLog -  Application Root: /root/SDRTrunk  [10MB/60MB 18%]
02:59:25.137 INFO  i.g.d.log.ApplicationLog -  Application Log:  /root/SDRTrunk/logs  [10MB/60MB 18%]
02:59:25.138 INFO  i.g.d.log.ApplicationLog -  Event Log:        /root/SDRTrunk/event_logs  [10MB/60MB 18%]
02:59:25.138 INFO  i.g.d.log.ApplicationLog -  Playlist:         /root/SDRTrunk/playlist  [10MB/60MB 18%]
02:59:25.139 INFO  i.g.d.log.ApplicationLog -  Recordings:       /root/SDRTrunk/recordings  [10MB/60MB 18%]
02:59:25.515 INFO  i.g.d.util.ThreadPool - Application thread pool created with [4] threads  [17MB/60MB 28%]
02:59:25.516 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: /root/SDRTrunk  [17MB/60MB 28%]
02:59:25.521 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/root/SDRTrunk/SDRTrunk.properties]  [17MB/60MB 28%]
02:59:25.522 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/root/SDRTrunk/SDRTrunk.properties]  [17MB/60MB 28%]
02:59:25.770 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [/root/SDRTrunk/settings/settings.xml]  [8MB/27MB 30%]
02:59:26.159 INFO  i.g.d.s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations  [12MB/27MB 47%]
02:59:26.177 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 16777478  [13MB/27MB 50%]
02:59:26.179 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.22.11312  [13MB/27MB 50%]
02:59:26.179 INFO  i.g.d.s.t.TunerManager - discovered [6] attached USB devices  [13MB/27MB 50%]
02:59:26.186 INFO  i.g.d.s.t.TunerManager - USB Bus [2] Device [2A45:0C01] Unknown Device - Class 0  [13MB/27MB 50%]
02:59:26.188 INFO  i.g.d.s.t.TunerManager - USB Bus [2] Device [1D19:1101] Unknown Device - Class 0  [13MB/27MB 50%]
02:59:26.190 INFO  i.g.d.s.t.TunerManager - USB Bus [2] Device [8087:0020] Hub Device  [13MB/27MB 50%]
02:59:26.190 INFO  i.g.d.s.t.TunerManager - USB Bus [2] Device [1D6B:0002] Hub Device  [13MB/27MB 50%]
02:59:26.191 INFO  i.g.d.s.t.TunerManager - USB Bus [1] Device [8087:0020] Hub Device  [13MB/27MB 50%]
02:59:26.191 INFO  i.g.d.s.t.TunerManager - USB Bus [1] Device [1D6B:0002] Hub Device  [13MB/27MB 50%]
02:59:26.192 INFO  i.g.d.s.t.TunerManager - -------------------------------------------------------------  [13MB/27MB 50%]
02:59:26.192 INFO  i.g.d.s.t.TunerManager - USB Bus - Potential Maximum Data Rates  [13MB/27MB 51%]
02:59:26.315 ERROR i.g.d.a.p.AudioOutput - Couldn't obtain audio source data line for audio output - mixer [MID [default]]  [15MB/27MB 58%]
02:59:27.393 INFO  i.g.d.icon.IconManager - loading icons file [/root/SDRTrunk/settings/icons.xml]  [10MB/33MB 32%]
02:59:27.759 INFO  i.g.d.p.PlaylistManager - Loading playlist file [/root/SDRTrunk/playlist/default.xml]  [20MB/33MB 60%]
02:59:27.844 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [21MB/33MB 65%]
02:59:39.139 INFO  i.g.dsheirer.gui.SDRTrunk - Application shutdown started ...  [29MB/45MB 66%]
02:59:39.144 INFO  i.g.dsheirer.gui.SDRTrunk - Stopping channels ...  [29MB/45MB 66%]
02:59:39.145 INFO  i.g.dsheirer.gui.SDRTrunk - Stopping spectral display ...  [29MB/45MB 66%]
02:59:39.156 INFO  i.g.d.s.t.TunerManager - Releasing [0] tuner(s) ...  [29MB/45MB 66%]
02:59:39.157 INFO  i.g.d.s.t.TunerManager - Shutting down LibUsb timeout processor ...  [29MB/45MB 66%]
02:59:39.157 INFO  i.g.d.s.t.TunerManager - Shutting down LibUsb ...  [29MB/45MB 66%]
02:59:39.158 INFO  i.g.dsheirer.gui.SDRTrunk - Shutdown complete.  [29MB/45MB 66%]
02:59:39.158 INFO  i.g.d.log.ApplicationLog - Stopping application logging  [29MB/45MB 66%]
```

my device is listed in dmesg as

```
[   42.326355] usb 2-1.1: New USB device found, idVendor=1d19, idProduct=1101, bcdDevice= 1.00
[   42.326358] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   42.326360] usb 2-1.1: Product: Rtl2832UDVB
[   42.326361] usb 2-1.1: Manufacturer: Realtek
```

and after the rtl_test i get

```
Found 1 device(s):
  0:  Realtek, Rtl2832UDVB, SN: 0

Using device 0: Dexatek DK DVB-T Dongle (Logilink VG0002A)
Found FCI 2580 tuner
Supported gain values (1): 0.0 
Sampling at 2048000 S/s.
```

device is also usable in gqrx

DSheirer, 2020-07-07:
@pgram1 The FCI 2580 tuner is not currently supported by sdrtrunk.

closed 2020-07-07 by DSheirer