DSheirer/sdrtrunk #423: LTR and LTR-Net Decoders Perform Poorly

DSheirer, 2018-09-16:
After updating the LTR decoder to use variable sample rates, it appears that the logic to filter out the DC offset is causing the channels to decode poorly or not at all.  In one case, I was trying to decode 151.865 and the stronger HSC on 151.49 was decoding.  This indicates that the filter is not attenuating adjacent channels sufficiently.  

And, the decoder is not locking on to the LTR idle bursts in time to decode them properly.

Clark985, 2023-03-17:
Not sure if this is the proper way to comment, but I'm also having LTR decode issues.  

The first transmission will decode, then it's as if it gets a phantom transmission that doesn't catch the end of call signaling.  Activity will often times show a multi-hour transmission.   Running 0.5.2 (Windows), also did this on 0.5.0

Idle bursts also aren't decoded every 10 seconds, maybe a third to a half of them decode.

DSheirer, 2023-06-05:
Original issue & Issue reported by @Clark985 resolved in PR #1567

closed 2023-06-05 by DSheirer