DSheirer/sdrtrunk #426: Screen Layout Issue - Buttons Clipped on Mac OS X

DCHuber, 2018-09-17:
Running v0.3.4-beta.2 on a Mac OS X v10.11.6 with java version 1.8.0_181

Various screen layouts have the buttons clipped as shown in the attached screen captures.

[image]

[image]

[image]

ZacharyDuBois, 2020-03-17:
Happens with the 0.4.0b5 too.

DSheirer, 2020-03-18:
This seems like a GUI layout manager issue.  For the Swing based GUI windows I used MigLayout. 

These GUI windows are all being redone with JavaFX GUI components, so hopefully the layout manager performs better with JavaFX on Mac OSX