DSheirer/sdrtrunk #439: Audio Streaming Metadata Updata Issues

DSheirer, 2018-10-26:
I have 4 feeds to Broadcastify and 1 to Shoutcast.  On the Broadcastify feeds the tags appear to be trying to display but seem to "hang" up on on a talk group then completely stop updating.

On the Shoutcast feed i will see "scanning" like i would expect but as the talkgroups change they do not seem to keep up.

 I have checked Broadcastify with various apps on both android and iphone.  I have checked Shoutcast with only VLC on both android and iphone.

The broadcastify feeds are displaying "TO:"Talkgroup" FROM: "RadioID"  but the Shoutcast feed displays the name of the talkgroup.  I am not sure if there is anything to config for this.

FrostTheFox, 2019-08-08:
Just dropping a +1 here. With an Icecast2 server, on the client I get the stream description for a little while. Eventually it switches over to one of the calls (ex: TO:10881 LaGrange Co. Sheriff Dispatch FROM:4410134) and never leaves it, even during silence.

KalenXI, 2019-08-20:
I'm seeing this issue as well. And the log files are filled with this error over and over:

```
20190820 184818.031 [sdrtrunk pool-1-thread-1] DEBUG o.a.mina.http.HttpClientCodec - encode org.apache.mina.http.HttpRequestImpl  %PARSER_ERROR[memory_usage]
20190820 184818.031 [sdrtrunk pool-1-thread-1] DEBUG o.a.mina.http.HttpClientCodec - HttpRequest  %PARSER_ERROR[memory_usage]
```

DSheirer, 2019-12-10:
Reported by a user:
-----------------------------
I am experiencing metadata updating issues with beta 1. Below are two snippets from the Shoutcast server metadata XML.

<SHOUTCASTSERVER>
<SONGMETADATA>
<TALB>Onondaga County Public Safety</TALB>
<TENC>sdrtrunk v0.4.0-beta.1</TENC>
<TCON>Scanner</TCON>
TO:48 OCSO MRD 4 - West Zone Dispatch FROM:911030 Onondaga County 911 - MRD West (County Channel 4)
</SONGMETADATA>
</SHOUTCASTSERVER>

In the above example, the title is not wrapped with <TIT2></TIT2> tags.

In the example below, the "Scanning ..." title appears correct:

<SHOUTCASTSERVER>
<SONGMETADATA>
<TALB>Onondaga County Public Safety</TALB>
<TENC>sdrtrunk v0.4.0-beta.1</TENC>
<TCON>Scanner</TCON>
<TIT2>Scanning ...</TIT2>
</SONGMETADATA>
</SHOUTCASTSERVER>

PR #615 (2019-12-15): 439 - Streaming Metadata Update Issues
Resolves #439 Icecast and Shoutcast metadata updating issue.  

Resolves issue with Icecast HTTP metadata incorrect XML tags.
Resolves issue with intermittent Broadcastify/Icecast Metadata updates.
closed 2019-12-15 by DSheirer