DSheirer/sdrtrunk #441: Audio Recording Stops After Encountering Error

DSheirer, 2018-10-27:
20181026 064858.641 [sdrtrunk pool-1-thread-7] ERROR i.g.d.audio.AudioManager - Encountered error while processing audio packets  [600MB/1GB 58%]
java.lang.IllegalStateException: User count is below zero.  This indicates that this buffer's decrement user count was invoked by more than the expected user count
    at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.recycle(AbstractReusableBuffer.java:118)
    at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.decrementUserCount(AbstractReusableBuffer.java:100)
    at io.github.dsheirer.audio.AudioManager$AudioPacketProcessor.run(AudioManager.java:445)
    at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
    at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
    at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
    at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
    at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
    at java.base/java.lang.Thread.run(Unknown Source)

ImagoTrigger, 2018-10-30:
another one

20181030 074230.058 [sdrtrunk pool-1-thread-5] ERROR i.g.d.audio.AudioManager - Encountered error while processing audio packets  [627MB/1GB 61%]
java.lang.IllegalStateException: User count is below zero.  This indicates that this buffer's decrement user count was invoked by more than the expected user count
	at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.recycle(AbstractReusableBuffer.java:118)
	at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.decrementUserCount(AbstractReusableBuffer.java:100)
	at io.github.dsheirer.audio.AudioManager$AudioPacketProcessor.run(AudioManager.java:445)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
20181030 074230.908 [sdrtrunk pool-1-thread-3] ERROR i.g.d.d.f.c.o.ChannelOutputProcessor - Error while processing polyphase channel samples  [677MB/1GB 66%]
java.lang.IllegalStateException: User count is below zero.  This indicates that this buffer's decrement user count was invoked by more than the expected user count
	at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.recycle(AbstractReusableBuffer.java:118)
	at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.decrementUserCount(AbstractReusableBuffer.java:100)
	at io.github.dsheirer.record.RecorderManager.receive(RecorderManager.java:108)
	at io.github.dsheirer.record.RecorderManager.receive(RecorderManager.java:40)
	at io.github.dsheirer.sample.buffer.ReusableBufferBroadcaster.broadcast(ReusableBufferBroadcaster.java:42)
	at io.github.dsheirer.sample.buffer.ReusableBufferBroadcaster.broadcast(ReusableBufferBroadcaster.java:23)
	at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:39)
	at io.github.dsheirer.module.decode.p25.audio.P25AudioModule.processAudio(P25AudioModule.java:166)
	at io.github.dsheirer.module.decode.p25.audio.P25AudioModule.receive(P25AudioModule.java:115)
	at io.github.dsheirer.module.decode.p25.audio.P25AudioModule.receive(P25AudioModule.java:39)
	at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:121)
	at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:39)
	at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:102)
	at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:95)
	at io.github.dsheirer.module.decode.p25.P25MessageProcessor.receive(P25MessageProcessor.java:90)
	at io.github.dsheirer.module.decode.p25.P25MessageProcessor.receive(P25MessageProcessor.java:29)
	at io.github.dsheirer.module.decode.p25.P25MessageFramer.dispatch(P25MessageFramer.java:174)
	at io.github.dsheirer.module.decode.p25.P25MessageFramer.access$300(P25MessageFramer.java:38)
	at io.github.dsheirer.module.decode.p25.P25MessageFramer$P25MessageAssembler.checkComplete(P25MessageFramer.java:317)
	at io.github.dsheirer.module.decode.p25.P25MessageFramer$P25MessageAssembler.receive(P25MessageFramer.java:247)
	at io.github.dsheirer.module.decode.p25.P25MessageFramer.receive(P25MessageFramer.java:183)
	at io.github.dsheirer.module.decode.p25.P25MessageFramer.receive(P25MessageFramer.java:38)
	at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:121)
	at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:39)
	at io.github.dsheirer.dsp.psk.PSKDemodulator.broadcast(PSKDemodulator.java:58)
	at io.github.dsheirer.dsp.psk.DQPSKGardnerDemodulator.calculateSymbol(DQPSKGardnerDemodulator.java:88)
	at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:115)
	at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:89)
	at io.github.dsheirer.module.decode.p25.P25DecoderLSM.receive(P25DecoderLSM.java:104)
	at io.github.dsheirer.module.decode.p25.P25DecoderLSM.receive(P25DecoderLSM.java:33)
	at io.github.dsheirer.sample.buffer.ReusableBufferBroadcaster.broadcast(ReusableBufferBroadcaster.java:42)
	at io.github.dsheirer.sample.buffer.ReusableBufferBroadcaster.broadcast(ReusableBufferBroadcaster.java:23)
	at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:39)
	at io.github.dsheirer.sample.buffer.ReusableComplexBufferAssembler.flush(ReusableComplexBufferAssembler.java:153)
	at io.github.dsheirer.sample.buffer.ReusableComplexBufferAssembler.receive(ReusableComplexBufferAssembler.java:127)
	at io.github.dsheirer.sample.buffer.ReusableComplexBufferAssembler.receive(ReusableComplexBufferAssembler.java:139)
	at io.github.dsheirer.dsp.filter.channelizer.output.TwoChannelOutputProcessor.process(TwoChannelOutputProcessor.java:117)
	at io.github.dsheirer.dsp.filter.channelizer.output.ChannelOutputProcessor.processChannelResults(ChannelOutputProcessor.java:125)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.processSamples(PolyphaseChannelSource.java:246)
	at io.github.dsheirer.source.tuner.channel.TunerChannelSource$ScheduledIntervalProcessor.run(TunerChannelSource.java:303)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
20181030 074234.606 [sdrtrunk pool-1-thread-8] WARN  i.g.d.record.RecorderManager - overflow - audio packets will be dropped until recording catches up  [270MB/1GB 26%]

ImagoTrigger, 2018-11-03:
got this one

20181103 055334.555 [sdrtrunk pool-1-thread-6] WARN  i.g.d.record.RecorderManager - overflow - audio packets will be dropped until recording catches up  [670MB/1GB 65%]

w/o any other errors/warnings and recording stops until the app is restarted again.

ImagoTrigger, 2018-11-04:
same thing again, system is not even that busy during the weekend mornings which is when i like to work on these things ;-)

closed 2018-12-24 by DSheirer