DSheirer/sdrtrunk #461: Consolidate Audio Streaming/Recording

DSheirer, 2018-12-27:
There is a potential for some of the audio packets produced by a P25 call to not contain all of the identifiers for the call until a later point in the call.  This causes the initial audio packets to not be recorded.  

Update the audio packet manager to record all audio calls by default to a temporary recording file.  On completion of the recording, evaluate the identifier collection:

1. If the identifier collection indicates that any of the identifiers are recordable, convert the .tmp file to a wav recording.

2. If the identifier collection indicates that any of the identifiers are streamable, convert the .tmp file to a streamable .mp3 recording.

davidmckenzie, 2019-11-15:
Hoping to see this one at some point 🙏 has been brutal on my conventional channels this fire season. Every call has the 'missing start' problem, and often see completely missing calls. Trunked channels are perfect though

DSheirer, 2020-03-19:
This issue is resolved with MR #675.

closed 2020-03-19 by DSheirer