DSheirer/sdrtrunk #487: [feature request] Portable install

Tom-Neverwinter, 2019-01-16:
can it install neatly onto the desktop. with configuration.

DSheirer, 2019-01-17:
There's really no installation required, other than the sdr drivers required for windows and the udev permissions for linux.  

You can ask questions like this in the google support group.  

If there's still an issue to document, please add to some more details, or you can close this issue.

Tom-Neverwinter, 2019-01-17:
is it possible to export the config for stations? I have been having issues with skywave Linux and the cursed pulse audio bug. I have been manually entering the stations in and the respective alias's (hopefully skywave linux will receive an update as well as your wonderful software.)

Tom-Neverwinter, 2019-01-17:
That will work, I will simply write up a bash file for making the transfer easier. Thank you!!! I love your software!

closed 2019-01-18 by DSheirer