DSheirer/sdrtrunk #502: Icecast HTTP Broadcaster Error

DSheirer, 2019-01-22:
Im unable to connect to an icecast server.  With the same settings that I've always used it reports "disconnected" as the status.  It seems to be a problem with the parsing of the http header, the error is below.  In the streaming settings if I change the server address to include "http://" before my ip address the behavior changes and the status stays at "connecting" rather than going to "disconnected" but it never connects.

Relevant log:

14:54:26.580 ERROR i.g.d.a.b.i.IcecastHTTPAudioBroadcaster - HTTP protocol decoder error  [313MB/496MB 63%]
org.apache.mina.filter.codec.ProtocolDecoderException: java.lang.ArrayIndexOutOfBoundsException: Index 1 out of bounds for length 1 (Hexdump: 48 54 54 50 2F 31 2E 30 20 32 30 30 20 4F 4B 0D 0A 0D 0A)
    at org.apache.mina.filter.codec.ProtocolCodecFilter.messageReceived(ProtocolCodecFilter.java:262)
    at org.apache.mina.core.filterchain.DefaultIoFilterChain.callNextMessageReceived(DefaultIoFilterChain.java:641)
    at org.apache.mina.core.filterchain.DefaultIoFilterChain.access$1300(DefaultIoFilterChain.java:48)
    at org.apache.mina.core.filterchain.DefaultIoFilterChain$EntryImpl$1.messageReceived(DefaultIoFilterChain.java:1114)
    at org.apache.mina.core.filterchain.IoFilterAdapter.messageReceived(IoFilterAdapter.java:121)
    at org.apache.mina.core.filterchain.DefaultIoFilterChain.callNextMessageReceived(DefaultIoFilterChain.java:641)
    at org.apache.mina.core.filterchain.DefaultIoFilterChain.fireMessageReceived(DefaultIoFilterChain.java:634)
    at org.apache.mina.core.polling.AbstractPollingIoProcessor.read(AbstractPollingIoProcessor.java:539)
    at org.apache.mina.core.polling.AbstractPollingIoProcessor.access$1200(AbstractPollingIoProcessor.java:68)
    at org.apache.mina.core.polling.AbstractPollingIoProcessor$Processor.process(AbstractPollingIoProcessor.java:1242)
    at org.apache.mina.core.polling.AbstractPollingIoProcessor$Processor.process(AbstractPollingIoProcessor.java:1231)
    at org.apache.mina.core.polling.AbstractPollingIoProcessor$Processor.run(AbstractPollingIoProcessor.java:683)
    at org.apache.mina.util.NamePreservingRunnable.run(NamePreservingRunnable.java:64)
    at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
    at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
    at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.lang.ArrayIndexOutOfBoundsException: Index 1 out of bounds for length 1
    at org.apache.mina.http.HttpClientDecoder.parseHttpReponseHead(HttpClientDecoder.java:221)
    at org.apache.mina.http.HttpClientDecoder.decode(HttpClientDecoder.java:108)
    at org.apache.mina.filter.codec.ProtocolCodecFilter.messageReceived(ProtocolCodecFilter.java:253)
    ... 15 common frames omitted
14:54:26.581 INFO  i.g.d.a.b.AudioBroadcaster - [Dane County] status: Disconnected  [313MB/496MB 63%]
14:54:57.580 ERROR i.g.d.a.b.i.IcecastHTTPAudioBroadcaster - HTTP protocol decoder error  [109MB/496MB 22%]
org.apache.mina.filter.codec.ProtocolDecoderException: java.lang.ArrayIndexOutOfBoundsException: Index 1 out of bounds for length 1 (Hexdump: 48 54 54 50 2F 31 2E 30 20 32 30 30 20 4F 4B 0D 0A 0D 0A)
    at org.apache.mina.filter.codec.ProtocolCodecFilter.messageReceived(ProtocolCodecFilter.java:262)
    at org.apache.mina.core.filterchain.DefaultIoFilterChain.callNextMessageReceived(DefaultIoFilterChain.java:641)
    at org.apache.mina.core.filterchain.DefaultIoFilterChain.access$1300(DefaultIoFilterChain.java:48)
    at org.apache.mina.core.filterchain.DefaultIoFilterChain$EntryImpl$1.messageReceived(DefaultIoFilterChain.java:1114)
    at org.apache.mina.core.filterchain.IoFilterAdapter.messageReceived(IoFilterAdapter.java:121)
    at org.apache.mina.core.filterchain.DefaultIoFilterChain.callNextMessageReceived(DefaultIoFilterChain.java:641)
    at org.apache.mina.core.filterchain.DefaultIoFilterChain.fireMessageReceived(DefaultIoFilterChain.java:634)
    at org.apache.mina.core.polling.AbstractPollingIoProcessor.read(AbstractPollingIoProcessor.java:539)
    at org.apache.mina.core.polling.AbstractPollingIoProcessor.access$1200(AbstractPollingIoProcessor.java:68)
    at org.apache.mina.core.polling.AbstractPollingIoProcessor$Processor.process(AbstractPollingIoProcessor.java:1242)
    at org.apache.mina.core.polling.AbstractPollingIoProcessor$Processor.process(AbstractPollingIoProcessor.java:1231)
    at org.apache.mina.core.polling.AbstractPollingIoProcessor$Processor.run(AbstractPollingIoProcessor.java:683)
    at org.apache.mina.util.NamePreservingRunnable.run(NamePreservingRunnable.java:64)
    at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
    at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
    at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.lang.ArrayIndexOutOfBoundsException: Index 1 out of bounds for length 1
    at org.apache.mina.http.HttpClientDecoder.parseHttpReponseHead(HttpClientDecoder.java:221)
    at org.apache.mina.http.HttpClientDecoder.decode(HttpClientDecoder.java:108)
    at org.apache.mina.filter.codec.ProtocolCodecFilter.messageReceived(ProtocolCodecFilter.java:253)
    ... 15 common frames omitted
14:54:57.581 INFO  i.g.d.a.b.AudioBroadcaster - [Dane County] status: Disconnected  [109MB/496MB 22%]

hans109h, 2019-01-22:
I had a little bit more time to investigate, but didn't see big aha's in the logs--in case these are helpful:

Icecast debug log from v4alpha6 (working): http://paste.ubuntu.com/p/j9vZsS9zYF/

Icecast debug log from v4alpha7 (nonworking): http://paste.ubuntu.com/p/6fqBRp28S7/

From what I can tell it seems to mount but then just doesn't send any data.

DSheirer, 2019-01-24:
@hans109h Which version of icecast server are you using?

I'll start digging into this on Saturday or Sunday.

hans109h, 2019-01-24:
V2.4.4, the latest release.

PR #518 (2019-01-27): 502 icecast http connect error
Resolves #502.

Adds new Hack to account for Icecast HTTP servers sending incomplete HTTP responses that are missing the header tokens. 

Updated Apache Mina exception handler to inspect the response bytes and update the streaming connection state to connected when we receive the HTTP/1.0 OK response.
commit c4bf0896 (2019-01-27): 502 icecast http connect error (#518)

* Resolves #502

Resolves issue with connecting to icecast http streaming server.

The issue stems from Icecast HTTP (2.4+) servers not sending a full
HTTP OK response (missing header tokens) to a connection request.

Prior to the upgrade to apache mina 2.0.19, we received a specific
exception and handled it and set the status to connected.

After the upgrade, apache mina is now sending an array index out of
bounds exception when it decodes the incomplete http response.

So, we now catch that specific exception, inspect the http response
hex bytes and set [... cut]
closed 2019-01-27 by DSheirer