DSheirer/sdrtrunk #505: Can't run or build Version 0.4.0 Alpha 7 on RPi 3B+

nstoik, 2019-01-24:
As the title says, I am not able to run 0.4.0 Alpha 7 or 6 on a RPi 3B+, using the latest Raspbian Stretch.  

I am able to run 0.4.0 Alpha 5 via the .jar file on the release page, but I am interested in trying the latest build available.

When trying to launch the precompiled sdr-trunk file downloaded and unzipped from the release page, I issue the following command `./sdr-trunk`

I then get the following error 
`./sdr-trunk: 154: exec: /home/pi/Downloads/sdr-trunk-linux-x64/bin/java: Exec format error`

I thought it may be becasue its 64 bit, so I tried to download the source code and then build with Gradle.

I downloaded the source file zip, unpacked it and then ran `./gradlew build` and got an error similiar to "the JavaFXPlugin has been compiled by a more recent version of the Java Runtime". The Gradle build then fails.

Is it possible to run the latest alpha build on a RPi?

DSheirer, 2019-01-25:
The linux release uses an embeded Java JDK intended for intel/amd 64-bit CPU's.  I'll have to update the build script to produce a build with a JDK that's compatible with the ARM processor.  I'll add that this weekend, but I don't have a 3B+ to test against.  Once I add support for ARM, if you could test the build and let me know how it works?

nstoik, 2019-01-28:
Yes I can definitely test it out on a 3B+! Let me know when it is ready to test.

DSheirer, 2019-02-02:
@nstoik I added support for linux aarch64 (rpi) platform builds.  Can you please test the following build:

https://drive.google.com/open?id=1oGXJb_dA1S_kkYO5yTTu7JdmkKmSe3PS

Specifically, can you test:
1) You can launch the application using the sdr-trunk launch script in the bin directory
2) Once launched, you can open the user preferences dialog (View > Preferences)

Thanks in advance,
Denny

nstoik, 2019-02-05:
I downloaded the .zip from the link you provided onto the same RPi 3 that is working with the Alpha 5 build

I then extracted the zip file and ran the sdr-trunk script in the bin directory, as follows:
`pi@raspberrypi:~/Downloads/sdr-trunk-linux-aarch64/bin $ ./sdr-trunk`

I got the following error:
`./sdr-trunk: 154: exec: /home/pi/Downloads/sdr-trunk-linux-aarch64/bin/java: Exec format error`

Let me know if there's something else I should try

DSheirer, 2019-02-05:
I just purchased a pi 3b+ ... I'll try to get it setup this weekend and see if I can figure out the right steps to getting the software running on the pi.

nstoik, 2019-02-06:
Okay thanks for the help on this. Let me know if there is anything else I can help test with.

delmonij, 2019-02-16:
If you haven't found an image that makes you happy on the pi3b+ I've had good luck with this for pure 64 bit: 
http://cdimage.ubuntu.com/releases/18.04/release/ubuntu-18.04.2-preinstalled-server-arm64+raspi3.img.xz

I got an alpha version around 5 working on 32 bit running but it was of dubious usefulness since I could only really track about 1 call at a time due to cpu...and in that case had to be careful to set messages to 0 and keep gui minimized, obviously spectrometer/waterfall off...Didn't have enough left to mp3enc...

I did get it working for analog via this path which might interest you...as it may apply to sdrtrunk.

The idea was to use soapysdr-server and it's great decimation enhancement (UDP packet drop) and remote control and support for all sorts of SDR hardware and eventually plumb it into sdrtrunk something along the lines of rtl_fm or rtl_tcp...oh and it supports my lime sdr...

I thought about asking you if we should try to support soapy as a work-saver since it does all of that C&C already but forgot until I noticed this post...It may be worth looking into soapy but I'm not able to take the discovery much further myself. 

If checking out that stuff sounds interesting you should be able to get roughly where I'm at with these steps although they don't get us from point A to point B which is using sdrtrunk with RPI since we don't talk soapy...also note that imo soapy is very tied to ubuntu even working on debian didn't go well. Also make sure you are using soapy v7 or later, it's all pretty new.

xzcat image > /dev/mmcblk?
first boot will take a while (3 min+) to resize filesystem and will only start that process after dhcp completes on eth0
ssh/auth: ubuntu:ubuntu
add-apt-repository ppa:myriadrf/drivers
apt-get install soapysdr-module-rtlsdr soapysdr-module-lms7 soapysdr-server rtl-sdr powertop
systemctl enable SoapySDRServer.service 
systemctl start SoapySDRServer.service 
added powertop --auto-tune to rc.local

soapysdr-server works __surprisingly well__ and advertises over avahi so it's easy to find remotely (with CubicSDR/gqrx of a new enough version) the manual devstring would look something like 'soapy=0,remote=X.X.X.X' also SoapyUtil --find takes same strings

# RPI things that wasted my time
finding a good "safe" compat aarch64 clean image
use a 40w+ laptop type power supply, not a phone charger
booting with HDMI unplugged saves power # hw accel drivers likely missing on 64 it's slooooowwwwww
powertop saves power
the more power we save the more likely we are to be 1Ghz and not crash
use a multi TT usb2 hub that is powered
running 1 rtlsdr is just about at the power limit YMMV but really power usb bus somehow not through GPIO or USB-Micro....
wifi is not sharing the single usb2 hub although eth0 is so it's probably worth installing wpasupplicant and wifi-tools so you could add ssid/psk to netplan although I had better results with wired for unknown reasons.

Can't say thanks enough for this great software!!!

DSheirer, 2019-02-17:
I spent most of the day trying to get version 0.4.0 alpha 9 running on a fresh 32-bit raspbian image on my brand new pi 3b+.  I used the Bellsoft Liberica arm v6 hard float 32-bit JDK and downloaded the Gluon early access JavaFX build for the same architecture.  There are currently no JavaFX builds available for arm 64-bit and a 64-bit arm OS image is not quite yet mainstream.

I updated the gradle build script to inline the Gluon JavaFX libraries into the Liberica build image.  I had to manually edit the sdrtrunk launch script to update the -classpath to remove the windows/mac/linux versions fo the java fx libraries and add back in the Gluon libraries.  The gradle beryx runtime plugin automatically creates the launch script.

I got to the point where the application launches successfully, but produces an error on launching any of the JavaFX windows (e.g. preferences editor).  The stack trace indicated that it couldn't initialize the udev for one of the graphics libraries.  I don't have any experience in troubleshooting/degugging linux graphics libraries and related permissions issues and google wasn't much help, so I'm at a stopping point.  I read that this may be an issue stemming from recent updates to graphics libraries in the most recent raspbian image.

I'm hoping that Gluon will update their JavaFX SDK libraries and post an updated guide to using JavaFX on OpenJDK 11.0.2 linux on arm.  For now, I"m putting this issue on the back burner.

Denny

nstoik, 2019-02-20:
Okay, thanks for taking this as far as you did. I might try and follow what @delmonij suggested and see how far I can get.

If not, then I'll wait and see if the JavaFX SDK gets updated, and we can get a little further on that approach.

I'll leave it up to @DSheirer if you want to keep this issue open or not.

delmonij, 2019-03-16:
The JavaFX got updated on 64bit .... adding this to RPI wasted time post above.....

# This took quite a while the first time so I installed squid-deb-proxy-client
# and squid-deb-proxy-server on lan and opened tcp 8000 and avahi
# fresh install (after blkdiscard /dev/mmcblk0)
apt-get update
apt-get -y dist-upgrade
edit /etc/update-manager/release-upgrades set variable to normal
do-release-upgrade # change to cosmic # yes to everything

apt-get install firmware-raspi3 unzip powertop openjfx openjdk-11-jre ubuntu-mate
# firmware makes it take a long time to load off sd card with rainbow screen
# ubuntu mate is huge compared to the pre-installed image

# HDMI audio
# add snd_bcm2835 enable_compat_alsa=1 to /etc/modules
# add dtparam=audio=on to /boot/firmware/config.txt
# when flash-kernel runs the config.txt line is removed :/
reboot

download and unzip alpha9 for linux
replace bin/java and bin/keytool with links to /usr/bin/java and /usr/bin/keytool

start bin/sdr-trunk

winning!

the system works normally except without accelerated graphics it really sucks
the system is too slow [for me] for local console or remote ssh/x11
gui applications without some form of acceleration working or maybe a tile
caching x server like nx? there may be some vc4 support that works it seems
genuine success has been reported and there are relics around...

found vc4 stuff and some transitional packages from bionic but it seems not
enabled in the cosmic builds

thanks for keeping this open, sdrtrunk a9 runs on aarch64! and for a mainstream install the same thing may work on the raspi2 bionic cdimage for armhf and that might be more easily accelerated if it's not already.

delmonij, 2019-03-16:
I'm sorry I'll learn the markup :( and fix those

TheButterZone, 2019-12-30:
Should I open a new Issue for this on Version 0.4.0 Beta 2 on RPi 3B+ 

`$ ./sdr-trunk-linux-x86_64/bin/sdr-trunk
./sdr-trunk-linux-x86_64/bin/sdr-trunk: 154: exec: /home/pie/sdr-trunk-linux-x86_64/bin/java: Exec format error
`

twigs538, 2020-04-04:
Hi, when I run ./bin/sdr-trunk for sdr-trunk-linux-x86_32-v0.4.0-beta.8 on Ubuntu18 I get the following error:

./bin/sdr-trunk: 154: exec: /home/twigs/sdr-trunk-linux-x86_32-v0.4.0-beta.8/bin/java: not found

What does it mean?

twigs538, 2020-04-04:
When I run the sdrtrunk-0.3.0-beta.12.tar.gz which runs but can't decode P25. It gives the following error:

Exception in thread "AWT-EvenQueue-0" java.lang.ClassCastException: jmbe.JMBEAudioLibrary cannot be cast to jmbe.iface.AudioConversionLibrary

0x4D3331, 2020-08-07:
@DSheirer Any chance of progress on this?

closed 2022-11-26 by DSheirer