DSheirer/sdrtrunk #523: Push Call Events to API

metaphoricalmarlinspike, 2019-01-29:
I've recently migrated my monitoring station from [trunk-recorder](https://github.com/robotastic/trunk-recorder) to sdrtrunk, which has cut down significantly on the number of missed or garbled transmissions. However, for each transmission, I would like to be able to push a compressed (mp3/whatever) audio file with all the metadata from the call up to an Amazon S3 bucket or something like [ScanOC](https://github.com/ScanOC). I have an [existing project](https://github.com/metaphoricalmarlinspike/trunk-recorder-cloud) that processes the metadata from a call in trunk-recorder, re-formats the filename to fit my needs, and uploads it to Amazon S3. However, sdrtrunk does not seem to have an option to generate the JSON file with this metadata. 

It seems like there's a better way to do this with sdrtrunk, given its event driven architecture. That said, I'm not too sure where to start--and Java isn't one of my strong points. Can someone point me in the right direction here?

ImagoTrigger, 2019-02-07:
set your aliases to Record then inspect the generated .wav files in /recordings

metaphoricalmarlinspike, 2019-02-13:
I don't see anything in the audio metadata for the recordings that isn't in the filename already. Maybe VLC is missing something or I just don't know where to look? SDRTrunk outputs most of the information that I need in the file names, so I can probably just zero out the unknown fields and move on. But [TrunkPlayer ](https://github.com/ScanOC/trunk-player) can also handle additional info such as emergency calls, multiple unit IDs in a call, and frequencies. At that point, the information would be getting too long to fit in the file name but could be embedded as metadata. 

I have noticed that there are several files that don't have a talkgroup ID or any radio IDs attached. Most of them are shorter, but this isn't always the case. Is there any way to minimize the number of "orphan" calls like this? It's not a huge deal--in a sample of 5,381 calls there are only 4 with no talkgroup, 129 with no metadata at all, and 864 with no radio ID. Longer calls are more likely to have all the information. Is this an issue with the information not getting added to the file names, or is SDRTrunk not capturing this information in the first place? It's hard to tell from the logs because there is so much information in them. Trunk Recorder has a different process for capturing audio though; as far as I can tell, SDRTrunk will split transmissions from different radios across separate files so there wouldn't ever be a file with multiple radio IDs or frequencies.

DSheirer, 2019-02-15:
Starting with release 0.4.0 alpha 7, each indiividual call segment is output as an individual recording, so ther should only be from and to radio identifiers.  Frequency is included in the metadata.  I'll have to add an emergency attribute.

When you see incomplete metadata for a recording, it's because the information wasn't decoded.  Sometimes a call will be decoded where the beginning is missed.  The from radio id is only treansmitted on every other voice frame, so there may be rare instances where that doesn't get decoded on a short call.

chuot, 2020-02-10:
I'm looking to interface [Rdio Scanner](https://github.com/chuot/rdio-scanner) with SDR Trunk.

My understanding is that SDR Trunk 0.4.0 can record wav audio files with their mbe metadata. I can also define a script to execute for specific aliases. I am currently testing these options on a P25-1 system.

The problems I have encountered so far are:

1. The volume of the wav file is so low that I can't hear anything when playing it.

2. The wav / mbe traffic recording function is performed for all calls from a system when I would like to be able to specify which newsgroup I want to record. Is it possible to have these same recording options but in the alias tab where I could define which talkgroups to record?

3. The alias scripting function works, but does not pass any arguments to the script. So I don't know which file to upload to Rdio Scanner. I also do not know if the script is executed during the creation of the wav file or at the end when it is created.

Do you have any clues that could help me?

Davepc291, 2025-12-01:
hello can someone please help download trunk play