DSheirer/sdrtrunk #527: SDRTrunk 0.4.0a8 still will not decode the P25-1 control frequency.

masterplayer30, 2019-02-01:
To whom may concern:

I am still having to go back to 0.3.4b2 because 0.4.0a8 still will not pick up the P25-1 control frequency at all. I am using Java 8 on Ubuntu 16.04 and this was discussed many times even on newer releases.

0.3.4b2 can for sure decode the control channel but not 0.4.0a8.

Why is this not still fixed? Why am I having to go back to 0.3.4b2?

Please respond:

Mr. Bobby Lee Burgess

DSheirer, 2019-02-01:
@masterplayer30 can you please make a baseband recording of the control channel using the latest version of the software and send it to me?

In the channel configuration, under the Recording tab, check the box next to the baseband recording.  Start the software, enable the channel and then let it run for 2 or 3 minutes.  

Denny

masterplayer30, 2019-02-02:
@DSheirer 

Here is the file you requested: 
[System_Site_Berwick_baseband_20190201_190717.wav.zip](https://github.com/DSheirer/sdrtrunk/files/2823863/System_Site_Berwick_baseband_20190201_190717.wav.zip)

Bobby Burgess

DSheirer, 2019-02-18:
@masterplayer30 Bobby, sorry for taking so long to get back to this.

I'm getting excellent decodes on this file using the P25 C4FM (not lsm) decoder.  Please see the attached screen shot.

[image]

Are you using the C4FM or the LSM decoder?  If you were using the LSM decoder, please try it again with the C4FM decoder and let me know if it works ok?

Denny

masterplayer30, 2019-02-20:
@DSheirer 

I was using the LSM decoder and with your suggestion I was able to make it work this time. But now I am dealing with the app freezing when there is an incoming transmission. 

Bobby Burgess

DSheirer, 2019-02-20:
@masterplayer30 Bobby, are there any errors showing up in the log file or the console before the freeze up or during the freeze up?  I need some more details to help troubleshoot.  What CPU and memory size are in your computer?

masterplayer30, 2019-02-20:
@DSheirer  It just freezes without viewing anything at all and I have to force close it via task manager. The computer is using Intel Celeron Processor N3060 with 2 gb of ram.

closed 2022-11-26 by DSheirer