DSheirer/sdrtrunk #534: Unable to open preferences. AARCH64. Alpha 8 and 9 (main). No P25 Audio.

GeneralZodGit, 2019-02-04:
I am able to compile and run the program but there is no audio as the P25AudioModule is not loading the JMBE library although it is in the same directory as the main jar.

```
10:07:05.201 INFO  i.g.d.m.d.p.a.P25AudioModule - JMBE audio library path is NOT SET in your User Preferences.  [23MB/49MB 48%]
10:07:05.203 INFO  i.g.d.m.d.p.a.P25AudioModule - JMBE audio conversion library NOT FOUND  [23MB/49MB 48%]
```

But when I attempt to open the preferences dialog to set it, I get the following stack trace:

```
Loading library prism_es2 from resource failed: java.lang.UnsatisfiedLinkError: /home/odroid/.openjfx/cache/11/libprism_es2.so: /home/odroid/.openjfx/cache/11/libprism_es2.so: cannot open shared object file: No such file or directory (Possible cause: can't load AMD 64-bit .so on a AARCH64-bit platform)
java.lang.UnsatisfiedLinkError: /home/odroid/.openjfx/cache/11/libprism_es2.so: /home/odroid/.openjfx/cache/11/libprism_es2.so: cannot open shared object file: No such file or directory (Possible cause: can't load AMD 64-bit .so on a AARCH64-bit platform)
	at java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
	at java.base/java.lang.ClassLoader$NativeLibrary.load(ClassLoader.java:2430)
	at java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(ClassLoader.java:2487)
	at java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2684)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2617)
	at java.base/java.lang.Runtime.load0(Runtime.java:767)
	at java.base/java.lang.System.load(System.java:1831)
	at com.sun.glass.utils.NativeLibLoader.installLibraryFromResource(NativeLibLoader.java:205)
	at com.sun.glass.utils.NativeLibLoader.loadLibraryFromResource(NativeLibLoader.java:185)
	at com.sun.glass.utils.NativeLibLoader.loadLibraryInternal(NativeLibLoader.java:157)
	at com.sun.glass.utils.NativeLibLoader.loadLibrary(NativeLibLoader.java:52)
	at com.sun.prism.es2.ES2Pipeline.lambda$static$0(ES2Pipeline.java:68)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at com.sun.prism.es2.ES2Pipeline.<clinit>(ES2Pipeline.java:50)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:315)
	at com.sun.prism.GraphicsPipeline.createPipeline(GraphicsPipeline.java:187)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.init(QuantumRenderer.java:91)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(QuantumRenderer.java:124)
	at java.base/java.lang.Thread.run(Thread.java:834)
Loading library prism_sw from resource failed: java.lang.UnsatisfiedLinkError: /home/odroid/.openjfx/cache/11/libprism_sw.so: /home/odroid/.openjfx/cache/11/libprism_sw.so: cannot open shared object file: No such file or directory (Possible cause: can't load AMD 64-bit .so on a AARCH64-bit platform)
java.lang.UnsatisfiedLinkError: /home/odroid/.openjfx/cache/11/libprism_sw.so: /home/odroid/.openjfx/cache/11/libprism_sw.so: cannot open shared object file: No such file or directory (Possible cause: can't load AMD 64-bit .so on a AARCH64-bit platform)
	at java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
	at java.base/java.lang.ClassLoader$NativeLibrary.load(ClassLoader.java:2430)
	at java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(ClassLoader.java:2487)
	at java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2684)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2617)
	at java.base/java.lang.Runtime.load0(Runtime.java:767)
	at java.base/java.lang.System.load(System.java:1831)
	at com.sun.glass.utils.NativeLibLoader.installLibraryFromResource(NativeLibLoader.java:205)
	at com.sun.glass.utils.NativeLibLoader.loadLibraryFromResource(NativeLibLoader.java:185)
	at com.sun.glass.utils.NativeLibLoader.loadLibraryInternal(NativeLibLoader.java:157)
	at com.sun.glass.utils.NativeLibLoader.loadLibrary(NativeLibLoader.java:52)
	at com.sun.prism.sw.SWPipeline.lambda$static$0(SWPipeline.java:42)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at com.sun.prism.sw.SWPipeline.<clinit>(SWPipeline.java:41)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:315)
	at com.sun.prism.GraphicsPipeline.createPipeline(GraphicsPipeline.java:187)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.init(QuantumRenderer.java:91)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(QuantumRenderer.java:124)
	at java.base/java.lang.Thread.run(Thread.java:834)
Graphics Device initialization failed for :  es2, sw
Error initializing QuantumRenderer: no suitable pipeline found
java.lang.RuntimeException: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
	at com.sun.javafx.tk.quantum.QuantumRenderer.getInstance(QuantumRenderer.java:280)
	at com.sun.javafx.tk.quantum.QuantumToolkit.init(QuantumToolkit.java:222)
	at com.sun.javafx.tk.Toolkit.getToolkit(Toolkit.java:260)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:267)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
	at javafx.embed.swing.JFXPanel.lambda$initFx$1(JFXPanel.java:234)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.init(QuantumRenderer.java:94)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(QuantumRenderer.java:124)
	... 1 more
Feb 04, 2019 10:42:43 AM com.google.common.eventbus.EventBus$LoggingHandler handleException
SEVERE: Exception thrown by subscriber method process(io.github.dsheirer.gui.preference.PreferenceEditorViewRequest) on subscriber io.github.dsheirer.gui.JavaFxWindowManager@5cc69cfe when dispatching event: io.github.dsheirer.gui.preference.PreferenceEditorViewRequest@794e1cdb
java.lang.RuntimeException: No toolkit found
	at com.sun.javafx.tk.Toolkit.getToolkit(Toolkit.java:272)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:267)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
	at javafx.embed.swing.JFXPanel.lambda$initFx$1(JFXPanel.java:234)
	at java.base/java.lang.Thread.run(Thread.java:834)
```

I cannot locate how to manually set this option as a workaround until fixed.
Also, is there a workaround I could do for now?

DSheirer, 2019-02-05:
It looks like you're trying to run this on an ARM processor ODROID but you're using an Intel/AMD x64 build of the Java Development Kit, which doesn't have the correct drivers for the JavaFX graphics.

Try installing the Liberica AARCH64 JDK from here: https://www.bell-sw.com/pages/java-11.0.2/ and then running it.

OR, I have a test build for the next release here: 
https://drive.google.com/open?id=1oGXJb_dA1S_kkYO5yTTu7JdmkKmSe3PS

GeneralZodGit, 2019-02-05:
That's odd, because I was using java jdk 11 for ARM64. (openjdk-11-jdk-headless_11.0.2+9-3~bpo9+1_arm64.deb)

I purged all jdk and jre from my system and downloaded the Liberica version. I then did a "gradlew clean buildSdr" and ran the software using the Liberica version. I am still getting the exact same error when I try to go into the preferences dialog. It seems like it isn't compiling the right version into the code or something.

```
Loading library prism_es2 from resource failed: java.lang.UnsatisfiedLinkError: /home/odroid/.openjfx/cache/11/libprism_es2.so: /home/odroid/.openjfx/cache/11/libprism_es2.so: cannot open shared object file: No such file or directory (Possible cause: can't load AMD 64-bit .so on a AARCH64-bit platform)
java.lang.UnsatisfiedLinkError: /home/odroid/.openjfx/cache/11/libprism_es2.so: /home/odroid/.openjfx/cache/11/libprism_es2.so: cannot open shared object file: No such file or directory (Possible cause: can't load AMD 64-bit .so on a AARCH64-bit platform)
	at java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
	at java.base/java.lang.ClassLoader$NativeLibrary.load(ClassLoader.java:2430)
	at java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(ClassLoader.java:2487)
	at java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2684)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2617)
	at java.base/java.lang.Runtime.load0(Runtime.java:767)
	at java.base/java.lang.System.load(System.java:1831)
	at com.sun.glass.utils.NativeLibLoader.installLibraryFromResource(NativeLibLoader.java:205)
	at com.sun.glass.utils.NativeLibLoader.loadLibraryFromResource(NativeLibLoader.java:185)
	at com.sun.glass.utils.NativeLibLoader.loadLibraryInternal(NativeLibLoader.java:157)
	at com.sun.glass.utils.NativeLibLoader.loadLibrary(NativeLibLoader.java:52)
	at com.sun.prism.es2.ES2Pipeline.lambda$static$0(ES2Pipeline.java:68)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at com.sun.prism.es2.ES2Pipeline.<clinit>(ES2Pipeline.java:50)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:315)
	at com.sun.prism.GraphicsPipeline.createPipeline(GraphicsPipeline.java:187)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.init(QuantumRenderer.java:91)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(QuantumRenderer.java:124)
	at java.base/java.lang.Thread.run(Thread.java:834)
Loading library prism_sw from resource failed: java.lang.UnsatisfiedLinkError: /home/odroid/.openjfx/cache/11/libprism_sw.so: /home/odroid/.openjfx/cache/11/libprism_sw.so: cannot open shared object file: No such file or directory (Possible cause: can't load AMD 64-bit .so on a AARCH64-bit platform)
java.lang.UnsatisfiedLinkError: /home/odroid/.openjfx/cache/11/libprism_sw.so: /home/odroid/.openjfx/cache/11/libprism_sw.so: cannot open shared object file: No such file or directory (Possible cause: can't load AMD 64-bit .so on a AARCH64-bit platform)
	at java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
	at java.base/java.lang.ClassLoader$NativeLibrary.load(ClassLoader.java:2430)
	at java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(ClassLoader.java:2487)
	at java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2684)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2617)
	at java.base/java.lang.Runtime.load0(Runtime.java:767)
	at java.base/java.lang.System.load(System.java:1831)
	at com.sun.glass.utils.NativeLibLoader.installLibraryFromResource(NativeLibLoader.java:205)
	at com.sun.glass.utils.NativeLibLoader.loadLibraryFromResource(NativeLibLoader.java:185)
	at com.sun.glass.utils.NativeLibLoader.loadLibraryInternal(NativeLibLoader.java:157)
	at com.sun.glass.utils.NativeLibLoader.loadLibrary(NativeLibLoader.java:52)
	at com.sun.prism.sw.SWPipeline.lambda$static$0(SWPipeline.java:42)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at com.sun.prism.sw.SWPipeline.<clinit>(SWPipeline.java:41)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:315)
	at com.sun.prism.GraphicsPipeline.createPipeline(GraphicsPipeline.java:187)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.init(QuantumRenderer.java:91)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(QuantumRenderer.java:124)
	at java.base/java.lang.Thread.run(Thread.java:834)
Graphics Device initialization failed for :  es2, sw
Error initializing QuantumRenderer: no suitable pipeline found
java.lang.RuntimeException: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
	at com.sun.javafx.tk.quantum.QuantumRenderer.getInstance(QuantumRenderer.java:280)
	at com.sun.javafx.tk.quantum.QuantumToolkit.init(QuantumToolkit.java:222)
	at com.sun.javafx.tk.Toolkit.getToolkit(Toolkit.java:260)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:267)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
	at javafx.embed.swing.JFXPanel.lambda$initFx$1(JFXPanel.java:234)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.init(QuantumRenderer.java:94)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(QuantumRenderer.java:124)
[... cut ...]
SEVERE: Exception thrown by subscriber method process(io.github.dsheirer.gui.preference.PreferenceEditorViewRequest) on subscriber io.github.dsheirer.gui.JavaFxWindowManager@1b11ef33 when dispatching event: io.github.dsheirer.gui.preference.PreferenceEditorViewRequest@5823bbaf
java.lang.RuntimeException: No toolkit found
	at com.sun.javafx.tk.Toolkit.getToolkit(Toolkit.java:272)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:267)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
	at javafx.embed.swing.JFXPanel.lambda$initFx$1(JFXPanel.java:234)
	at java.base/java.lang.Thread.run(Unknown Source)

```

DSheirer, 2019-02-05:
Apparently Gluon is providing an early access release of JavaFX for armv6hf (hard float for raspbian lite) embedded processors here (bottom of page): https://gluonhq.com/products/javafx/

If you start with the aarch64 liberica jdk.  Unzip the gluon file and then copy the javafx*.jar files in the jmods directory and the other .so files in the libs directory, supposedly this will get JavaFX running on arm.  But, I haven't tried this yet myself.

GeneralZodGit, 2019-02-06:
I downloaded the Gluon early access build and unzipped it into a directory. I then copies the javafx .jar files into the jmods directory where liberica installed it and then copied the Gluon .so files into the lib directory where liberica installed them. Like this:

```
odroid@odroid:~/javafx/armv6hf-sdk/lib$ sudo cp *.jar /usr/lib/jvm/jdk-11-bellsoft-aarch64/jmods/
odroid@odroid:~/javafx/armv6hf-sdk/lib$ sudo cp *.so /usr/lib/jvm/jdk-11-bellsoft-aarch64/lib/
```

I then ran the SDRTrunk code and when into the preferences and the crash still occurs. I also deleted the .openjfx directory in the users home directory to make sure no cache was occuring. And it still crashes when going into the preferences dialog:

```
odroid@odroid:~/SDR/arm64$ java -jar sdr-trunk-all-0.4.0-alpha.9.jar 
13:31:21.791 INFO  i.g.d.log.ApplicationLog - Application Log File: /home/odroid/SDRTrunk/logs/sdrtrunk_app.log  [7MB/27MB 26%]
13:31:21.975 INFO  i.g.d.log.ApplicationLog -   [8MB/27MB 30%]
13:31:22.003 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [8MB/27MB 30%]
13:31:22.008 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [8MB/27MB 30%]
13:31:22.010 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [8MB/27MB 30%]
13:31:22.012 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [8MB/27MB 30%]
13:31:22.014 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [8MB/27MB 30%]
13:31:22.018 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Linux  [8MB/27MB 30%]
13:31:22.032 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          aarch64  [8MB/27MB 30%]
13:31:22.036 INFO  i.g.d.log.ApplicationLog - Host OS Version:       3.16.62-35  [8MB/27MB 30%]
13:31:22.041 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        4  [8MB/27MB 30%]
13:31:22.046 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  417 MB  [8MB/27MB 30%]
13:31:22.048 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [8MB/27MB 30%]
13:31:22.056 INFO  i.g.d.log.ApplicationLog -  Application Root: /home/odroid/SDRTrunk  [8MB/27MB 30%]
13:31:22.061 INFO  i.g.d.log.ApplicationLog -  Application Log:  /home/odroid/SDRTrunk/logs  [8MB/27MB 30%]
13:31:22.067 INFO  i.g.d.log.ApplicationLog -  Event Log:        /home/odroid/SDRTrunk/event_logs  [8MB/27MB 30%]
13:31:22.072 INFO  i.g.d.log.ApplicationLog -  Playlist:         /home/odroid/SDRTrunk/playlist  [8MB/27MB 30%]
13:31:22.077 INFO  i.g.d.log.ApplicationLog -  Recordings:       /home/odroid/SDRTrunk/recordings  [8MB/27MB 30%]
13:31:22.708 INFO  i.g.d.util.ThreadPool - Application thread pool created with [4] threads  [11MB/27MB 42%]
13:31:22.713 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: /home/odroid/SDRTrunk  [11MB/27MB 42%]
13:31:22.746 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/home/odroid/SDRTrunk/SDRTrunk.properties]  [11MB/27MB 43%]
13:31:22.749 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/home/odroid/SDRTrunk/SDRTrunk.properties]  [11MB/27MB 43%]
13:31:22.933 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [/home/odroid/SDRTrunk/settings/settings.xml]  [7MB/27MB 28%]
13:31:26.085 INFO  i.g.d.s.m.MixerManager - loading system mixer devices
	[LOADED]     Input:  default [default] CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: default [default] CHANNELS: [MONO, STEREO]
	[LOADED]     Input:  ODROIDHDMI [plughw:0,0] CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: ODROIDHDMI [plughw:0,0] CHANNELS: [MONO, STEREO]
	[LOADED]     Input:  U0x47f0xad01 [plughw:1,0] CHANNELS: [MONO]
	[LOADED]     Output: U0x47f0xad01 [plughw:1,0] CHANNELS: [MONO, STEREO]
  [11MB/27MB 41%]
13:31:26.094 INFO  i.g.d.s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations  [11MB/27MB 41%]
13:31:26.171 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 16777478  [11MB/27MB 44%]
13:31:26.176 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.22.11312  [11MB/27MB 44%]
13:31:26.179 INFO  i.g.d.s.t.TunerManager - discovered [6] attached USB devices  [11MB/27MB 44%]
13:31:26.196 INFO  i.g.d.s.t.TunerManager - usb device [047F:AD01] NOT LOADED: Unknown Device  [11MB/27MB 44%]
13:31:26.200 INFO  i.g.d.s.t.TunerManager - usb device [148F:5370] NOT LOADED: Unknown Device  [11MB/27MB 44%]
13:31:26.203 INFO  i.g.d.s.t.TunerManager - usb device [05E3:0610] NOT LOADED: Unknown Device  [12MB/27MB 44%]
13:31:26.208 INFO  i.g.d.s.t.TunerManager - usb device [1D6B:0002] NOT LOADED: Unknown Device  [12MB/27MB 44%]
13:31:27.777 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:2838] LOADED: RTL2832 SDR/R820T 00000001  [13MB/27MB 48%]
13:31:27.780 INFO  i.g.d.s.t.TunerManager - usb device [1D6B:0002] NOT LOADED: Unknown Device  [13MB/27MB 48%]
13:31:28.150 INFO  i.g.d.a.AudioPacketManager - Audio packet manager started  [13MB/27MB 50%]
13:31:32.583 INFO  i.g.d.icon.IconManager - loading icons file [/home/odroid/SDRTrunk/settings/icons.xml]  [15MB/27MB 56%]
13:31:34.264 INFO  i.g.d.p.PlaylistManager - Loading playlist file [/home/odroid/SDRTrunk/playlist/default.xml]  [18MB/27MB 68%]
13:31:34.735 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [17MB/27MB 63%]
13:31:45.697 INFO  i.g.d.c.c.ChannelAutoStartFrame - Channel auto-start canceled by user  [23MB/27MB 87%]
[... cut ...]
SEVERE: Exception thrown by subscriber method process(io.github.dsheirer.gui.preference.PreferenceEditorViewRequest) on subscriber io.github.dsheirer.gui.JavaFxWindowManager@6bab2585 when dispatching event: io.github.dsheirer.gui.preference.PreferenceEditorViewRequest@56214d03
java.lang.RuntimeException: No toolkit found
	at com.sun.javafx.tk.Toolkit.getToolkit(Toolkit.java:272)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:267)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
	at javafx.embed.swing.JFXPanel.lambda$initFx$1(JFXPanel.java:234)
	at java.base/java.lang.Thread.run(Thread.java:834)

```

Thanks,

William

hans109h, 2019-02-06:
William,
Does /home/odroid/.openjfx/cache/11/libprism_sw.so exist?  If it does, what are the permissions set to?

GeneralZodGit, 2019-02-06:
It does, here is the long listing:

```
odroid@odroid:~$ ls -l /home/odroid/.openjfx/cache/11/*
-rw-rw-r-- 1 odroid odroid 64039 Feb  6 13:33 /home/odroid/.openjfx/cache/11/libprism_es2.so
-rw-rw-r-- 1 odroid odroid 71548 Feb  6 13:33 /home/odroid/.openjfx/cache/11/libprism_sw.so

```

That seems to be a cache, so it must be generated when it tries to run. I even deleted the directory to make sure and it regenerates.

hans109h, 2019-02-06:
I noticed that the libprism_sw.so file in your cache is larger than that in the zip.  Maybe your lib path isn't correct.  Perhaps try to copy the .so files directly from the zip to the cache?

hans109h, 2019-02-06:
There are also a couple of files in the zip that aren't jar's or so's, but are property files, perhaps these are necessary?  I believe you'd place them in the lib folder.

GeneralZodGit, 2019-02-07:
I tried copying the whole javafx zip into my lib folder for Liberica, and I tried copying the files into the cache. I still get the same stack trace when attempting to open the preferences.

GeneralZodGit, 2019-02-08:
Can you make the software check the same directory and sdr-trunk for jmbe-0.3.3.jar if it is not set in the preferences until this bug can be corrected?

dreinhold, 2019-02-12:
It not clear if others have said it also, but I am unable to open the user preferences using Version 0.4.0 Alpha 8 on my Ubuntu box.
It segfaults.

hans109h, 2019-02-12:
> It is not clear if others have said it also, but I am unable to open the user preferences using Version 0.4.0 Alpha 8 on my Ubuntu box.
> It segfaults.

Anything useful in the log?

dreinhold, 2019-02-12:
[hs_err_pid24153.log](https://github.com/DSheirer/sdrtrunk/files/2857358/hs_err_pid24153.log)
Attached is the full log.

GeneralZodGit, 2019-02-12:
> It is not clear if others have said it also, but I am unable to open the user preferences using Version 0.4.0 Alpha 8 on my Ubuntu box.
> It segfaults.

This thread is for ARM processors (AARCH64). Your crash is caused by a different issue with the X11 library. You should open up a different bug report for your issue separately.

GeneralZodGit, 2019-02-28:
DSheirer,

Since I can't open the preferences, will it work if I change the line of code pointing to the path of the P25 library in:

./src/main/java/io/github/dsheirer/module/decode/p25/audio/P25AudioModule.java        

FROM:
Path path = mUserPreferences.getJmbeLibraryPreference().getPathJmbeLibrary();

TO:
Path path = "/home/generalzod/sdr/";

Will that work? Or can it be coded to look in the same directory first as the default and only look elsewhere if it not null in the preferences dialog?

Thanks,

GeneralZod

DocJr90, 2019-05-04:
Attempting to manually set the user preference in /user/x/.java/.userPrefs/io/github/dsheirer/preference/decoder/ resulted in a IncorrectFormatPrefs.xml

I believe the key name to be "path.jmbe.library" - can someone provide an example xml file from their own instance that would shed light on how exactly the xml file should appear?

DocJr90, 2019-05-05:
> Attempting to manually set the user preference in /user/x/.java/.userPrefs/io/github/dsheirer/preference/decoder/ resulted in a IncorrectFormatPrefs.xml
> I believe the key name to be [...]

One issue resolved - another one appears: now the version of Java I compiled jmbe with is newer than the version of Java that is running sdr-trunk (either the distributed version or the installed version on Linux?).

Below is an example from my environment that I somehow managed to piece together despite having about 10 hours of experience with Linux (and Java) the past few days...

<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!DOCTYPE map SYSTEM "http://java.sun.com/dtd/preferences.dtd">
<map MAP_XML_VERSION="1.0">
        <entry key="path.jmbe.library" value="//home//mark//Desktop//sdr-trunk-linux-aarch64//lib//jmbe-0.3.3.jar"/>
</map>

DocJr90, 2019-05-05:
Finally, some progress! I resolved the version mismatch by adding the following option/value pairs in "build.gradle" of the jmbe source files:

Specifically, in section "project(':jmbe')" and sub-section "dependencies":

sourceCompatibility = 1.7
targetCompatibility = 1.7

Additionally, as detailed here (https://github.com/DSheirer/jmbe/issues/11), if using a JDK version higher than 11, you must also change the "distributionUrl" in the "gradle-wrapper.properties" file in the "/gradle/wrapper/" subdirectory of the jmbe source files.

Again, the user preference for the jmbe library should appear as below in "/user/x/.java/.userPrefs/io/github/dsheirer/preference/decoder/prefs.xml"

`<?xml version="1.0" encoding="UTF-8" standalone="no"?>`
`<!DOCTYPE map SYSTEM "http://java.sun.com/dtd/preferences.dtd">`
`<map MAP_XML_VERSION="1.0">`
`<entry key="path.jmbe.library" value="//home//mark//Desktop//sdr-trunk-linux-aarch64//lib//jmbe-0.3.3.jar"/>`
`</map>`

DocJr90, 2019-05-05:
Confirmed P25 is now working when using the aarch64 build provided by dsheirer earlier in this thread (https://github.com/DSheirer/sdrtrunk/issues/534#issuecomment-460563414). In order to get it working it will require the fixes listed in my previous posts, specifically:

1) One cannot access the preferences menu for whatever reason (the root issue at hand here), so an XML file will need to be created manually with the path to the jmbe library.

2) If the jmbe version is going to be compiled with anything higher than JDK 11 then the compatibility values will need to be set in the dependencies section of the jmbe project settings. Additionally, the distributionUrl for gradle will need to be changed.

3) Lastly, I quickly ran out of heap space and started encountering Java OutOfMemory errors; to resolve this, I edited the "sdr-trunk" shell script file in the bin directory of sdr-trunk; specifically, I added the following arguments to the DEFAULT_JVM_OPTS variable: "-Xms2048m -Xmx2048m" (thus specifying that Java should allocate 2GB to sdr-trunk on startup).

ineedhelphelper, 2019-07-03:
I'm running into a similar issue on Raspberry Pi 4. The CPU is 64-bit, but Raspbian is 32-bit. 
View Preferences is a no go.

OpenJDK Server VM warning: You have loaded library /home/pi/.openjfx/cache/11.0.2/libprism_es2.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
Loading library prism_es2 from resource failed: java.lang.UnsatisfiedLinkError: /home/pi/.openjfx/cache/11.0.2/libprism_es2.so: /home/pi/.openjfx/cache/11.0.2/libprism_es2.so: wrong ELF class: ELFCLASS64 (Possible cause: architecture word width mismatch)
java.lang.UnsatisfiedLinkError: /home/pi/.openjfx/cache/11.0.2/libprism_es2.so: /home/pi/.openjfx/cache/11.0.2/libprism_es2.so: wrong ELF class: ELFCLASS64 (Possible cause: architecture word width mismatch)
	at java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
	at java.base/java.lang.ClassLoader$NativeLibrary.load(ClassLoader.java:2430)
	at java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(ClassLoader.java:2487)
	at java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2684)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2617)
	at java.base/java.lang.Runtime.load0(Runtime.java:767)
	at java.base/java.lang.System.load(System.java:1831)
	at com.sun.glass.utils.NativeLibLoader.installLibraryFromResource(NativeLibLoader.java:205)
	at com.sun.glass.utils.NativeLibLoader.loadLibraryFromResource(NativeLibLoader.java:185)
	at com.sun.glass.utils.NativeLibLoader.loadLibraryInternal(NativeLibLoader.java:157)
	at com.sun.glass.utils.NativeLibLoader.loadLibrary(NativeLibLoader.java:52)
	at com.sun.prism.es2.ES2Pipeline.lambda$static$0(ES2Pipeline.java:68)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at com.sun.prism.es2.ES2Pipeline.<clinit>(ES2Pipeline.java:50)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:315)
	at com.sun.prism.GraphicsPipeline.createPipeline(GraphicsPipeline.java:187)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.init(QuantumRenderer.java:91)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(QuantumRenderer.java:124)
	at java.base/java.lang.Thread.run(Thread.java:834)
Loading library prism_sw from resource failed: java.lang.UnsatisfiedLinkError: /home/pi/.openjfx/cache/11.0.2/libprism_sw.so: /home/pi/.openjfx/cache/11.0.2/libprism_sw.so: wrong ELF class: ELFCLASS64 (Possible cause: architecture word width mismatch)
java.lang.UnsatisfiedLinkError: /home/pi/.openjfx/cache/11.0.2/libprism_sw.so: /home/pi/.openjfx/cache/11.0.2/libprism_sw.so: wrong ELF class: ELFCLASS64 (Possible cause: architecture word width mismatch)
	at java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
	at java.base/java.lang.ClassLoader$NativeLibrary.load(ClassLoader.java:2430)
	at java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(ClassLoader.java:2487)
	at java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2684)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2617)
	at java.base/java.lang.Runtime.load0(Runtime.java:767)
	at java.base/java.lang.System.load(System.java:1831)
	at com.sun.glass.utils.NativeLibLoader.installLibraryFromResource(NativeLibLoader.java:205)
	at com.sun.glass.utils.NativeLibLoader.loadLibraryFromResource(NativeLibLoader.java:185)
	at com.sun.glass.utils.NativeLibLoader.loadLibraryInternal(NativeLibLoader.java:157)
	at com.sun.glass.utils.NativeLibLoader.loadLibrary(NativeLibLoader.java:52)
	at com.sun.prism.sw.SWPipeline.lambda$static$0(SWPipeline.java:42)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at com.sun.prism.sw.SWPipeline.<clinit>(SWPipeline.java:41)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:315)
	at com.sun.prism.GraphicsPipeline.createPipeline(GraphicsPipeline.java:187)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.init(QuantumRenderer.java:91)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(QuantumRenderer.java:124)
	at java.base/java.lang.Thread.run(Thread.java:834)
Graphics Device initialization failed for :  es2, sw
Error initializing QuantumRenderer: no suitable pipeline found
java.lang.RuntimeException: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
	at com.sun.javafx.tk.quantum.QuantumRenderer.getInstance(QuantumRenderer.java:280)
	at com.sun.javafx.tk.quantum.QuantumToolkit.init(QuantumToolkit.java:222)
	at com.sun.javafx.tk.Toolkit.getToolkit(Toolkit.java:260)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:267)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
	at javafx.embed.swing.JFXPanel.lambda$initFx$1(JFXPanel.java:234)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.lang.RuntimeException: Error initializing QuantumRenderer: no suitable pipeline found
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.init(QuantumRenderer.java:94)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(QuantumRenderer.java:124)
	... 1 more
Jul 03, 2019 1:53:25 AM com.google.common.eventbus.EventBus$LoggingHandler handleException
[... cut ...] r.gui.JavaFxWindowManager@197b197 when dispatching event: io.github.dsheirer.gui.preference.PreferenceEditorViewRequest@1208c80
java.lang.RuntimeException: No toolkit found
	at com.sun.javafx.tk.Toolkit.getToolkit(Toolkit.java:272)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:267)
	at com.sun.javafx.application.PlatformImpl.startup(PlatformImpl.java:158)
	at javafx.embed.swing.JFXPanel.lambda$initFx$1(JFXPanel.java:234)
	at java.base/java.lang.Thread.run(Thread.java:834)

closed 2022-11-26 by DSheirer