DSheirer/sdrtrunk #544: will not recognize RTL-SDR V3 in latest SDRTrunk release. worked fine in older releases

Electro2019, 2019-02-26:
<pre><font color="#8AE234"><b>jc@jc</b></font>:<font color="#729FCF"><b>~/git/sdrtrunk-master (1)/build/libs</b></font>$ java -jar sdr-trunk-all-0.4.0-beta.1.jar
22:39:17.836 INFO  i.g.d.log.ApplicationLog - Application Log File: /home/jc/SDRTrunk/logs/sdrtrunk_app.log  [10MB/92MB 11%]
22:39:17.876 INFO  i.g.d.log.ApplicationLog -   [11MB/92MB 12%]
22:39:17.877 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [11MB/92MB 12%]
22:39:17.877 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [11MB/92MB 12%]
22:39:17.877 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [11MB/92MB 12%]
22:39:17.877 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [11MB/92MB 12%]
22:39:17.877 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [11MB/92MB 12%]
22:39:17.878 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Linux  [11MB/92MB 12%]
22:39:17.879 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [11MB/92MB 12%]
22:39:17.879 INFO  i.g.d.log.ApplicationLog - Host OS Version:       4.18.0-15-generic  [11MB/92MB 12%]
22:39:17.880 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        4  [11MB/92MB 12%]
22:39:17.881 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  1 GB  [11MB/92MB 12%]
22:39:17.881 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [11MB/92MB 12%]
22:39:17.883 INFO  i.g.d.log.ApplicationLog -  Application Root: /home/jc/SDRTrunk  [11MB/92MB 12%]
22:39:17.884 INFO  i.g.d.log.ApplicationLog -  Application Log:  /home/jc/SDRTrunk/logs  [11MB/92MB 12%]
22:39:17.884 INFO  i.g.d.log.ApplicationLog -  Event Log:        /home/jc/SDRTrunk/event_logs  [11MB/92MB 12%]
22:39:17.885 INFO  i.g.d.log.ApplicationLog -  Playlist:         /home/jc/SDRTrunk/playlist  [11MB/92MB 12%]
22:39:17.886 INFO  i.g.d.log.ApplicationLog -  Recordings:       /home/jc/SDRTrunk/recordings  [11MB/92MB 12%]
22:39:18.009 INFO  i.g.d.util.ThreadPool - Application thread pool created with [4] threads  [9MB/92MB 10%]
22:39:18.010 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: /home/jc/SDRTrunk  [9MB/92MB 10%]
22:39:18.017 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/home/jc/SDRTrunk/SDRTrunk.properties]  [9MB/92MB 10%]
22:39:18.018 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/home/jc/SDRTrunk/SDRTrunk.properties]  [10MB/92MB 10%]
22:39:18.051 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [/home/jc/SDRTrunk/settings/settings.xml]  [8MB/92MB 9%]
22:39:18.431 INFO  i.g.d.s.m.MixerManager - loading system mixer devices
	[LOADED]     Input:  default [default] CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: default [default] CHANNELS: [MONO, STEREO]
	[LOADED]     Input:  PCH [plughw:0,0] CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: PCH [plughw:0,0] CHANNELS: [MONO, STEREO]
	[LOADED]     Input:  PCH [plughw:0,2] CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: PCH [plughw:0,3] CHANNELS: [MONO, STEREO]
  [26MB/92MB 28%]
22:39:18.432 INFO  i.g.d.s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations  [26MB/92MB 28%]
22:39:18.450 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 16777478  [26MB/92MB 29%]
22:39:18.451 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.22.11312  [26MB/92MB 29%]
22:39:18.452 INFO  i.g.d.s.t.TunerManager - discovered [10] attached USB devices  [26MB/92MB 29%]
22:39:18.458 INFO  i.g.d.s.t.TunerManager - usb device [8087:0024] NOT LOADED: Unknown Device  [26MB/92MB 29%]
22:39:18.459 INFO  i.g.d.s.t.TunerManager - usb device [1D6B:0002] NOT LOADED: Unknown Device  [26MB/92MB 29%]
22:39:18.459 INFO  i.g.d.s.t.TunerManager - usb device [093A:2510] NOT LOADED: Unknown Device  [26MB/92MB 29%]
22:39:18.460 INFO  i.g.d.s.t.TunerManager - usb device [04F2:1112] NOT LOADED: Unknown Device  [26MB/92MB 29%]
22:39:18.460 INFO  i.g.d.s.t.TunerManager - usb device [05E3:0608] NOT LOADED: Unknown Device  [26MB/92MB 29%]
22:39:18.460 INFO  i.g.d.s.t.TunerManager - usb device [8087:0024] NOT LOADED: Unknown Device  [26MB/92MB 29%]
22:39:18.461 INFO  i.g.d.s.t.TunerManager - usb device [1D6B:0002] NOT LOADED: Unknown Device  [26MB/92MB 29%]
22:39:18.462 INFO  i.g.d.s.t.TunerManager - usb device [1D6B:0003] NOT LOADED: Unknown Device  [26MB/92MB 29%]
22:39:18.471 ERROR i.g.d.s.t.TunerManager - couldn&apos;t determine RTL2832 tuner type  [27MB/92MB 29%]
io.github.dsheirer.source.SourceException: couldn&apos;t open device - check permissions Linux: udev.rule? , or Windows: reinstall Zadig? [LIBUSB_ERROR_ACCESS]
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.identifyTunerType(RTL2832TunerController.java:343)
	at io.github.dsheirer.source.tuner.TunerManager.initRTL2832Tuner(TunerManager.java:383)
	at io.github.dsheirer.source.tuner.TunerManager.initTuner(TunerManager.java:225)
	at io.github.dsheirer.source.tuner.TunerManager.initTuners(TunerManager.java:140)
	at io.github.dsheirer.source.tuner.TunerManager.&lt;init&gt;(TunerManager.java:72)
	at io.github.dsheirer.source.SourceManager.&lt;init&gt;(SourceManager.java:48)
	at io.github.dsheirer.gui.SDRTrunk.&lt;init&gt;(SDRTrunk.java:185)
	at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:639)
22:39:18.492 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:2838] NOT LOADED: SDRTRunk doesn&apos;t currently support RTL2832 Dongle with [UNKNOWN] tuner for tuner class[USB Tuner:RTL2832_VARIOUS Vendor:RTL2832 Device:SDR Address:0BDA:2838]  [27MB/92MB 30%]
22:39:18.493 INFO  i.g.d.s.t.TunerManager - usb device [1D6B:0002] NOT LOADED: Unknown Device  [28MB/92MB 30%]
22:39:18.539 INFO  i.g.d.a.AudioPacketManager - Audio packet manager started  [9MB/92MB 10%]
22:39:19.395 INFO  i.g.d.icon.IconManager - loading icons file [/home/jc/SDRTrunk/settings/icons.xml]  [16MB/111MB 14%]
22:39:19.671 INFO  i.g.d.p.PlaylistManager - Loading playlist file [/home/jc/SDRTrunk/playlist/default.xml]  [26MB/111MB 23%]
22:39:19.751 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [27MB/111MB 24%]

</pre>

Electro2019, 2019-02-26:
the only thing ive done different is changed from skywave linux to Ubuntu 18.04. Im sure i have probably overlooked something simple. If you have any idea on how to fix this denny that would be great. I have compiled this with gradle clean build & ./gradlew clean build and get the same problem both ways.

davidmckenzie, 2019-02-26:
```couldn't open device - check permissions Linux: udev.rule? , or Windows: reinstall Zadig? [LIBUSB_ERROR_ACCESS]```

Seems to be the relevant bit - did you install the relevant rtlsdr libraries after switching to Ubuntu 18? Does the dongle work in any other apps like CubicSDR?

Electro2019, 2019-02-26:
i got it thanks i had to black list it as a tv tuner

closed 2019-02-26 by DSheirer