DSheirer/sdrtrunk #546: "RTL-2832 with Rafael Micro R820T or R820T2"

TheButterZone, 2019-03-04:
https://github.com/DSheirer/sdrtrunk/wiki/Tuners_V0.3.0 

https://www.nooelec.com/store/nesdr-smartee-sdr.html doesn't show up in Tuners, nothing does.

davidmckenzie, 2019-03-04:
Does it show up in other sdr programs like gqrx? Make sure you close any programs using the dongle before you start sdrtrunk too. If the device is being used it won't show up in the list.

TheButterZone, 2019-03-04:
When I launch those programs, yes.

DSheirer, 2019-03-05:
Could you please post the console output or the application log?  I need the part where it lists each of the USB devices that it scans on startup when looking for tuners.

TheButterZone, 2019-03-05:
01:21:06.644 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 16777478  [20MB/256MB 8%]
01:21:06.645 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.22.11312  [20MB/256MB 8%]
01:21:06.646 INFO  i.g.d.s.t.TunerManager - discovered [9] attached USB devices  [20MB/256MB 8%]
01:21:06.660 ERROR i.g.d.s.t.r.RTL2832TunerController - error while determining tuner type  [21MB/256MB 8%]
io.github.dsheirer.source.SourceException: couldn't claim usb interface [LIBUSB_ERROR_OTHER]
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.claimInterface(RTL2832TunerController.java:201)
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.identifyTunerType(RTL2832TunerController.java:350)
	at io.github.dsheirer.source.tuner.TunerManager.initRTL2832Tuner(TunerManager.java:383)
	at io.github.dsheirer.source.tuner.TunerManager.initTuner(TunerManager.java:225)
	at io.github.dsheirer.source.tuner.TunerManager.initTuners(TunerManager.java:140)
	at io.github.dsheirer.source.tuner.TunerManager.<init>(TunerManager.java:72)
	at io.github.dsheirer.source.SourceManager.<init>(SourceManager.java:48)
	at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:185)
	at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:639)
01:21:06.688 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:2838] NOT LOADED: SDRTRunk doesn't currently support RTL2832 Dongle with [UNKNOWN] tuner for tuner class[USB Tuner:RTL2832_VARIOUS Vendor:RTL2832 Device:SDR Address:0BDA:2838]  [22MB/256MB 8%]
01:21:06.692 INFO  i.g.d.s.t.TunerManager - usb device [046D:C01D] NOT LOADED: Unknown Device  [22MB/256MB 8%]
01:21:06.693 INFO  i.g.d.s.t.TunerManager - usb device [05AC:0220] NOT LOADED: Unknown Device  [22MB/256MB 8%]
01:21:06.693 INFO  i.g.d.s.t.TunerManager - usb device [05AC:8281] NOT LOADED: Unknown Device  [22MB/256MB 8%]
01:21:06.694 INFO  i.g.d.s.t.TunerManager - usb device [05AC:1006] NOT LOADED: Unknown Device  [22MB/256MB 8%]
01:21:06.694 INFO  i.g.d.s.t.TunerManager - usb device [05AC:8242] NOT LOADED: Unknown Device  [22MB/256MB 8%]
01:21:06.695 INFO  i.g.d.s.t.TunerManager - usb device [0A5C:4500] NOT LOADED: Unknown Device  [22MB/256MB 8%]
01:21:06.702 INFO  i.g.d.s.t.TunerManager - usb device [0424:2513] NOT LOADED: Unknown Device  [22MB/256MB 8%]
01:21:06.705 INFO  i.g.d.s.t.TunerManager - usb device [0424:2513] NOT LOADED: Unknown Device  [22MB/256MB 8%]

DSheirer, 2019-03-07:
@TheButterZone It appears as if your dongle doesn't have an R820T or R820T2.  Instead, it has an unrecognized (for sdrtrunk) tuner.

There's at least 2 tuners that are used with the RTL2832 for these dongles that I haven't added support for yet.  The FC0012 and FC0013.  I recently bought a new dongle that has yet a 3rd tuner type that I've not seen before.

TheButterZone, 2019-03-07:
[My dongle](https://www.nooelec.com/store/nesdr-smartee-sdr.html) does have 

> R820T2 tuner IC

  according to Nooelec. Are they lying?
[image]

hans109h, 2019-03-08:
Did you purchase your tuner directly from Nooelec.com?

TheButterZone, 2019-03-08:
Nooelec.com/store/ linked Where To Buy at the bottom of the page: [Amazon (US)](http://www.amazon.com/s?marketplaceID=ATVPDKIKX0DER&me=A2JO7YP9I9Y3D6&merchant=A2JO7YP9I9Y3D6&tag=nooelec-20)

copy-pasted from my Amazon invoice: 

Sold by: Nooelec ([seller profile](http://smile.amazon.com/gp/help/seller/at-a-glance.html/ref=od_sold_by_link?ie=UTF8&isAmazonFulfilled=1&marketplaceSeller=1&orderID=112-2373201-2825856&seller=A2JO7YP9I9Y3D6)) 

Merchant/seller code on both matches: A2JO7YP9I9Y3D6

hans109h, 2019-03-08:
what's the output of rtl_epprom?

TheButterZone, 2019-03-08:
$ rtl_eeprom
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________

hans109h, 2019-03-08:
Anything interesting in the kernel log?  Just thinking maybe udev is doing something weird.

TheButterZone, 2019-03-08:
Only thing interesting in the logs are the errors in sdrtrunk_app.log I pasted into https://github.com/DSheirer/sdrtrunk/issues/546#issuecomment-469604548 whenever I run sdrtrunk.

hans109h, 2019-03-08:
What platform are you running sdrtrunk on?

TheButterZone, 2019-03-08:
> https://github.com/DSheirer/sdrtrunk/releases/download/v0.4.0-alpha.9/sdr-trunk-0.4.0-alpha.9-osx-x64.zip

OSX 10.13.6

hans109h, 2019-03-09:
I'm not familiar with OSX, but I do know that it doesn't use udev which is specific to linux.  Hopefully someone else who is better equip will come along.

DSheirer, 2019-03-09:
@TheButterZone sorry if I'm asking the same question again, but I wasn't clear on your answer.  Are you  able to use any other applications with this same tuner dongle on this same OSX machine? (originally asked by @davidmckenzie) 

I just checked librtlsdr and it doesn't looks like they've changed or added anything new to support variations of R820T series tuners.  I don't have the NOOELEC dongle, but I believe others have successfully used it with sdrtrunk in the past.

TheButterZone, 2019-03-09:
> When I launch those closed programs, **yes**.

eekdood, 2019-04-09:
I have three of these dongles and initially had issues setting them up on both OSX and Linux. What I found was these are shipped with the same serial number (00000001) and until I changed them using `rtl_eeprom -s 00000010` they would show up as Unknown.

TheButterZone, 2019-04-09:
Isn't the same serial issue only an issue IF you insert multiple dongles? I've only ever had one dongle.

hans109h, 2019-04-09:
You would think so, but perhaps the OS is using the s/n to type the device.  It's worth a test.

TheButterZone, 2019-04-09:
Made no difference to execute & confirm `rtl_eeprom -s 00000010` while connected to the RPi, then move it to the Mac.

hans109h, 2019-04-09:
Thanks for reporting back.

eekdood, 2019-04-09:
Yeah, It was an issue with multiple dongles. I experienced the Unknown Device error on each additional dongle beyond the first one sdrtrunk found.

n8twj, 2019-04-20:
Hi - I have discovered that the UNKNOWN tuner situation first appeared in Version 0.4.0 Alpha 6

0.4.0 Alpha 5 and prior I would get:

14:32:10.591 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:2838] LOADED: RTL2832 SDR/R820T 00000010  [13MB/256MB 5%]

However, I cannot get the application to actually pop a window in alpha 5 and prior:  

Exception in thread "main" java.lang.NoClassDefFoundError: com/sun/java/swing/plaf/windows/WindowsLookAndFeel
	at com.jidesoft.plaf.LookAndFeelFactory.installJideExtension(LookAndFeelFactory.java:845)
	at com.jidesoft.plaf.LookAndFeelFactory.installJideExtension(LookAndFeelFactory.java:633)
	at com.jidesoft.plaf.LookAndFeelFactory.installJideExtension(LookAndFeelFactory.java:598)
	at com.jidesoft.swing.JideSplitButton.updateUI(JideSplitButton.java:109)
	at java.desktop/javax.swing.JMenuItem.init(JMenuItem.java:208)
	at java.desktop/javax.swing.JMenuItem.<init>(JMenuItem.java:147)
	at java.desktop/javax.swing.JMenuItem.<init>(JMenuItem.java:124)
	at java.desktop/javax.swing.JMenu.<init>(JMenu.java:161)
	at com.jidesoft.swing.JideMenu.<init>(JideMenu.java:64)
	at com.jidesoft.swing.JideSplitButton.<init>(JideSplitButton.java:49)
	at io.github.dsheirer.module.decode.event.MessageActivityPanel$MessageHistoryButton.<init>(MessageActivityPanel.java:132)
	at io.github.dsheirer.module.decode.event.MessageActivityPanel$MessageManagementPanel.<init>(MessageActivityPanel.java:100)
	at io.github.dsheirer.module.decode.event.MessageActivityPanel.<init>(MessageActivityPanel.java:59)
	at io.github.dsheirer.channel.metadata.ChannelMetadataViewer.<init>(ChannelMetadataViewer.java:51)
	at io.github.dsheirer.controller.ControllerPanel.<init>(ControllerPanel.java:79)
	at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:175)
	at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:585)
Caused by: java.lang.ClassNotFoundException: com.sun.java.swing.plaf.windows.WindowsLookAndFeel
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:582)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:190)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:499)
	... 17 more

n8twj, 2019-04-20:
I can make things work great on Windows 7, using the same exact Dongle on 0.4.0 Alpha 9

TheButterZone, 2019-05-03:
https://github.com/DSheirer/sdrtrunk/releases/tag/v0.4.0-alpha.5 recognizes it & pops the GUI in OSX.
 
All I want to listen to is Motorola Type II SmartZone systems in San Diego.

TheButterZone, 2019-05-09:
Same BS with the RTL-SDR Blog V3 R820T2 RTL2832U & https://github.com/DSheirer/sdrtrunk/releases/download/v0.4.0-alpha.9/sdr-trunk-0.4.0-alpha.9-osx-x64.zip

12:19:37.943 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:2838] NOT LOADED: SDRTRunk doesn't currently support RTL2832 Dongle with [UNKNOWN] tuner for tuner class[USB Tuner:RTL2832_VARIOUS Vendor:RTL2832 Device:SDR Address:0BDA:2838]  [23MB/256MB 9%]

DSheirer, 2019-05-10:
I just ordered one of the RTL-SDR blog V3 dongles ... I try to see why it's not recognizing them.

D-E-German, 2019-09-09:
I'm having the same issue as well, any news? I'm running OSX and also a VM Windows. I'm using the Nooelec NESDR SMART dongle and had no luck getting SDRTrunk to recognize the tuner. I'm currently running the latest Alpha release. I can also run the tuner quite well in CubicSDR

DSheirer, 2019-12-21:
I don't have any means to test on versions older that 10.15.  I updated the minimum system requirements to list 10.14 or higher.

closed 2019-12-21 by DSheirer

TheButterZone, 2026-09-03:
Not that anyone cares but...

Finally got both my RSP1 working in https://github.com/DSheirer/sdrtrunk/releases#release-v0.6.1 via 

sudo killall sdrplay_apiService 2>/dev/null; /Applications/sdr-trunk-osx-x86_64-v0.6.1/bin/sdr-trunk 

on Catalina (also had to remove lib\shaders.metallib at least, forgot what else in all the hours of ChatGPT troubleshooting). 

My RTL-2832/R820T (NESDR SMArTee) works too now standalone, since one system rebanded to a smaller frequency range, well-within its usable bandwidth.