DSheirer/sdrtrunk #547: SDRtrunk forgetting settings

RetiredDude, 2019-03-06:
SDRtrunk on macbook pro running 10.14.4 Beta works great. The hiccup is... It never remembers anything once I shut it down. Each time I open it I have to set the Receiver, Control channel, and any Aliases. Every time. 
After installing SDRtrunk I open it by going to Terminal, SDRtrunk, cd build, cd libs, and then i past the jar command of java -jar sdr-trunk-all-0.4.0-beta.1.jar. It opens and runs great. I can't change any of the Directories under Preferences other than the top one, the Root directory, which I haven't changed. Any help would be greatly appreciated!! Thanks, Jim
[image]

DSheirer, 2019-03-07:
Are you able to download and use version 0.4.0 Beta 9?  Does it exhibit the same issue?

RetiredDude, 2019-03-07:
Well I downloaded the OSX version sdr-trunk-0.4.0-alpha.9-osx-x64.zip <https://github.com/DSheirer/sdrtrunk/releases/download/v0.4.0-alpha.9/sdr-trunk-0.4.0-alpha.9-osx-x64.zip> but frankly I’m new(er) to Mac OS and need some instructions on how to install this version. SDTrunk I have now was installed watching a video by Zach Whitiker. He is running  linux system and there were a couple commands that would not run in my OSX terminal. It was a bit ugly on my end… but somehow I got it running. 
The reason I say my install was ugly is because to get mine up and running I have to run the Jar string this way:
Open Terminal
cd sdrtrunk
cd build
cd libs
paste jar string "java -jar sdr-trunk-all-0.4.0-beta.1.jar” and hit Enter.
SDRtrunk opens and runs

Is there a video or other location explaining how to install sdr-trunk-0.4.0-alpha.9-osx-x64.zip <https://github.com/DSheirer/sdrtrunk/releases/download/v0.4.0-alpha.9/sdr-trunk-0.4.0-alpha.9-osx-x64.zip> on OSX once you have it downloaded?

Thanks,
Jim

RetiredDude, 2019-03-07:
I fixed it!! I was trying to run 0.4.0 from the downloads folder but figured out each time it was somehow defaulting to the other version in the root directory. I had to delete everything sdrtrunk in the root directory and then ran 0.4.0 again and it ran correctly and now saves everything! I did like reading in the description everything was already part of it and I didn't have to build all the related files. 
This thing is awesome!! Thank You!

closed 2019-12-21 by DSheirer