DSheirer/sdrtrunk #548: Aliases not playing

RetiredDude, 2019-03-07:
About 30% of the time the Aliases I have show in the Events log or Control log (sometimes both) but they fail to play any audio. Is this a limited bandwidth issue with my SDR dongle? Im monitoring Motorola p25 phase 1 running from 854.5625 to 859.9875. Using a NooElec RTL-SDR.  I would think if the dongle is seeing it and SDRtrunk is displaying it it would play.  Is there something I'm missing?
Thanks,
Jim (retired dude)

davidmckenzie, 2019-03-07:
You'd need multiple dongles to cover that range fully - the bandwidth of an RTL dongle is only around 2mhz max.

RetiredDude, 2019-03-08:
I suspected....
and your right its only 2.4mhz
Do you know how I set these up to span the whole bandwidth if I did pickup a second dongle?

davidmckenzie, 2019-03-08:
It just magically works :) SDRTrunk automatically moves the centre frequency around on both dongles to encompass the bandwidth as required.

hans109h, 2019-03-08:
From the Wiki:

> Tuner

 Management

The application automatically manages all tuners as a pooled set of resources. As you enable channels for decoding, sdrtrunk polls each tuner to see if it can support the requested channel frequency, and then assigns the channel to the tuner. If no tuners can support the channel, then the channel will not start and you will receive a popup message indicating that no tuners are available.

Each tuner can source one or more channel frequencies so long as the set of channel frequencies fit within the current center tuned frequency and bandwidth (sample rate) of the tuner. As additional channels are sourced by the tuner, the tuner controller will automatically adjust the center frequency to provide a best-fit of all sourced channels, while ensuring that no channels overlap any DC-spike that may be present in the center of the spectrum for some types of tuners.

> 
On thing to keep in mind is that each tuner will need to be assigned a unique serial number, there are utilities to do this.

nr0q, 2019-03-08:
Anyone have a name or link to a utility to change a dongles serial number.

hans109h, 2019-03-08:
rtl_eeprom, it's part of the rtl_sdr package.  Use it carefully.

closed 2023-11-15 by DSheirer