DSheirer/sdrtrunk #551: couldn't apply teh sample rate setting

NoraCodes, 2019-03-24:
When I try to select my R820T dongle, I get:

```
21:15:50.505 ERROR i.g.d.s.t.r.r.R820TTunerEditor - R820T Tuner Controller - couldn't apply sample rate setting [2.400 MHz]  [87MB/252MB 34%]
org.usb4java.LibUsbException: USB error 9: error writing byte buffer: Pipe error
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.write(RTL2832TunerController.java:822)
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.writeDemodRegister(RTL2832TunerController.java:718)
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.setSampleRate(RTL2832TunerController.java:972)
	at io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerEditor$3.actionPerformed(R820TTunerEditor.java:144)
	at java.desktop/javax.swing.JComboBox.fireActionEvent(Unknown Source)
	at java.desktop/javax.swing.JComboBox.setSelectedItem(Unknown Source)
	at io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerEditor.setItem(R820TTunerEditor.java:589)
	at io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerEditor.setItem(R820TTunerEditor.java:55)
	at io.github.dsheirer.source.tuner.TunerEditor.setTunerConfiguration(TunerEditor.java:337)
	at io.github.dsheirer.source.tuner.TunerEditor$1.valueChanged(TunerEditor.java:117)
	at java.desktop/javax.swing.DefaultListSelectionModel.fireValueChanged(Unknown Source)
	at java.desktop/javax.swing.DefaultListSelectionModel.fireValueChanged(Unknown Source)
	at java.desktop/javax.swing.DefaultListSelectionModel.fireValueChanged(Unknown Source)
	at java.desktop/javax.swing.DefaultListSelectionModel.changeSelection(Unknown Source)
	at java.desktop/javax.swing.DefaultListSelectionModel.changeSelection(Unknown Source)
	at java.desktop/javax.swing.DefaultListSelectionModel.setSelectionInterval(Unknown Source)
	at java.desktop/javax.swing.JTable.setRowSelectionInterval(Unknown Source)
	at io.github.dsheirer.source.tuner.TunerEditor.setItem(TunerEditor.java:264)
	at io.github.dsheirer.source.tuner.TunerViewPanel$1.valueChanged(TunerViewPanel.java:100)
	at java.desktop/javax.swing.DefaultListSelectionModel.fireValueChanged(Unknown Source)
	at java.desktop/javax.swing.DefaultListSelectionModel.fireValueChanged(Unknown Source)
	at java.desktop/javax.swing.DefaultListSelectionModel.setValueIsAdjusting(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicTableUI$Handler.setValueIsAdjusting(Unknown Source)
	at java.desktop/javax.swing.plaf.basic.BasicTableUI$Handler.mouseReleased(Unknown Source)
	at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
	at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
	at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
Exception in thread "AWT-EventQueue-0" org.usb4java.LibUsbException: USB error 9: error writing byte buffer: Pipe error
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.write(RTL2832TunerController.java:822)
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.writeDemodRegister(RTL2832TunerController.java:718)
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.enableI2CRepeater(RTL2832TunerController.java:622)
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.writeI2CRegister(RTL2832TunerController.java:681)
	at io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerController.writeR820TRegister(R820TTunerController.java:656)
	at io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerController.setLNAGain(R820TTunerController.java:696)
	at io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerController.setGain(R820TTunerController.java:686)
	at io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerEditor$6.actionPerformed(R820TTunerEditor.java:242)
[... 28 lines cut ...]
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
```

hans109h, 2019-03-24:
What dongle are you using?  Have you tried changing the sample rate setting?

Please post the log generated at the startup of sdrtrunk.

NoraCodes, 2019-03-24:
I'm using: https://www.rtl-sdr.com/buy-rtl-sdr-dvb-t-dongles/

I cannot change the sample rate:
[image]

Log is as below:

```
21:14:55,847 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback-test.xml]
21:14:55,847 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback.groovy]
21:14:55,848 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Found resource [logback.xml] at [jar:file:/home/nora/.bin/sdr-trunk-linux-x64/lib/sdr-trunk-0.4.0-alpha.9.jar!/logback.xml]
21:14:55,848 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs multiple times on the classpath.
21:14:55,848 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs at [jar:file:/home/nora/.bin/sdr-trunk-linux-x64/lib/sdr-trunk-0.4.0-alpha.9.jar!/logback.xml]
21:14:55,848 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs at [jar:file:/home/nora/.bin/sdr-trunk-linux-x64/lib/radio-reference-api-15.0.1.jar!/logback.xml]
21:14:55,856 |-INFO in ch.qos.logback.core.joran.spi.ConfigurationWatchList@7219ec67 - URL [jar:file:/home/nora/.bin/sdr-trunk-linux-x64/lib/sdr-trunk-0.4.0-alpha.9.jar!/logback.xml] is not of type file
21:14:55,937 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - debug attribute not set
21:14:55,938 |-INFO in ch.qos.logback.core.joran.action.ConversionRuleAction - registering conversion word memory_usage with class [io.github.dsheirer.util.MemoryUsageLogger]
21:14:55,938 |-INFO in ch.qos.logback.core.joran.action.TimestampAction - Using current interpretation time, i.e. now, as time reference.
21:14:55,955 |-INFO in ch.qos.logback.core.joran.action.TimestampAction - Adding property to the context with key="appstart" and value="211455" to the LOCAL scope
21:14:55,955 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - About to instantiate appender of type [ch.qos.logback.core.ConsoleAppender]
21:14:55,958 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - Naming appender as [STDOUT]
21:14:55,962 |-INFO in ch.qos.logback.core.joran.action.NestedComplexPropertyIA - Assuming default type [ch.qos.logback.classic.encoder.PatternLayoutEncoder] for [encoder] property
21:14:55,976 |-INFO in ch.qos.logback.classic.joran.action.RootLoggerAction - Setting level of ROOT logger to DEBUG
21:14:55,976 |-INFO in ch.qos.logback.core.joran.action.AppenderRefAction - Attaching appender named [STDOUT] to Logger[ROOT]
21:14:55,977 |-INFO in ch.qos.logback.classic.joran.action.LoggerAction - Setting level of logger [org.apache.mina] to INFO
21:14:55,977 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - End of configuration.
21:14:55,978 |-INFO in ch.qos.logback.classic.joran.JoranConfigurator@45018215 - Registering current configuration as safe fallback point

21:14:56.245 INFO  i.g.d.log.ApplicationLog - Application Log File: /root/SDRTrunk/logs/sdrtrunk_app.log  [19MB/252MB 7%]
21:14:56.271 INFO  i.g.d.log.ApplicationLog -   [20MB/252MB 8%]
21:14:56.271 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [20MB/252MB 8%]
21:14:56.271 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [20MB/252MB 8%]
21:14:56.271 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [20MB/252MB 8%]
21:14:56.271 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [20MB/252MB 8%]
21:14:56.271 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [20MB/252MB 8%]
21:14:56.272 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Linux  [20MB/252MB 8%]
21:14:56.272 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [20MB/252MB 8%]
21:14:56.272 INFO  i.g.d.log.ApplicationLog - Host OS Version:       4.18.0-16-generic  [20MB/252MB 8%]
21:14:56.272 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        12  [20MB/252MB 8%]
21:14:56.273 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  3 GB  [20MB/252MB 8%]
21:14:56.273 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [20MB/252MB 8%]
21:14:56.273 INFO  i.g.d.log.ApplicationLog -  Application Root: /root/SDRTrunk  [20MB/252MB 8%]
21:14:56.274 INFO  i.g.d.log.ApplicationLog -  Application Log:  /root/SDRTrunk/logs  [20MB/252MB 8%]
21:14:56.274 INFO  i.g.d.log.ApplicationLog -  Event Log:        /root/SDRTrunk/event_logs  [20MB/252MB 8%]
21:14:56.274 INFO  i.g.d.log.ApplicationLog -  Playlist:         /root/SDRTrunk/playlist  [20MB/252MB 8%]
21:14:56.275 INFO  i.g.d.log.ApplicationLog -  Recordings:       /root/SDRTrunk/recordings  [20MB/252MB 8%]
21:14:56.345 INFO  i.g.d.util.ThreadPool - Application thread pool created with [12] threads  [24MB/252MB 9%]
21:14:56.345 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: /root/SDRTrunk  [24MB/252MB 9%]
21:14:56.348 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/root/SDRTrunk/SDRTrunk.properties]  [24MB/252MB 9%]
21:14:56.349 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/root/SDRTrunk/SDRTrunk.properties]  [24MB/252MB 9%]
21:14:56.369 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [/root/SDRTrunk/settings/settings.xml]  [7MB/252MB 2%]
21:14:56.661 INFO  i.g.d.s.m.MixerManager - loading system mixer devices
[... cut ...]
21:14:59.396 INFO  i.g.d.s.t.TunerManager - usb device [1D6B:0002] NOT LOADED: Unknown Device  [10MB/252MB 4%]
21:14:59.425 INFO  i.g.d.a.AudioPacketManager - Audio packet manager started  [12MB/252MB 4%]
21:14:59.849 INFO  i.g.d.icon.IconManager - loading icons file [/root/SDRTrunk/settings/icons.xml]  [29MB/252MB 11%]
21:15:00.001 INFO  i.g.d.p.PlaylistManager - Loading playlist file [/root/SDRTrunk/playlist/default.xml]  [23MB/252MB 9%]
21:15:00.036 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [25MB/252MB 10%]
```

DSheirer, 2019-03-24:
You're gettting a USB pipe error.  Try unplugging the dongle and replugging it.  You might also try plugging it into a different USB port.

Another thought ... have you applied the UDEV rule to allow your user account to have access to the hardware device?  I see that you're running as root, so that may not matter.

NoraCodes, 2019-03-24:
> You're gettting a USB pipe error. Try unplugging the dongle and replugging it. You might also try plugging it into a different USB port.

Did this, no change.

> Another thought ... have you applied the UDEV rule to allow your user account to have access to the hardware device? I see that you're running as root, so that may not matter.

Yes, GNURadio, GQRX, et cetera all work fine with my udev settings

DSheirer, 2019-03-25:
Are you running this in a VM?

NoraCodes, 2019-03-25:
No, on the bare metal. Also, I tried measuring voltage/current going into the dongle. It was low, but using a powered hub did not solve the problem.

aaomidi, 2020-06-04:
Hitting this issue as well on Windows

[image]

aaomidi, 2020-06-04:
So I did the process on https://www.rtl-sdr.com/rtl-sdr-quick-start-guide/

On two seperate operating systems. Both of them ended up with the same error.

DSheirer, 2020-10-25:
Resolved under #973

closed 2020-10-25 by DSheirer

cliffjao, 2021-04-03:
@DSheirer This is still happening for me with 0.5.0 Alpha 6 on Linux with a R820T. I can see the waterfall. I've tried plugging into all the different USB ports, and I've tried running as root as well.

Nat-As, 2022-05-01:
Same issue on windows with NESDR Smart XTR E4000 Tuner. RTL2832U
[image]

DSheirer, 2022-05-01:
@Nat-As Try unplugging and replugging the dongle to see if that clears the pipe error.