DSheirer/sdrtrunk #553: call event logging no longer shows talkgroup name/alias

ImagoTrigger, 2019-04-10:
v3 event log would write the text of the alias name into the event log file as well as the id.  v4 just writes the decimal id of the talkgroup.

ImagoTrigger, 2019-04-11:
[alias_in_event_log.diff.txt](https://github.com/DSheirer/sdrtrunk/files/3066869/alias_in_event_log.diff.txt)

ImagoTrigger, 2019-04-11:
[alias_in_event_log.diff.txt](https://github.com/DSheirer/sdrtrunk/files/3066874/alias_in_event_log.diff.txt)

ImagoTrigger, 2019-04-11:
there is probably a better way to do this, without loading the alias list each time?  it works for me for now

DSheirer, 2019-04-11:
I pulled the alias list out of a lot of the code to try and clean up the code.  I was having to pass the alias list down to each and every message that was decoded from the channel in order to get the aliases included in the log files.

If it's important, I can figure out a way to add support back in, but in a cleaner way that doesn't have me embedding the aliast list in the hundreds of message parsing classes.

closed 2023-11-15 by DSheirer