DSheirer/sdrtrunk #556: Recorded audio WAV files incorrrect padding in metadata INFO chunk

jasoveen, 2019-04-17:
I'm trying to read the recorded audio WAV file MetaData in c# and it looks like the WAV files have incorrect padding in the MetaData INFO chunks. I have tried a handful of programs on Windows and they all seem to have some trouble reading the MetaData.

The RIFF format says there should be a "pad byte" after the variable-sized data field if the chunk's length is not even.
https://en.wikipedia.org/wiki/Resource_Interchange_File_Format

It appears that sdrtrunk is doing the opposite and padding when the chunk length is even which causes the tags to incorrectly get offset.

For example take the "ISFT" tag, the chunk length value is set to "09". The data for this tag is "sdrtrunk" plus a NULL termination character which totals 9. Since the chunk length is odd it should have a pad byte added, but one is not getting added in the WAV file..

The "CHFQ" tag has a chunk length value set to 10 (0A in hex). The data for this tag is "858487500" plus a NULL termination character which totals 10. Since the chunk length is even it should not have a pad byte added, but one is getting added in the WAV file.

It also seems like when it does add a pad byte that it's sometimes adding that in the chunk length value.
For the "ICMT" tag it has a chunk length value set to 13 (0D in hex) and the data for this tag is "P25 Phase I" plus a NULL termination character which totals 12. There is an extra pad byte before the start of the next tag which makes 13 bytes and that matches the chunk length that's getting written. The correct chunk length would be 12 (0C in hex).

I'm using sdrtrunk 0.4.0 Alpha 9

DSheirer, 2019-04-19:
I'll get this fixed for the next release.

DSheirer, 2020-03-19:
Resolved in MR #675

closed 2020-03-19 by DSheirer