DSheirer/sdrtrunk #560: Java Exception Out of Memory

radito, 2019-05-25:
I just got this error today. This thing happens when i enable waterfall. Suddenly the app is crashing.
```
My system
OS: Kali GNU/Linux Rolling x86_64
Kernel: Linux dream 5.0.1 #1 SMP Tue Mar 12 10:12:37 WIB 2019 x86_64 GNU/Linux
Java: 
java version "11.0.2" 2019-01-15 LTS
Java(TM) SE Runtime Environment 18.9 (build 11.0.2+9-LTS)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11.0.2+9-LTS, mixed mode)

```
```
07:37:31.312 INFO  i.g.d.m.d.p.a.P25AudioModule - Loading JMBE library from [/home/radito/jmbe/build/libs/jmbe-0.3.3.jar]  [86MB/124MB 69%]
07:37:31.367 INFO  i.g.d.m.d.p.a.P25AudioModule - JMBE audio conversion library successfully loaded - P25 audio will be available  [86MB/124MB 70%]
07:37:31.440 DEBUG i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2880000.0] providing [114] channels at [25263.157894736843] Hz each  [63MB/124MB 50%]
07:38:10.332 INFO  i.g.d.s.t.FrequencyErrorCorrectionManager - Auto-Correcting Tuner PPM to [-9.982047795254886]  [1GB/1GB 97%]
07:38:30.048 ERROR i.g.d.d.f.c.PolyphaseChannelManager - Error  [1GB/1GB 99%]
java.lang.OutOfMemoryError: Java heap space
        at io.github.dsheirer.sample.buffer.ReusableChannelResultsBuffer.getEmptyBuffer(ReusableChannelResultsBuffer.java:69)
        at io.github.dsheirer.dsp.filter.channelizer.ComplexPolyphaseChannelizerM2.process(ComplexPolyphaseChannelizerM2.java:357)
        at io.github.dsheirer.dsp.filter.channelizer.ComplexPolyphaseChannelizerM2.receive(ComplexPolyphaseChannelizerM2.java:218)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:596)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:557)
        at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor.process(ContinuousBufferProcessor.java:189)
        at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor$Processor.run(ContinuousBufferProcessor.java:224)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
        at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
        at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
07:38:32.194 ERROR i.g.d.d.f.c.ContinuousBufferProcessor - Error while dispatching buffers to listener  [1GB/1GB 99%]
java.lang.IllegalStateException: User count is below zero.  This indicates that this buffer's decrement user count was invoked by more than the expected user count
        at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.recycle(AbstractReusableBuffer.java:119)
        at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.decrementUserCount(AbstractReusableBuffer.java:100)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:609)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:557)
        at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor.process(ContinuousBufferProcessor.java:189)
        at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor$Processor.run(ContinuousBufferProcessor.java:224)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
        at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
        at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at java.base/java.lang.Thread.run(Unknown Source)

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "AWT-EventQueue-0"
^C^C^C
Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "Timer-0"
07:39:27.007 ERROR i.g.d.d.f.c.PolyphaseChannelManager - Error  [1GB/1GB 99%]
java.lang.OutOfMemoryError: Java heap space
07:39:28.138 ERROR i.g.d.d.f.c.ContinuousBufferProcessor - Error while dispatching buffers to listener  [1GB/1GB 99%]
java.lang.IllegalStateException: User count is below zero.  This indicates that this buffer's decrement user count was invoked by more than the expected user count
        at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.recycle(AbstractReusableBuffer.java:119)
        at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.decrementUserCount(AbstractReusableBuffer.java:100)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:609)
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:557)
        at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor.process(ContinuousBufferProcessor.java:189)
        at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor$Processor.run(ContinuousBufferProcessor.java:224)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
        at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
        at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at java.base/java.lang.Thread.run(Unknown Source)
Exception in thread "JavaFX Application Thread" java.lang.NullPointerException
        at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
        at com.sun.glass.ui.gtk.GtkApplication.lambda$runLoop$11(GtkApplication.java:277)
        at java.base/java.lang.Thread.run(Unknown Source)
                                                                                                                      
```

DSheirer, 2019-05-25:
Once the application runs out of memory, you'll have to restart the application.  Try either reducing the workload (number of channels decoding) or run the software on a computer with a more powerful CPU and/or more memory.

closed 2019-05-25 by DSheirer

radito, 2019-05-30:
Okay, but my java only allocated 1gb for the application that's why it's always crashing when decoding the calls. How do I increase it? Or is it coded to always allocating 1 gb ?

```
17:30:22.389 INFO  i.g.d.log.ApplicationLog - Application Log File: /home/radito/SDRTrunk/logs/sdrtrunk_app.log  [11MB/124MB 9%]
17:30:22.514 INFO  i.g.d.log.ApplicationLog -   [12MB/124MB 10%]
17:30:22.516 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [12MB/124MB 10%]
17:30:22.517 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [12MB/124MB 10%]
17:30:22.518 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [12MB/124MB 10%]
17:30:22.519 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [12MB/124MB 10%]
17:30:22.520 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [12MB/124MB 10%]
17:30:22.530 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Linux  [12MB/124MB 10%]
17:30:22.534 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [12MB/124MB 10%]
17:30:22.537 INFO  i.g.d.log.ApplicationLog - Host OS Version:       5.0.1  [12MB/124MB 10%]
17:30:22.540 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        2  [12MB/124MB 10%]
17:30:22.541 INFO  i.g.d.log.ApplicationLog - **Host Max Java Memory:  1 GB  [12MB/124MB 10%]**
```

DocJr90, 2019-05-30:
radito, see my post as linked below regarding settings memory options.

https://github.com/DSheirer/sdrtrunk/issues/534#issuecomment-489435700

radito, 2019-05-30:
@DocJr90 ok, thanks, will look into it.

radito, 2019-05-30:
Ok solved for now, just allocated 4GB memory for the application by changing the line in the sdr-trunk startup script.

``
DEFAULT_JVM_OPTS="-Xms4096m -Xmx4096m"
``

DocJr90, 2019-05-30:
Awesome!

radito, 2019-05-30:
Ok after increasing the memory it turned out the problem doesn't go away it still showing memory errors after running it for a couple of minutes.

I just think it doesn't free the previous memory.
```
18:16:05.676 INFO  i.g.d.s.t.FrequencyErrorCorrectionManager - Auto-Correcting Tuner PPM to [76.09504319480786]  [1GB/4GB 35%]                                          
18:30:45.903 INFO  i.g.d.s.t.FrequencyErrorCorrectionManager - Auto-Correcting Tuner PPM to [75.62728216222538]  [2GB/4GB 69%]                                          
18:38:30.683 INFO  i.g.d.s.t.FrequencyErrorCorrectionManager - Auto-Correcting Tuner PPM to [75.09169577991844]  [2GB/4GB 60%]                                          
18:50:26.201 ERROR i.g.d.d.f.c.PolyphaseChannelManager - Error  [3GB/4GB 99%]
java.lang.OutOfMemoryError: Java heap space
        at io.github.dsheirer.sample.buffer.ReusableChannelResultsBuffer.getEmptyBuffer(ReusableChannelResultsBuffer.java:69)                                           
        at io.github.dsheirer.dsp.filter.channelizer.ComplexPolyphaseChannelizerM2.process(ComplexPolyphaseChannelizerM2.java:357)                                      
        at io.github.dsheirer.dsp.filter.channelizer.ComplexPolyphaseChannelizerM2.receive(ComplexPolyphaseChannelizerM2.java:218)                                      
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:596)                         
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:557)                         
        at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor.process(ContinuousBufferProcessor.java:189)                                              
        at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor$Processor.run(ContinuousBufferProcessor.java:224)                                        
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)                                                                                
        at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
        at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)                                                           
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)                                                                                  
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)                                                                                 
        at java.base/java.lang.Thread.run(Unknown Source)
18:50:31.471 ERROR i.g.d.d.f.c.ContinuousBufferProcessor - Error while dispatching buffers to listener  [3GB/4GB 99%]                                                   
java.lang.IllegalStateException: User count is below zero.  This indicates that this buffer's decrement user count was invoked by more than the expected user count     
        at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.recycle(AbstractReusableBuffer.java:119)                                                             
        at io.github.dsheirer.sample.buffer.AbstractReusableBuffer.decrementUserCount(AbstractReusableBuffer.java:100)                                                  
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:609)                         
        at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$BufferSourceEventMonitor.receive(PolyphaseChannelManager.java:557)                         
        at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor.process(ContinuousBufferProcessor.java:189)                                              
        at io.github.dsheirer.dsp.filter.channelizer.ContinuousBufferProcessor$Processor.run(ContinuousBufferProcessor.java:224)                                        
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)                                                                                
        at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
        at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)                                                           
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)                                                                                  
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)                                                                                 
        at java.base/java.lang.Thread.run(Unknown Source)
Exception in thread "Timer-0" 18:51:36.070 ERROR i.g.d.s.t.u.USBTransferProcessor - Error while processing USB transfer buffers  [3GB/4GB 99%]
java.lang.OutOfMemoryError: Java heap space                                                                                                                             
18:51:36.070 ERROR i.g.d.d.f.c.ContinuousBufferProcessor - Error while dispatching buffers to listener  [3GB/4GB 99%]                                                   
java.lang.OutOfMemoryError: Java heap space                                    
java.lang.OutOfMemoryError: Java heap space  
```

DSheirer, 2019-06-02:
@radito The log is showing that your CPU only has 2 cores.  sdrtrunk does reclaim used memory but it's handled by Java in background threads.  I don't think a 2-core CPU is up to the task, even if you have enough physical memory.

vishalmalli, 2019-11-12:
Hey everyone, I had the same out of memory error. sdrtrunk had been running smoothly on a dedicated box (i7, 8GB) for almost a year, and then it started crashing. The fix for me was deleting the event_logs folder, which had accumulated to over a million files, several GB in size. I then disabled logging for the channels I am monitoring. sdrtrunk is working smoothly again.

Nokoa, 2022-02-01:
I was having the same issue, my event_logs folder was 32GB... However after deleting it the issue kept happening. SDRTrunk has been running fine on my machine before. But now I am getting that same error, and it's using a very high percentage of the CPU. No audio gets decoded now, and when viewing the messages tab of the control channels it is moving extremely slow. So slow that I can read each message. Not sure what changed, maybe my CPU died?