DSheirer/sdrtrunk #563: wont recognize the dongle, rtlsdr v3.  macosx high sierra.  got

texas1880, 2019-06-01:
20:25:24.246 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [/Users/markzachary/SDRTrunk/settings/settings.xml]  [9MB/128MB 7%]
20:25:24.782 INFO  i.g.d.s.m.MixerManager - loading system mixer devices
	[LOADED]     Input:  Default Audio Device CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: Default Audio Device CHANNELS: [MONO, STEREO]
	[LOADED]     Input:  Built-in Microphone CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Input:  Built-in Input CHANNELS: [LEFT, RIGHT, MONO]
	[LOADED]     Output: Built-in Output CHANNELS: [MONO, STEREO]
  [20MB/128MB 15%]
20:25:24.784 INFO  i.g.d.s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations  [20MB/128MB 15%]
20:25:24.821 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 16777478  [20MB/128MB 16%]
20:25:24.824 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.22.11312  [20MB/128MB 16%]
20:25:24.826 INFO  i.g.d.s.t.TunerManager - discovered [9] attached USB devices  [20MB/128MB 16%]
20:25:24.841 ERROR i.g.d.s.t.r.RTL2832TunerController - error while determining tuner type  [21MB/128MB 16%]
io.github.dsheirer.source.SourceException: couldn't claim usb interface [LIBUSB_ERROR_OTHER]
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.claimInterface(RTL2832TunerController.java:201)
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.identifyTunerType(RTL2832TunerController.java:350)
	at io.github.dsheirer.source.tuner.TunerManager.initRTL2832Tuner(TunerManager.java:383)
	at io.github.dsheirer.source.tuner.TunerManager.initTuner(TunerManager.java:225)
	at io.github.dsheirer.source.tuner.TunerManager.initTuners(TunerManager.java:140)
	at io.github.dsheirer.source.tuner.TunerManager.<init>(TunerManager.java:72)
	at io.github.dsheirer.source.SourceManager.<init>(SourceManager.java:48)
	at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:185)
	at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:639)
20:25:24.865 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:2838] NOT LOADED: SDRTRunk doesn't currently support RTL2832 Dongle with [UNKNOWN] tuner for tuner class[USB Tuner:RTL2832_VARIOUS Vendor:RTL2832 Device:SDR Address:0BDA:2838]  [22MB/128MB 17%]
20:25:24.865 INFO  i.g.d.s.t.TunerManager - usb device [05AC:8218] NOT LOADED: Unknown Device  [22MB/128MB 17%]
20:25:24.866 INFO  i.g.d.s.t.TunerManager - usb device [05AC:0236] NOT LOADED: Unknown Device  [22MB/128MB 17%]
20:25:24.867 INFO  i.g.d.s.t.TunerManager - usb device [05AC:8403] NOT LOADED: Unknown Device  [22MB/128MB 17%]
20:25:24.867 INFO  i.g.d.s.t.TunerManager - usb device [05AC:8242] NOT LOADED: Unknown Device  [22MB/128MB 17%]
20:25:24.872 INFO  i.g.d.s.t.TunerManager - usb device [05AC:8507] NOT LOADED: Unknown Device  [22MB/128MB 17%]
20:25:24.872 INFO  i.g.d.s.t.TunerManager - usb device [0A5C:4500] NOT LOADED: Unknown Device  [22MB/128MB 17%]
20:25:24.873 INFO  i.g.d.s.t.TunerManager - usb device [0424:2514] NOT LOADED: Unknown Device  [22MB/128MB 17%]
20:25:24.874 INFO  i.g.d.s.t.TunerManager - usb device [0424:2514] NOT LOADED: Unknown Device  [22MB/128MB 17%]
20:25:24.950 INFO  i.g.d.a.AudioPacketManager - Audio packet manager started  [23MB/128MB 18%]
20:25:27.222 INFO  i.g.d.icon.IconManager - loading icons file [/Users/markzachary/SDRTrunk/settings/icons.xml]  [37MB/128MB 29%]
20:25:27.642 INFO  i.g.d.p.PlaylistManager - Loading playlist file [/Users/markzachary/SDRTrunk/playlist/default.xml]  [17MB/128MB 13%]
20:25:27.740 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [18MB/128MB 14%]

texas1880, 2019-06-01:
so i installed sdrtrunk, made it through that, used java 8, that works-java 12 does not, set java home properly, got the p25 libraries installed no problem there, but now starting up i dont have a tuner.  

cubic sdr works fine.

hans109h, 2019-06-01:
In the google group Denny recently reported that he ordered one of these dongles to experiment with as other were having similar results to yours.  He hasn't reported back as far as I am aware, but he's been focusing on P25 phase II development.

DSheirer, 2019-06-02:
I got the new RTL-SDR.com V3 dongle.  I'll have to wait to try it on my daughter's macbook.  

On linux the dongle is recognized and usable by sdrtrunk.  This seems to be an issue that's specific to OSX High Sierra.

DSheirer, 2019-12-21:
I was able to test this on my daughter's laptop and it's running 10.15 catalina and it works correctly with the V3 dongle.  I don't have any means to test on older versions.

I updated the minimum system requirements for OSX to be 10.14.6 or higher.  If you find other versions that work, please let me know.

closed 2019-12-21 by DSheirer