DSheirer/sdrtrunk #565: No Audio Output

brookesrook, 2019-06-19:
I'm running SDR-trunk-0.4.0 and I'm getting no audio out of my speakers... log below. 

I do get a waterfall that is functioning... like I can see the SDR is picking up activity. I'm just not getting any audio. 

23:04:04.683 INFO  i.g.d.log.ApplicationLog - Application Log File: C:\Users\SDRTrunk\logs\sdrtrunk_app.log
23:04:04.708 INFO  i.g.d.log.ApplicationLog -
23:04:04.708 INFO  i.g.d.log.ApplicationLog - *******************************************************************
23:04:04.710 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***
23:04:04.710 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***
23:04:04.711 INFO  i.g.d.log.ApplicationLog - *******************************************************************
23:04:04.711 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]
23:04:04.712 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Windows 10
23:04:04.712 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64
23:04:04.713 INFO  i.g.d.log.ApplicationLog - Host OS Version:       10.0
23:04:04.714 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        4
23:04:04.717 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  1 GB
23:04:04.718 INFO  i.g.d.log.ApplicationLog - Storage Directories:
23:04:04.720 INFO  i.g.d.log.ApplicationLog -  Application Root: C:\Users\SDRTrunk
23:04:04.723 INFO  i.g.d.log.ApplicationLog -  Application Log:  C:\Users\SDRTrunk\logs
23:04:04.724 INFO  i.g.d.log.ApplicationLog -  Event Log:        C:\Users\SDRTrunk\event_logs
23:04:04.725 INFO  i.g.d.log.ApplicationLog -  Playlist:         C:\Users\SDRTrunk\playlist
23:04:04.726 INFO  i.g.d.log.ApplicationLog -  Recordings:       C:\Users\SDRTrunk\recordings
23:04:04.736 INFO  i.g.d.util.ThreadPool - Application thread pool created with [4] threads
23:04:04.737 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: C:\Users\SDRTrunk
23:04:04.746 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [C:\Users\SDRTrunk\SDRTrunk.properties]
23:04:04.747 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [C:\Users\SDRTrunk\SDRTrunk.properties]
23:04:04.775 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [C:\Users\SDRTrunk\settings\settings.xml]
23:04:05.119 INFO  i.g.d.s.m.MixerManager - loading system mixer devices
        [LOADED]     Output: Primary Sound Driver CHANNELS: [MONO, STEREO]
        [LOADED]     Output: Speakers (Realtek(R) Audio) CHANNELS: [MONO, STEREO]

23:04:05.122 INFO  i.g.d.s.r.RecordingSourceManager - RecordingSourceManager - discovered [0] recording configurations
23:04:05.329 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 16777478
23:04:05.329 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.22.11312
23:04:05.344 INFO  i.g.d.s.t.TunerManager - discovered [5] attached USB devices
23:04:05.360 INFO  i.g.d.s.t.TunerManager - usb device [17EF:6099] NOT LOADED: Unknown Device
23:04:07.219 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:2838] LOADED: RTL2832 SDR/R820T 00000001
23:04:07.219 INFO  i.g.d.s.t.TunerManager - usb device [046D:C52B] NOT LOADED: Unknown Device
23:04:07.219 INFO  i.g.d.s.t.TunerManager - usb device [8087:0A2A] NOT LOADED: Unknown Device
23:04:07.219 INFO  i.g.d.s.t.TunerManager - usb device [8086:A2AF] NOT LOADED: Unknown Device
23:04:07.266 INFO  i.g.d.a.AudioPacketManager - Audio packet manager started
23:04:08.033 INFO  i.g.d.icon.IconManager - loading icons file [C:\Users\SDRTrunk\settings\icons.xml]
23:04:08.267 INFO  i.g.d.p.PlaylistManager - Loading playlist file [C:\Users\SDRTrunk\playlist\default.xml]
23:04:08.346 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui
23:04:10.919 INFO  i.g.d.c.c.ChannelAutoStartFrame - Starting [1] channels now - user invoked
23:04:10.992 INFO  i.g.d.m.d.p.a.P25AudioModule - Loading JMBE library from [C:\SDR\sdr-trunk v9\bin\jmbe-0.3.3.jar]
23:04:11.009 INFO  i.g.d.m.d.p.a.P25AudioModule - JMBE audio conversion library successfully loaded - P25 audio will be available
23:04:11.029 DEBUG i.g.d.s.t.m.HeterodyneChannelSourceManager - Created/registered complex sample delay buffer of size [80] buffers and delay duration [2000ms] for tuner controller [io.github.dsheirer.source.tuner.rtl.r820t.R820TTunerController] with buffer duration [25]

DSheirer, 2019-06-19:
Are you seeing additional traffic channels being created as calls occur?  

You can right-click on the audio panel (left/right/mono) and select a different mixer (audio card/output).  sdrtrunk uses the system default audio mixer, but if you have your speakers plugged into a different sound card, then you might have to adjust this setting.

brookesrook, 2019-06-19:
No additional traffic channels being created. I do see it say "2 channels unlocked" or "1 channel unlocked" occasionally. 

I'm wondering when I compiled JMBE should I have moved more than jmbe-0.3.3.jar into sdr-trunk/bin/?

DSheirer, 2019-06-20:
No additional files are needed ... it looks like you have the JMBE library setup correctly.

Next thing to check ... in the 'Now Playing' screen, select the control channel that you enabled for decoding and then look in the bottom screen under the Messages tab and make sure it's continuously showing messages decoding.

If you're getting a good decode, then look at the 'Events' tab and see if call events are showing.  Look in the details column to see any details for each call event.

twigs538, 2020-04-05:
I have a similar problem... Running Ubuntu 18 and sdrtrunk v4 final with jmbe 1.0.4. 
[image]
[image]
[image]
[image]

DSheirer, 2020-04-06:
@twigs538 There's several issues with your setup:
1) P25 Phase 2 systems use a Phase 1 control channel.  Setup 1 channel config that uses the Phase 1 decoder and the control channel frequency.  Disable all of the Phase 2 channel configurations that you're currently using.

2) It looks like you're picking up a lot of interference/noise on your spectral display.  All of those spikes that are spaced out the same distance apart appear to be noise.  You may want to investigate eliminating that noise.

As a side note, the reason you're not getting audio out of each of the traffic channels that you've setup is because you need to specify the WACN, SYSTEM and NAC in the phase 2 config.  Your system doesn't broadcast those values on Phase 2 channels, therefore sdrtrunk can't auto-discover them.  When you use the Phase 1 decoder on the control channel, sdrtrunk is able to discover those values from the phase 1 control channel and it automatically sets up each of the phase 2 traffic channels for you.

twigs538, 2020-04-06:
I'll try and eliminate the noise by setting a PPM from rtl_test.

I don't know if the channel I input is the control channel. 

Does sdrtrunk discover traffic channels if the channel isn't the control channel?

Can't I get the control channel from the traffic channels?

DSheirer, 2020-04-07:
The noise is likely being picked up either by the antenna, or leaking in through the cabling or USB cable.  It may be coming from your computer.

Do you have a link to a list of frequencies for the system, such as radioreference.com?  A NAC code?

When you've selected the correct phase 1 control channel, the status column will light up with an orange label instead of the blue label you're seeing for calls.

You definitely have a good traffic channel for frequency  443.9896 ... just need to get the correct WACN, System and NAC from the control channel.

OccamsTaser, 2020-04-23:
I'm having a similar issue to all of this.  i have the JMBE compiled and installed correctly.  as SDRTrunk is loading, in the console it says 

12:19:14.305 INFO  i.g.d.a.c.m.JmbeAudioModule - Loading JMBE library from [C:\Users\space\Desktop\jmbe-1.0.4\codec\build\libs\jmbe-1.0.4.jar]  [33MB/50MB 67%]
12:19:14.313 INFO  i.g.d.a.c.m.JmbeAudioModule - JMBE audio conversion library loaded: JMBE Audio Conversion Library v1.0.4  [34MB/50MB 68%]
12:19:14.314 INFO  i.g.d.a.c.m.ImbeAudioModule - JMBE audio conversion library IMBE CODEC successfully loaded - P25-1 audio will be available  [34MB/50MB 68%]
12:19:14.331 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [35MB/50MB 70%]
12:20:31.359 INFO  i.g.d.a.c.m.AmbeAudioModule - AMBE CODEC successfully loaded - P25-2/DMR/NXDN audio will be available  [49MB/64MB 76%]

I am getting traffic and calls coming through but just no audio is coming out.  If i change to NBFM i do get the noise, but it just seems that, for whatever reason, JMBE is not working and for the life of me I cant figure out why..
[image]
[image]

DSheirer, 2020-04-24:
@OccamsTaser can you check the PPM on your tuner(s) ... in the first screen shot (attached) you have 2 traffic channels active, but the signal spikes look like they're out of adjustment.  What's weird is the control channel spike which is at the far right of the  screen seems aligned, but the signal that it's aligned with is weak/fading.
[image]

OccamsTaser, 2020-04-24:
@DSheirer My PPM was checked for Auto-Correction.  I unchecked and changed it, but still no sound...
[image]
[image]

lwvmobile, 2020-06-16:
@DSheirer 
> As a side note, the reason you're not getting audio out of each of the traffic channels that you've setup is because you need to specify the WACN, SYSTEM and NAC in the phase 2 config. Your system [...]

Just out of curiosity, I stumbled upon this because I wasn't getting any audio out of P25 Phase 2 channels in Linux, and I was curious if there was a way to tell SDR Trunk to just play all audio calls if there is no specified values for WACN, SYSTEM, and NAC. The reason I ask is in some areas, those exact values may not be entirely known, or just exploring what signals are available I find plenty of P25 phase 2 channels but no corresponding control channel. Either that, or the CC is pretty weak at different times of the day due to distance or weather related issues. In some ways, it would be nice to just be able to listen and figure out what is there, unless there is a technical type of issue where you have to know those values for decoding.

DSheirer, 2020-06-16:
@lwvmobile Some Phase II traffic channels do broadcast those parameters, but that seems to be rare.

The 3 values (wacn, system and nac) are required to descramble most of the phase 2 MAC messages, including the audio frames.  Without correct values, the audio contents are scrambled and can't be decoded.

closed 2023-11-10 by DSheirer