DSheirer/sdrtrunk #566: Not working on Mac OSX Mojave

quochuy, 2019-07-10:
I'm getting this error on OSX

`15:39:16,224 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback-test.xml]
15:39:16,225 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback.groovy]
15:39:16,225 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Found resource [logback.xml] at [jar:file:/Users/huyn/Downloads/sdr-trunk-osx-x64/lib/sdr-trunk-0.4.0-alpha.9.jar!/logback.xml]
15:39:16,226 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs multiple times on the classpath.
15:39:16,226 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs at [jar:file:/Users/huyn/Downloads/sdr-trunk-osx-x64/lib/sdr-trunk-0.4.0-alpha.9.jar!/logback.xml]
15:39:16,226 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs at [jar:file:/Users/huyn/Downloads/sdr-trunk-osx-x64/lib/radio-reference-api-15.0.1.jar!/logback.xml]
15:39:16,244 |-INFO in ch.qos.logback.core.joran.spi.ConfigurationWatchList@27ae2fd0 - URL [jar:file:/Users/huyn/Downloads/sdr-trunk-osx-x64/lib/sdr-trunk-0.4.0-alpha.9.jar!/logback.xml] is not of type file
15:39:16,358 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - debug attribute not set
15:39:16,358 |-INFO in ch.qos.logback.core.joran.action.ConversionRuleAction - registering conversion word memory_usage with class [io.github.dsheirer.util.MemoryUsageLogger]
15:39:16,358 |-INFO in ch.qos.logback.core.joran.action.TimestampAction - Using current interpretation time, i.e. now, as time reference.
15:39:16,386 |-INFO in ch.qos.logback.core.joran.action.TimestampAction - Adding property to the context with key="appstart" and value="153916" to the LOCAL scope
15:39:16,386 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - About to instantiate appender of type [ch.qos.logback.core.ConsoleAppender]
15:39:16,394 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - Naming appender as [STDOUT]
15:39:16,399 |-INFO in ch.qos.logback.core.joran.action.NestedComplexPropertyIA - Assuming default type [ch.qos.logback.classic.encoder.PatternLayoutEncoder] for [encoder] property
15:39:16,421 |-INFO in ch.qos.logback.classic.joran.action.RootLoggerAction - Setting level of ROOT logger to DEBUG
15:39:16,421 |-INFO in ch.qos.logback.core.joran.action.AppenderRefAction - Attaching appender named [STDOUT] to Logger[ROOT]
15:39:16,422 |-INFO in ch.qos.logback.classic.joran.action.LoggerAction - Setting level of logger [org.apache.mina] to INFO
15:39:16,422 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - End of configuration.
15:39:16,424 |-INFO in ch.qos.logback.classic.joran.JoranConfigurator@29176cc1 - Registering current configuration as safe fallback point

Exception in thread "main" java.lang.NullPointerException
        at java.desktop/java.awt.Window.init(Unknown Source)
        at java.desktop/java.awt.Window.<init>(Unknown Source)
        at java.desktop/java.awt.Frame.<init>(Unknown Source)
        at java.desktop/java.awt.Frame.<init>(Unknown Source)
        at java.desktop/javax.swing.JFrame.<init>(Unknown Source)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:119)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:639)
`

DSheirer, 2019-07-10:
It's throwing an error at the point where it's trying to create the main user interface window.  It there anything different about the computer that you're running it on, like video cards or monitor?  

Are you using the MAC OS specific sdrtrunk build?

quochuy, 2019-07-10:
@DSheirer thanks for the response.
I'm just using a Macbook Pro 15 inch 2017.
[image]

I used sdr-trunk-0.4.0-alpha.9-osx-x64.zip

closed 2019-12-21 by DSheirer