DSheirer/sdrtrunk #568: Build Error with Gradle

KrumbEMT, 2019-08-11:
Every time I try and build the code using Gradlew, It completes without error, Generating the proper .jar file. However when starting the jar file, it throws the following error, no matter the OS.

> no main manifest attribute, in /sdrtrunk-296-p25-phase-2-decoder-3/build/libs/sdr-trunk-0.4.0-beta.1.jar

Tested on Windows 10, macOS 10.15, macOS 10.14, Ubuntu 18.04

DSheirer, 2019-08-12:
I believe I removed that method of launching the program.  Now that the gradle script bundles the Java runtime with the output, you have to run the launch script in the bin directory of the output product.  

If you have to run it from the jar, you can use this command:
java -jar sdr-trunk-0.4.0-beta.1.jar io.github.dsheirer.gui.SDRTrunk

KrumbEMT, 2019-08-12:
Not sure if I am building the program wrong, but both methods are still generating errors. Below is the terminal output of starting SDRTrunk form the script built with ./gradlew build 
OS: macOS 10.15

>/sdrtrunk-296-p25-phase-2-decoder-2/build/scripts/sdr-trunk ; exit;
/sdrtrunk-296-p25-phase-2-decoder-2/build/scripts/sdr-trunk: line 154: /sdrtrunk-296-p25-phase-2-decoder-2/build/bin/java: No such file or directory
/sdrtrunk-296-p25-phase-2-decoder-2/build/scripts/sdr-trunk: line 154: exec: /sdrtrunk-296-p25-phase-2-decoder-2/build/bin/java: cannot execute: No such file or directory
[Process completed]

Perhaps it is something with the Beta of macOS, I will test on a different OS tonight. Thanks

KrumbEMT, 2019-08-12:
More info from trying a few things. After copying my java install folder to the build folder, I get basically the same error as calling the jar from the command line. 

> Error: Could not find or load main class io.github.dsheirer.gui.SDRTrunk

DSheirer, 2019-08-13:
The build script bundles the Java runtime with the output of the build.  In order to do that, it needs the location of your Java Development Kit (JDK) installation.  If you open the build.gradle file, there is a section where it points to the JDK for each of the operating systems:

/**
 * Java Development Kit (JDK) locations.  These paths must point to the 'bin' directory
 * within a JDK for each of the specified architectures.  These JDKs are used by the 
 * runtime and runtimeZip tasks to produce platform-specific builds
 */
def jdkLinux_x64 = '/media/denny/WD250GB/java/bellsoft/linux-x64/jdk-11.0.2'
def jdkOsx_x64 = '/media/denny/WD250GB/java/bellsoft/osx-x64/jdk-11.0.2.jdk'
def jdkWindows_x64 = '/media/denny/WD250GB/java/bellsoft/windows-x64/jdk-11.0.2'

The path that is listed should point to the folder that contains the 'bin' directory for the JDK.  You can probably get away with just listing the OSX JDK location since you're not interested in the other 2 OSs.

Once you update the OSX JDK path, then run the following:

./gradlew distZip

this will create the zipped sdrtrunk build product.  When you unzip it, then run the sdrtrunk launch script from there.

closed 2019-12-09 by KrumbEMT