DSheirer/sdrtrunk #569: Alias List explicit match for decode

NavyBOFH, 2019-08-11:
I have read into the features and used the Playlist web tool to create my Alias list. However I end up with the following issue. When I set a mask for "all other aliases" to ignore (do not monitor), the software still decodes the audio just doesn't play it - tying up resources on the SDR receivers and system... creating slow-downs while showing active calls for talkgroups either encrypted or not in my alias list.

I am hoping for either a feature change or perhaps guidance if this is just something that was not defined well by documentation. I am running an "Atomic Pi" Atom board using 2x RTL-SDR dongles and Ubuntu 19.04. System behaves well with 25% CPU usage and 40% RAM usage at idle - but my issue ends up being that I have 3-4 talkgroups normally monitored, and another ~20 sorted into Priority 2/3 (mutual aid and interop TGs) that are normally quiet... but the trunked system also has another ~15+ TGs that are encrypted or not applicable for the area which quickly turn the setup from "tolerable" to "stalled".

DSheirer, 2019-08-12:
There's currently no option to do what you're describing ... to not follow a traffic channel.  I'll keep this ticket open to add that feature.

chris7948, 2020-05-05:
Would really like to see this feature. Area is to busy to listen to it all :)