DSheirer/sdrtrunk #571: Low decoded audio volume

LordDonkey, 2019-08-30:
The decoded P25 P2 audio seems very quiet. The volume within SDRTrunk set to 100 and the computer volume is set to 100 and it is still quiet. Is there a workaround to try to get the volume any higher? 

On a side note, sometimes it sounds like they are talking through a fan. It doesn't happen all the time. Most of the time it sounds pretty good. I included two clips to help demonstrate the "fan" sound. 
[Clip 1](https://soundcloud.com/michael-nugent-500351597/20190829-2117031186-25-traffic-to-7521-from-2138470/s-wWQsC)
[Clip 2: There was a fire tone out at the beginning of this clip, it didn't make its way through](https://soundcloud.com/michael-nugent-500351597/20190829-2123411186-25-traffic-to-12601-from-2135505/s-dwBK1)

DSheirer, 2019-08-30:
Would you be able to get me a TRAFFIC CHANNEL DEMODULATED BITSTREAM recording of a traffic channel that has the fire tone out?

In the control channel configuration on the recording tab, you can enable this type of recording.  This is going to probably generate a number of *.bits recordings for you until you get the right call type with the tone out.

This would help me to figure out why the tones are not generating correctly in the JMBE for those tone outs.

LordDonkey, 2019-08-31:
Ok, I have it set to record the traffic channel demodulated bitstream. I'll just have to wait until there's a call now. On a side note, the fire tone outs are encoded audio patched from the old analog system over to the new phase 2 system. The county is still in the transition to digital.

DSheirer, 2019-08-31:
If the tone outs are patched from the audio panel, the AMBE codec may still recognize them as tones and encode them as tone frames.  I don't have any actual tone frames to test against, so I haven't been able to test that part of the code yet.  I did test the tone generators, but there may be a problem with the tone frame message decoding.

LordDonkey, 2019-09-01:
[20190831_2251141186_25_TRAFFIC__TO_12601_FROM_2135505.zip](https://github.com/DSheirer/sdrtrunk/files/3563062/20190831_2251141186_25_TRAFFIC__TO_12601_FROM_2135505.zip)
I heard some silence before they started talking. Hopefully there were tones. I uploaded 2 bits files and the corresponding recordings. Both times cut out at the end with some stuttering. I've been hearing a lot of that garbled stuttering too, entire voice calls do it sometimes.

DSheirer, 2019-09-01:
Which system is this from?  Or, can you post the WACN and System values?  It looks like the NAC is 1189 (4A5).

LordDonkey, 2019-09-01:
Yes the NAC is 1189 (4A5), the system is 1186 (4A2) and the WACN is 781824 (BEE00).

LordDonkey, 2019-09-05:
Here are 2 bits files that should contain AMBE tones from the neighboring county. It is the same NAC and System. 
[20190905_115415_9600BPS_APCO25PHASE1_1186_25_TRAFFIC.zip](https://github.com/DSheirer/sdrtrunk/files/3580142/20190905_115415_9600BPS_APCO25PHASE1_1186_25_TRAFFIC.zip)

LordDonkey, 2019-10-07:
[20191007_1810431186_25_TRAFFIC__TO_12601_FROM_2135505.zip](https://github.com/DSheirer/sdrtrunk/files/3699896/20191007_1810431186_25_TRAFFIC__TO_12601_FROM_2135505.zip)

Here is another tone out. The lower frequency tones could be heard (346.7 KHz,330.5 KHz), the other ones are just hisses.