DSheirer/sdrtrunk #577: Error while processing polyphase channel samples

Hermite123456, 2019-09-25:
Hello, 

I just installed Alpha on Windows 64. (This is a great program!!!) Thanks Denny.

While running Phase 2 decoding (using 2 Airspy R2 tuners, I randomly see these messages in the Log:

20190925 153625.438 [main] DEBUG i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2500000.0] providing [100] channels at [25000.0] Hz each  [23MB/256MB 9%]
20190925 153625.443 [main] INFO  i.g.d.s.tuner.TunerManager - usb device [1D50:60A1] LOADED: Airspy 573064DC-307DA4C1  [23MB/256MB 9%]
20190925 153625.472 [main] INFO  i.g.d.audio.AudioPacketManager - Audio packet manager started  [25MB/256MB 9%]
20190925 153626.018 [main] INFO  i.g.dsheirer.icon.IconManager - loading icons file [C:\Users\quietpc\SDRTrunk\settings\icons.xml]  [24MB/40MB 60%]
20190925 153626.173 [main] INFO  i.g.d.playlist.PlaylistManager - Loading playlist file [C:\Users\quietpc\SDRTrunk\playlist\default.xml]  [19MB/40MB 47%]
20190925 153626.225 [main] INFO  i.github.dsheirer.gui.SDRTrunk - starting main application gui  [21MB/40MB 52%]
20190925 153745.876 [AWT-EventQueue-0] INFO  i.g.d.a.c.mbe.JmbeAudioModule - Loading JMBE library from [C:\Users\quietpc\Downloads\sdrtrunk-296-p25-phase-2-decoder\sdrtrunk-296-p25-phase-2-decoder\jmbe-5-ambe-codec-2\codec\build\libs\jmbe-1.0.0.jar]  [28MB/52MB 54%]
20190925 153745.883 [AWT-EventQueue-0] INFO  i.g.d.a.c.mbe.JmbeAudioModule - JMBE audio conversion library loaded: JMBE Audio Conversion Library v1.0.0  [28MB/52MB 55%]
20190925 153745.884 [AWT-EventQueue-0] INFO  i.g.d.a.c.mbe.ImbeAudioModule - JMBE audio conversion library IMBE CODEC successfully loaded - P25-1 audio will be available  [28MB/52MB 55%]
20190925 153745.896 [AWT-EventQueue-0] DEBUG i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2500000.0] providing [100] channels at [25000.0] Hz each  [29MB/52MB 57%]
20190925 153746.595 [sdrtrunk pool-1-thread-4] INFO  i.g.d.a.c.mbe.AmbeAudioModule - AMBE CODEC successfully loaded - P25-2/DMR/NXDN audio will be available  [44MB/80MB 55%]
20190925 153746.622 [sdrtrunk pool-1-thread-4] DEBUG i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2500000.0] providing [100] channels at [25000.0] Hz each  [46MB/80MB 57%]
20190925 153751.710 [sdrtrunk pool-1-thread-2] ERROR i.g.d.d.f.c.o.ChannelOutputProcessor - Error while processing polyphase channel samples  [57MB/123MB 46%]
java.lang.ClassCastException: class io.github.dsheirer.module.decode.p25.phase1.DecodeConfigP25Phase1 cannot be cast to class io.github.dsheirer.module.decode.p25.phase2.DecodeConfigP25Phase2 (io.github.dsheirer.module.decode.p25.phase1.DecodeConfigP25Phase1 and io.github.dsheirer.module.decode.p25.phase2.DecodeConfigP25Phase2 are in unnamed module of loader 'app')
	at io.github.dsheirer.module.decode.p25.P25TrafficChannelManager.processPhase2ChannelGrant(P25TrafficChannelManager.java:437)
	at io.github.dsheirer.module.decode.p25.P25TrafficChannelManager.processChannelGrant(P25TrafficChannelManager.java:231)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.processTSBK(P25P1DecoderState.java:1206)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.receive(P25P1DecoderState.java:281)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderState.receive(P25P1DecoderState.java:142)
	at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:152)
	at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:59)
	at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:103)
	at io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:96)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageProcessor.receive(P25P1MessageProcessor.java:95)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageProcessor.receive(P25P1MessageProcessor.java:37)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.dispatchMessage(P25P1MessageFramer.java:311)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.receive(P25P1MessageFramer.java:198)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1MessageFramer.receive(P25P1MessageFramer.java:55)
	at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:152)
	at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:59)
	at io.github.dsheirer.dsp.psk.PSKDemodulator.broadcast(PSKDemodulator.java:58)
	at io.github.dsheirer.dsp.psk.DQPSKDecisionDirectedDemodulator.calculateSymbol(DQPSKDecisionDirectedDemodulator.java:88)
	at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:115)
	at io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:89)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderC4FM.receive(P25P1DecoderC4FM.java:108)
	at io.github.dsheirer.module.decode.p25.phase1.P25P1DecoderC4FM.receive(P25P1DecoderC4FM.java:40)
	at io.github.dsheirer.sample.buffer.ReusableBufferBroadcaster.broadcast(ReusableBufferBroadcaster.java:42)
	at io.github.dsheirer.sample.buffer.ReusableBufferBroadcaster.broadcast(ReusableBufferBroadcaster.java:23)
	at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:59)
	at io.github.dsheirer.sample.buffer.ReusableComplexBufferAssembler.flush(ReusableComplexBufferAssembler.java:158)
	at io.github.dsheirer.sample.buffer.ReusableComplexBufferAssembler.receive(ReusableComplexBufferAssembler.java:132)
	at io.github.dsheirer.sample.buffer.ReusableComplexBufferAssembler.receive(ReusableComplexBufferAssembler.java:144)
	at io.github.dsheirer.dsp.filter.channelizer.output.OneChannelOutputProcessor.process(OneChannelOutputProcessor.java:97)
	at io.github.dsheirer.dsp.filter.channelizer.output.ChannelOutputProcessor.processChannelResults(ChannelOutputProcessor.java:124)
	at io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.processSamples(PolyphaseChannelSource.java:248)
	at io.github.dsheirer.source.tuner.channel.TunerChannelSource$ScheduledIntervalProcessor.run(TunerChannelSource.java:312)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at java.base/java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:835)

Tks.

DSheirer, 2019-09-26:
Good catch.  It looks like this is happening when they allocate an SNDCP DATA channel and the data channels are not Phase II, they're Phase 1.  I'll get this fixed.  Thanks for the report.

DSheirer, 2019-11-17:
Resolved on P25 Phase II branch.

closed 2019-11-17 by DSheirer