DSheirer/sdrtrunk #578: Alias Audio Identifier does not generate a text box for input

Hermite123456, 2019-09-25:
Hello,
I'm running sdr-trunk-all-0.4.0-beta.1 on Windows 10/64.
When I try to create an Alias from the Audio/Identifier tab, and I press the 'NEW' button and select any of the 5 options, it never returns a text box for my input.
Anyone else observing this behavior?
Tks.
(SDRTrunk 101 novice here...)

DSheirer, 2019-09-26:
When you click the new button and select one of the 5 options, it inserts a new entry in the list and auto-selects the new item so that the editor changes to an entry form (if appropriate).

Are you saying that the new item is not being added to the list, or is it not being auto-selected?

Hermite123456, 2019-09-26:
Rookie mistake on my behalf. I was not using the ‘new’ button from the alises tab, but the ‘new’ button  from the Audio/Identifier tab without first creating the new entry. Thank you.

closed 2019-09-26 by Hermite123456