DSheirer/sdrtrunk #580: MPT-1327 Invalid Channel Map

Hermite123456, 2019-09-26:
Hello,
I'm using sdr-trunk-all-0.4.0-beta.1 on Windows 10. I'm trying to setup MPT1327 channel mapping but it appears to be invalid. I am comparing my setup with an old Winradio MPT1327 monitor.
Basically, the CC No. is 805 (401.00000 base frequency). I know of 2 active traffic channels No. (745 & 931). I setup SDRTrunk channel mapping with a range 0-1025 with a base of 401.000000 @12500 stepsize. (First 401.000000, Last 413.812500). 
Why am I getting '**Invalid channel map. No frequency for channel xxx**' ?
Tks.
[image]
[image]

DSheirer, 2019-09-27:
In the channel configuration for the MPT1327 traffic channel, do you have the channel map listed with the same name that you assigned to the map 'Text MPT1327'??

Hermite123456, 2019-09-27:
> In the channel configuration for the MPT1327 traffic channel, do you have the channel map listed with the same name that you assigned to the map 'Text MPT1327'??

Yes I do. I also renamed it to remove upper/lower case and blank space just in case.

[image]

[image]

Find below except of call_events & decoded_messages log that occurred @ 08:04:37 ;

"2019:09:27:08:04:37","","MPT-1327","Call","9354","9487","745","","","Invalid Channel Map - No Frequency For Channel 745"

20190927 080437,PASSED,BCAST - System Parameters SYSTEM:1801 CALL MAINT PARAMETERS PERIODIC MAINT MSG REQD:true INTERVAL:0(sec) PRESSEL ON REQD:false IDENT1 VALUE:INDIVIDUAL ADDRESS
20190927 080437,PASSED,ALH - Aloha SYSTEM:1801 WAIT:1 RSVD:0 M:0 N:0
20190927 080437,PASSED,AHOY - General Availability Check
20190927 080437,PASSED,ALH - Aloha SYSTEM:1801 WAIT:1 RSVD:0 M:20 N:0
**20190927 080437,PASSED,GTC - Goto Channel FROM:001-1162 TO:001-1295 CHAN:745
20190927 080437,PASSED,GTC - Goto Channel FROM:001-1162 TO:001-1295 CHAN:745
20190927 080437,PASSED,GTC - Goto Channel FROM:001-1162 TO:001-1295 CHAN:745**
20190927 080437,PASSED,ALH - Aloha SYSTEM:1801 WAIT:1 RSVD:0 M:0 N:10
20190927 080437,PASSED,BCAST - System Parameters SYSTEM:1801 REGISTRATION PARAMETERS

jurek4321, 2020-01-19:
I have the same problem.
[image]
[image]

lama9800, 2020-03-28:
I can also confirm the same problem
[image]
Channel Map data is correct as per
https://vicradiozone.com/smr-network/smr-network-channel-plan/
[image]

DSheirer, 2020-03-28:
Can you please post a screen shot of your channel configuration for the control channel with the Decoder Tab displayed?

lama9800, 2020-03-28:
sure
[image]

DSheirer, 2020-03-28:
@lama9800 your channel map looks good.  In the channel map editor, select the SMR CH MAP and then in the right side click each channel range.  Look at the bottom of of the editor (just above the buttons) and make sure that it's showing the start and end channel numbers and frequencies for each.

Next, go to the control channel configuration and in the decoder tab clear the channel map name and then save.  Then, select the channel map name from the drop down box and save again.  Start the channel and see if this resolves the error.

Since the channel map looks valid, I'm guessing the disconnect is that the channel configuration might have an issue that's preventing it from selecting the correct channel map when the channel starts, thus why I want to clear the map and then reset it and save.

lama9800, 2020-03-28:
@DSheirer I checked the channel map start and end frequencys as you sugested and they are correct. However i am unable to clear the channel map name in the decodeing tab, I can only select the current map (SMR CH MAP). should i create another map and use that to switch between?

DSheirer, 2020-03-28:
That might work.

DSheirer, 2020-03-28:
If that doesn't work, can you post your playlist file here and I'll take a look to see why it's not picking up the channel map

lama9800, 2020-03-28:
its acting a bit strange when i try to select the "test" channel map it automaticly reverts back to the "SMR CH MAP". anyway, heres my playlist file.

[playlist.zip](https://github.com/DSheirer/sdrtrunk/files/4397230/playlist.zip)

DSheirer, 2020-03-28:
Do you have a playlist named 'playlist_v2.xml' ?

lama9800, 2020-03-28:
Is that in the same playlist directory or someware else?

DSheirer, 2020-03-28:
Disregard

DSheirer, 2020-03-28:
I loaded your playlist and ran it here locally and all seems to be good.

Can restart the application, start the channel, and see if it's still giving you invalid channel map on channel grants?

lama9800, 2020-03-28:
Ok tested it just tested it then. It is finding the correct frequencys/channels for trafic (awesome). However i am not getting any audio. I have encountered this audio bug previously when trying to use MPT decodeing however i have not found a solution yet. would you have any ideas on this?

[image]

DSheirer, 2020-03-28:
I think I may know what might be the problem.  I want to fix something and then send you a new build to try.  

Which variant of sdrtrunk are you using (windows/linux) and bits (32/64)?

lama9800, 2020-03-28:
64 bit windows

DSheirer, 2020-03-28:
I put a test build on my google drive here:
https://drive.google.com/open?id=10fdcJZc627mMClAK60Z6ctkqkSiGzM2U

Sorry that I can't test this locally.  We have MPT at a local casino and one at a school bus facility, both are shutdown for corona virus, so no activity to test against.

lama9800, 2020-03-28:
cool no worries. Might take me a bit to get a signal, its early in the morning here in the land down under :) .There still should be some activity on the network though.

ghoward02, 2020-03-28:
The test build fixed my no-audio issue with MPT1327, only for me to discover the local power company is mostly using it for data bursts...

As for the original channel map issue, this discussion prodded me to delve into my playlist file. I found that despite the GUI showing an assigned channel map the actual default.xml just had an empty <channelMapName/> for the map. Once I manually edited in start and end tags as well as the map name I was able to properly follow channel grants.

lama9800, 2020-03-28:
Audio works! Thanks @DSheirer for the help.
@ghoward02 good to see its MPT is working for you too.

[image]

DSheirer, 2020-03-28:
@ghoward02 and @lama9800 Thanks to both of you for testing.  I'll get this fix added and it will be permanent for the next release.

The channel configuration editor and the channel map issue will be resolved in the next 0.5.0 release. I've already rewritten all of that code in JavaFX.

PR #711 (2020-03-28): Resolves #580 channel map selection and editing
Resolves #580 
Updates MPT1327 channel editor to resolve issue with selecting and editing MPT1327 channel maps.
commit 277a8470 (2020-03-28): Resolves #580 channel map selection and editing (#711)
closed 2020-03-28 by DSheirer