DSheirer/sdrtrunk #593: Control channel not "locking" with multiple frequencies set

hans109h, 2019-11-19:
This was reported early on in the google group, but I couldn't find an actual ticket her for it so relaying.

Versions after phase II was introduced 4.0 (beta), channels set to "Tuner - Multiple Frequencies" find a control channel, then after a call is decoded it fails to remaining on the initial control channel and starts the scanning process until it again finds the initial control channel in the frequency list.

DSheirer, 2019-11-23:
There's a distinct difference in the P25 decoder between using the single frequency vs multi-frequency tuner sources.  Something in the multi-frequency source is causing the decoder to lose sync lock every 1-2 seconds.  In a poor signal environment this can delay the decoder recoverying the sync for over 2 seconds and causing the multi-frequency channel rotation monitor to start scanning the frequency list.

DSheirer, 2019-11-23:
This is resolved with PR #603

closed 2019-11-23 by DSheirer