DSheirer/sdrtrunk #597: Using decoders with an external source

bobbark, 2019-11-21:
This isn't really an issue.... Have you ever used your decoders with an external source?  We are using a framework called REDHAWK that already abstracts out our tuners.  I would like to use your awesome decoders in conjunction with that framework if possible.  

I created an implementation of IReusableComplexBufferProvider connected to a REDHAWK DDC.  I then attached an instance of P25DecoderC4FM as a listener and added a few print statements to see if any messages get decoded but they don't.  I am tuned to a control channel as determined by running sdrtrunk normally.

bobbark, 2019-11-21:
Looks like the issue was the sample rate.  I had the RTL set for 2.5 MHz and nothing decoded.  I changed it to 2.4 MHz and it works like a champ.  Now to get the traffic channel manager tuning external DDCs as well.

DSheirer, 2019-11-22:
I've only used them within sdrtrunk.  I looked at the REDHAWK framework a couple years ago but haven't revisited it in a while.  Let me know how it works out for you?

bobbark, 2019-11-22:
I have been using REDHAWK for a few years now and it works pretty well although the IDE is a little quirky sometimes.  I will let you know how it goes trying to use the decoders.  I am hoping to make minimal changes so I can roll in any updates you make.  If there are changes that could be made to your code to make this cleaner, and those changes make sense from your point of view of course, would you be opposed to that?  I am definitely not ready to do that at this point but wanted to throw that out there.

DSheirer, 2019-11-23:
Yes, absolutely no problem.

bobbark, 2019-12-09:
Your software is very well written.  I was able to create an application that uses your decoders for p25 and only had to modify three of your files - and that was only because I have to use Java 8.

I have a squelch broadcaster attached to my ChannelState and P25AudioModule objects so the SQUELCH messages are being passed.  I notice that in P25AudioModule, receiving a SQUELCH state is the only way to reset mEncryptedCall and mEncryptedCallStateEstablished.  Those both remain true during some unencrypted calls.  I removed the check for mEncryptedCall in P25AudioModule.processAudio and when I receive unencrypted calls while mEncryptedCall is set I am able to hear them.  Does this sound like I am missing something or a bug?