DSheirer/sdrtrunk #607: Doesn't work on macOS Sierra 10.12.6 with RTL-SDR

ah1102, 2019-11-29:
I am using "Version 0.4.0 Beta 1" and the program does not want to connect my RTL-SDR v3. I wrote that the "unknown device." Please correct the program or advise how to get the program to connect RTL-SDR.

For computers running mac os x, your application has no alternative, since the FMP24 program from the DSD + package and Unitrunker only work on Windows. I tried to run them under the Parallels Desktop virtual machine, but I ran into problems with their work there.

P.S. I would also like your program to optionally be able to output sound to external decoders, for example, dsd +.

`20:46:02.987 INFO  i.g.d.s.t.TunerManager - LibUSB API Version: 16777478
20:46:02.988 INFO  i.g.d.s.t.TunerManager - LibUSB Version: 1.0.22.11312
20:46:02.989 INFO  i.g.d.s.t.TunerManager - discovered [11] attached USB devices
20:46:03.000 ERROR i.g.d.s.t.r.RTL2832TunerController - error while determining tuner type
io.github.dsheirer.source.SourceException: couldn't claim usb interface [LIBUSB_ERROR_OTHER]
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.claimInterface(RTL2832TunerController.java:201)
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.identifyTunerType(RTL2832TunerController.java:350)
	at io.github.dsheirer.source.tuner.TunerManager.initRTL2832Tuner(TunerManager.java:383)
	at io.github.dsheirer.source.tuner.TunerManager.initTuner(TunerManager.java:225)
	at io.github.dsheirer.source.tuner.TunerManager.initTuners(TunerManager.java:140)
	at io.github.dsheirer.source.tuner.TunerManager.<init>(TunerManager.java:72)
	at io.github.dsheirer.source.SourceManager.<init>(SourceManager.java:48)
	at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:187)
	at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:670)
20:46:03.011 INFO  i.g.d.s.t.TunerManager - usb device [0BDA:2838] NOT LOADED: SDRTRunk doesn't currently support RTL2832 Dongle with [UNKNOWN] tuner for tuner class[USB Tuner:RTL2832_VARIOUS Vendor:RTL2832 Device:SDR Address:0BDA:2838]
20:46:03.012 INFO  i.g.d.s.t.TunerManager - usb device [10C4:EA70] NOT LOADED: Unknown Device
20:46:03.012 INFO  i.g.d.s.t.TunerManager - usb device [08BB:29B3] NOT LOADED: Unknown Device
20:46:03.013 INFO  i.g.d.s.t.TunerManager - usb device [0424:2512] NOT LOADED: Unknown Device
20:46:03.013 INFO  i.g.d.s.t.TunerManager - usb device [05AC:828B] NOT LOADED: Unknown Device
20:46:03.013 INFO  i.g.d.s.t.TunerManager - usb device [0A5C:4500] NOT LOADED: Unknown Device
20:46:03.014 INFO  i.g.d.s.t.TunerManager - usb device [05AC:8511] NOT LOADED: Unknown Device
20:46:03.014 INFO  i.g.d.s.t.TunerManager - usb device [0424:2412] NOT LOADED: Unknown Device
20:46:03.015 INFO  i.g.d.s.t.TunerManager - usb device [1532:005C] NOT LOADED: Unknown Device
20:46:03.015 INFO  i.g.d.s.t.TunerManager - usb device [8087:0024] NOT LOADED: Unknown Device
20:46:03.016 INFO  i.g.d.s.t.TunerManager - usb device [8087:0024] NOT LOADED: Unknown Device
20:46:03.071 INFO  i.g.d.a.AudioPacketManager - Audio packet manager started
20:46:04.686 INFO  i.g.d.icon.IconManager - loading icons file [/Users/Apple/SDRTrunk/settings/icons.xml]
20:46:04.922 INFO  i.g.d.p.PlaylistManager - Loading playlist file [/Users/Apple/SDRTrunk/playlist/default.xml]
20:46:04.985 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui
`

KrumbEMT, 2019-12-09:
How did you get Version 0.4.0? Looking through the logs it appears there is an issue with LIBUSB where it is failing to bind the USB port the SDR is plugged into, Could be a build error if you went that route. Try changing the USB port the SDR plugged into, and see if that helps.

ah1102, 2019-12-09:
I took from you an already assembled version of "sdr-trunk-0.4.0-beta.1-osx-x64.zip" https://github.com/DSheirer/sdrtrunk/releases/tag/v0.4.0-beta.1 Switching rtl -sdr in all 4 different usb imac does not help, the device does not connect to the program. Now I use the old version of "sdrtrunk-0.4.0-alpha.5" in it the old libusb library. In newer versions of your program, updated libusb is used with which an error occurs connecting the device to the program.

ah1102, 2019-12-13:
Maybe the author will replace LibUSB Version 1.0.22. on LibUSB Version 1.0.18. with which there were no problems, until finding out the reasons for the incorrect operation of Version 1.0.22? Or write how to do it yourself.

ah1102, 2019-12-16:
I upgraded to macOS Mojave 10.14 and my tuner was found. It turns out that it does not work only on macOS Sierra.

[image]

closed 2019-12-21 by DSheirer

dotsam, 2020-02-23:
This looks to have been caused by https://github.com/libusb/libusb/issues/519 which was fixed in libusb 1.0.23. This means that a new version of libusb4java will be needed that builds from this version of libusb. To get this working in the mean time, I installed libusb from Homebrew, and compiled libusb4java myself and replaced libusb4java-1.3.0-darwin-x86-64.jar in the lib folder

DSheirer, 2020-02-23:
@dotsam Thanks for investigating this.  I opened an issue https://github.com/usb4java/usb4java/issues/78 with usb4java